raspberrypi/firmware #1888: Raspberry pi 0W v1.1 randomly crashes after a few minutes

Totix92, 2024-04-23:
**Describe the bug**
The Raspberry Pi 0W freezes after a few minutes of being started.

**To reproduce**
I install Raspberry Pi OS Lite with rpi-imager, I start the raspberry and it works for a few minutes, then no matter what I make it do or idle, it freezes completely and the wifi disconnects, the same sd card on a Raspberry Pi 1B It works and the system doesn't crash.
I also tried updating the firmware via rpi-update, but it crashed, so I had to manually download firmware, kernel and modules and update them manually from my Ubuntu laptop, even so the system crashes after a few minutes.

**System**
`System Information
------------------

Raspberry Pi Zero W Rev 1.1
PRETTY_NAME="Raspbian GNU/Linux 12 (bookworm)"
NAME="Raspbian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"

Raspberry Pi reference 2024-03-15
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 11096428148f0f2be3985ef3126ee71f99c7f1c2, stage2

Linux Raspberry-Pi-0 6.6.28+ #1758 Fri Apr 19 10:36:10 BST 2024 armv6l GNU/Linux
Revision        : 9000c1
Serial          : 00000000a5a93f17
Model           : Raspberry Pi Zero W Rev 1.1
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0, libcamera interfaces=0

Videocore information
---------------------

Apr 17 2024 17:29:03
Copyright (c) 2012 Broadcom
version 86ccc427f35fdc604edc511881cdf579df945fb4 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
File system    1K-blocchi   Usati Disponib. Uso% Montato su
udev                82728       0     82728   0% /dev
tmpfs               43828     904     42924   3% /run
/dev/mmcblk0p2    3297068 1950504   1160868  63% /
tmpfs              219128       0    219128   0% /dev/shm
tmpfs                5120       0      5120   0% /run/lock
/dev/mmcblk0p1     522230  105282    416948  21% /boot/firmware
tmpfs               43824       0     43824   0% /run/user/1000

Filename                                Type            Size            Used   Priority
/var/swap                               file            102396          0      -2
`

**Logs**
Not having access to the log from ssh once it crashes I had to use rsyslog, managed to install it before it crashed and pointed it at my NAS server.
`Message from syslogd@Raspberry-Pi-1 at Apr 23 17:48:27 ...
 kernel:[   88.957230] Internal error: Oops: 37 [#2] ARM

Message from syslogd@Raspberry-Pi-1 at Apr 23 17:48:27 ...
 kernel:[   89.051760] Process dmesg (pid: 733, stack limit = 0x49454e90)

Message from syslogd@Raspberry-Pi-1 at Apr 23 17:48:27 ...
 kernel:[   89.054060] Stack: (0xdc9c5e80 to 0xdc9c6000)

Message from syslogd@Raspberry-Pi-1 at Apr 23 17:48:27 ...
 kernel:[   89.056369] 5e80: 00000000 dc9c5e84 dc9c5e84 c0d875f8 c0d875f8 c0d875e0 c0d875e0 c367b000

Message from syslogd@Raspberry-Pi-1 at Apr 23 17:48:27 ...
 kernel:[   89.060994] 5ea0: c0d7b030 c367b54c dc9c5f2c c0057fc0 c0d875f8 c0d875f8 c0d875e0 c0058d6c

[... 109 lines cut ...]

Raspberry-Pi-0 kern kernel [ 835.356059] df20: 00000000 1c435ebd c100d600 c113e000 c10c3bac c100d600 c0deda40 c100d61c

Raspberry-Pi-0 kern kernel [ 835.356096] df40: 61c88647 c10c372c dc85df5c c09b61bc c10c3b80 c10c3bac c100d658 c003d404

Raspberry-Pi-0 kern kernel [ 835.356132] df60: c10c3b80 c113e000 dc83ded0 c11aae80 c113e000 c003d214 c10c3b80 c10c1d40

Raspberry-Pi-0 kern kernel [ 835.356168] df80: dc83ded0 00000000 00000000 c004407c c11aae80 c0043fb0 00000000 00000000

Raspberry-Pi-0 kern kernel [ 835.356204] dfa0: 00000000 00000000 00000000 c000833c 00000000 00000000 00000000 00000000

Raspberry-Pi-0 kern kernel [ 835.356238] dfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000

Raspberry-Pi-0 kern kernel [ 835.356274] dfe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000

Raspberry-Pi-0 kern kernel [ 835.356312] __inet_lookup_listener from 0xc113e000

Raspberry-Pi-0 kern kernel [ 835.356403] Code: 0a00002d e59f3158 e3700a01 83a00000 (e5932000)

Raspberry-Pi-0 kern kernel [ 835.356437] ---[ end trace 0000000000000000 ]---

Raspberry-Pi-0 kern kernel [ 835.356461] note: kworker/u2:1[15] exited with irqs disabled`

**Additional context**
Add any other relevant context for the problem.

popcornmix, 2024-04-23:
What are you using as a power supply? Do you have any other option to test?

Totix92, 2024-04-23:
I'm using the official raspberry 5.1v 2.5A power supply.
I can try with another power supply.

popcornmix, 2024-04-23:
I don't think this is a software issue.
If Pi0W only ran for a few minutes for everyone then there would be many reports.

So, it's down to something unique to you. The options are:
Faulty Pi0W
Faulty power supply
Faulty sdcard

What happens if you don't connect to wifi. Is it still dead in a few minutes?

You could try adding `over_voltage=2` to config.txt. Does that make any difference?

Totix92, 2024-04-23:
This is what I get if I try another PSU and try running rpi-update:

curl: (23) Failure writing output to destination

Message from syslogd@Raspberry-Pi-1 at Apr 23 20:27:18 ...
 kernel:[  286.956475] Internal error: Oops: 37 [#1] ARM

Message from syslogd@Raspberry-Pi-1 at Apr 23 20:27:18 ...
 kernel:[  286.956475] Internal error: Oops: 37 [#1] ARM

Message from syslogd@Raspberry-Pi-1 at Apr 23 20:27:18 ...
 kernel:[  286.957750] Process tar (pid: 971, stack limit = 0x81fabb03)

Message from syslogd@Raspberry-Pi-1 at Apr 23 20:27:18 ...
 kernel:[  286.957784] Stack: (0xdcd8ddd0 to 0xdcd8e000)

Message from syslogd@Raspberry-Pi-1 at Apr 23 20:27:18 ...
 kernel:[  286.957813] ddc0:                                     c0d875e0 c07693a4 c18e3000 c0d875f8

Message from syslogd@Raspberry-Pi-1 at Apr 23 20:27:18 ...
 kernel:[  286.957850] dde0: c0d875f8 c0d875e0 c0d875e0 c1d60000 00000440 c1d60548 dcd8de7c c0057fc0
/usr/bin/rpi-update: riga 583:   969 Uscita 1                eval curl ${CURL_OPTIONS} "${FW_TARBALL_URI}"
       971 Errore di segmentazione | tar xzf - -C "${FW_REPOLOCAL}" --strip-components=1

Message from syslogd@Raspberry-Pi-1 at Apr 23 20:27:18 ...
 kernel:[  286.957750] Process tar (pid: 971, stack limit = 0x81fabb03)

Message from syslogd@Raspberry-Pi-1 at Apr 23 20:27:18 ...
 kernel:[  286.957784] Stack: (0xdcd8ddd0 to 0xdcd8e000)

Message from syslogd@Raspberry-Pi-1 at Apr 23 20:27:18 ...
 kernel:[  286.957813] ddc0:                                     c0d875e0 c07693a4 c18e3000 c0d875f8

Message from syslogd@Raspberry-Pi-1 at Apr 23 20:27:18 ...
 kernel:[  286.957886] de00: c0d875f8 c0d875f8 c0d875e0 c0058d6c c1d60000 c1d60000 c0d875e0 00000001

Message from syslogd@Raspberry-Pi-1 at Apr 23 20:27:18 ...
 kernel:[  286.957922] de20: c1d60000 00000440 c1d60548 c09b5908 cb153fb8 c1447c00 c1114d90 c0e13e9c

Message from syslogd@Raspberry-Pi-1 at Apr 23 20:27:18 ...
 kernel:[  286.957957] de40: c09b61bc 00000000 00000440 00002000 00000000 07290644 6627fd86 c1d60000

Message from syslogd@Raspberry-Pi-1 at Apr 23 20:27:18 ...
 kernel:[  286.957993] de60: c2436d2c 00000010 00000000 c1d60000 00000440 00002000 dcd8de8c c09b61bc

Message from syslogd@Raspberry-Pi-1 at Apr 23 20:27:18 ...
 kernel:[  286.958029] de80: c2436d20 c2436d2c 00000440 c023e6d8 c1d60000 00000000 2d5ad2ad c2436d70

Message from syslogd@Raspberry-Pi-1 at Apr 23 20:27:18 ...
 kernel:[  286.958065] dea0: c1a92640 00000000 dcd8df08 dcd8df28 00000001 c1d60000 c006013c c2436d2c

Message from syslogd@Raspberry-Pi-1 at Apr 23 20:27:18 ...
 kernel:[  286.958101] dec0: c2436d2c 07290644 00002600 00002000 c1a92640 00000000 00a89800 c1d60000

Message from syslogd@Raspberry-Pi-1 at Apr 23 20:27:18 ...
 kernel:[  286.958136] dee0: 00000001 00000000 00004004 c02346f8 00000000 c0234d9c 00000004 00000000

Message from syslogd@Raspberry-Pi-1 at Apr 23 20:27:18 ...
 kernel:[  286.958171] df00: 000a41ed 00000000 00000005 00000001 00000000 00a89800 00002000 00000000

Message from syslogd@Raspberry-Pi-1 at Apr 23 20:27:18 ...
 kernel:[  286.958207] df20: 00000001 00000000 c1a92640 00000000 00000000 00000000 00000000 00000000

Message from syslogd@Raspberry-Pi-1 at Apr 23 20:27:18 ...
 kernel:[  286.958242] df40: 00000000 00004004 00000000 00000000 00000000 07290644 00002000 c1a92640

Message from syslogd@Raspberry-Pi-1 at Apr 23 20:27:18 ...
 kernel:[  286.958277] df60: c1a92640 00a89800 00002000 c000843c c1d60000 00000000 000001ff c023506c

Message from syslogd@Raspberry-Pi-1 at Apr 23 20:27:18 ...
 kernel:[  286.958313] df80: 00002600 00000000 00000000 07290644 00079208 00000800 00002000 7ff00000

Message from syslogd@Raspberry-Pi-1 at Apr 23 20:27:18 ...
 kernel:[  286.958349] dfa0: 00000003 c0008260 00000800 00002000 00000003 00a89800 00002000 00000001

Message from syslogd@Raspberry-Pi-1 at Apr 23 20:27:18 ...
 kernel:[  286.958384] dfc0: 00000800 00002000 7ff00000 00000003 00000003 00079190 00000000 000001ff

Message from syslogd@Raspberry-Pi-1 at Apr 23 20:27:18 ...
 kernel:[  286.958420] dfe0: 00077078 bebb6128 00056e8c b6e5b5f0 20000010 00000003 00000000 00000000

Message from syslogd@Raspberry-Pi-1 at Apr 23 20:27:18 ...
 kernel:[  286.958458]  __inet_lookup_listener from 0xc1d60000

Message from syslogd@Raspberry-Pi-1 at Apr 23 20:27:18 ...
 kernel:[  286.958548] Code: 0a00002d e59f3158 e3700a01 83a00000 (e5932000)

After this the system crashes, the same SD running on a Pi 1B works, I doubt that's the problem.

Totix92, 2024-04-24:
I tried with over_voltage=2
Stable but it got too hot, so I tried with over_voltage=1 and so far it hasn't stopped, but how is this possible? I don't do any overclocking yet it requires overvoltage to be stable? Why?

pelwell, 2024-04-24:
Silicon variation. No matter how hard you try, no two chips will behave identically. Some need more, some less. Even worse, the variations affect different parts of the chips in different ways. There are mechanisms in place to work out the best voltage to use on the current board, but you do occasionally get outliers like yours that need just a bit more.

closed 2024-08-28 by popcornmix