raspberrypi/firmware #1897: Raspberry Pi 5 (8GB) Cannot startup: kernel or device tree not found

weiexcelpro, 2024-06-07:
Hi all,

I bought a Raspberry Pi 5 (8GB) a few days ago. The OS is RPI OS bookworm image. 

**Describe the bug**
I got the same problem that RPI cannot startup twice.  
BOOT ERROR: Code 7 - Kernel or device-tree not found or is not compatible. 
But what I install is the latest version and update all packages as the system notify. 

**To reproduce**
Not clear how to reproduce. 

**Expected behaviour**
Hope get a clear todo list to recover the system. 

[image]

pelwell, 2024-06-07:
> RPI cannot startup twice

What do you mean by this? Does it work most of the time? 

The errors suggest there is a problem with the SD card - have you tried a different card? Or rewriting the image on the same card?

weiexcelpro, 2024-06-07:
> > RPI cannot startup twice
> 
> What do you mean by this? Does it work most of the time?
> 
> The errors suggest there is a problem with the SD card - have you tried a different card? Or rewriting the image on the same card?

Yes, I tired to rewriting the image on the same card. It worked for a few hours and got the issue again.

pelwell, 2024-06-07:
I think you should try another card.

weiexcelpro, 2024-06-07:
> I think you should try another card.

many thanks for your reply. I will try.

MarjovanLier, 2024-06-26:
Interesting. My Samsung 980 512Gb did the same thing after months of working just fine.

I tried a USB caddy, as it was working fine on a [NVMe Base for Raspberry Pi 5 – NVMe](https://shop.pimoroni.com/products/nvme-base) till a few days ago.

I'm not sure whats up.

weiexcelpro, 2024-06-28:
I bought a new SD card (Sandisk 128 GB) and it works well so far. Thanks.

I tried a USB caddy, as it was working fine on a NVMe Base for Raspberry Pi 5 C NVMe<https://shop.pimoroni.com/products/nvme-base> till a few days ago.

I'm not sure whats up.

MarjovanLier, 2024-06-28:
I eventually found a way to fix this. (for whoever has to do this in future)

I installed Rasbian on an SD card and then put the SSD into a caddy.

Booted on the SD card, then inserted the USB SSD into the Pi.

### Step 1: Access and Mount Partitions
1. **Identify Partitions**: Use `lsblk` to list all devices and identify the partitions on the corrupted SD card (likely `/dev/sda1`, `/dev/sda2`).
   ```bash
   lsblk
   ```
2. **Mount Partitions**: Create mount points and mount the partitions.
   ```bash
   sudo fsck /dev/sda2
   sudo mkdir /mnt/external
   sudo mount /dev/sda2 /mnt/external
   sudo mount /dev/sda1 /mnt/external/boot
   ```

### Step 2: Chroot into the Mounted Filesystem
1. **Prepare Chroot Environment**:
   ```bash
   sudo mount --bind /dev /mnt/external/dev
   sudo mount --bind /proc /mnt/external/proc
   sudo mount --bind /sys /mnt/external/sys
   sudo cp /etc/resolv.conf /mnt/external/etc/
   ```
2. **Enter Chroot**:
   ```bash
   sudo chroot /mnt/external
   ```

### Step 3: Repair the Kernel
1. **Update Package Lists**:
   ```bash
   apt-get update
   ```
2. **Reinstall Kernel Packages**:
   ```bash
   apt-get install --reinstall raspberrypi-kernel raspberrypi-bootloader
   ```
3. **Update Firmware**:
   ```bash
   rpi-update
   ```

### Step 4: Exit Chroot and Unmount
1. **Exit Chroot**:
   ```bash
   exit
   ```
2. **Unmount Partitions**:
   ```bash
   sudo umount /mnt/external/boot
   sudo umount /mnt/external/dev
   sudo umount /mnt/external/proc
   sudo umount /mnt/external/sys
   sudo umount /mnt/external
   ```

### Step 5:

Power off and reboot with the SSD where it was, without the SD card.

closed 2024-06-28 by popcornmix