raspberrypi/firmware #1901: fatal firmware crash on attempt to play video on chromium-browser which leads to complete system freeze after some time

qrp73, 2024-06-10:
**Describe the bug**

attempt to play videos on chromium browser leads to fatal firmware crash. When it happens it also lead to complete system freeze after some runtime.

** Steps to reproduce**

1) Reboot the system
2) Open terminal and run ```dmesg -w```
3) Open chromium-browser and open page https://t.me/s/wildlifen
4) Play videos one-by-one from last to previous, etc

Expected result: videos are played with no firmware crash

Actual result: on third video (flying fox) you can notice some strange delays in playback. Forth video will have problems to play, but there is still no errors in the log. But on video #5 (Wild dogs are running predators) you can catch firmware crash. See log for details.

**System**

Hardware: Raspberry Pi 4 4GB (rev 1.5)

```
$ cat /etc/rpi-issue
Raspberry Pi reference 2023-09-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 40f37458ae7cadea1aec913ae10b5e7008ebce0a, stage4

$ vcgencmd version
May 24 2024 15:30:04 
Copyright (c) 2012 Broadcom
version 4942b7633c0ff1af1ee95a51a33b56a9dae47529 (clean) (release) (start)

$ uname -a
Linux raspi 6.6.32-v8+ #1 SMP PREEMPT Fri Jun  7 16:39:58 UTC 2024 aarch64 GNU/Linux
```

```
BOOTLOADER: up to date
   CURRENT: Fri 17 May 11:26:58 UTC 2024 (1715945218)
    LATEST: Fri 17 May 11:26:58 UTC 2024 (1715945218)
   RELEASE: latest (/lib/firmware/raspberrypi/bootloader-2711/latest)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138c0
    LATEST: 000138c0
```

**Logs**
dmesg
```
[  320.481630] bcm2835_mmal_vchiq: timed out waiting for sync completion
[  320.481649] bcm2835-codec bcm2835-codec: bcm2835_codec_stop_streaming: Failed disabling i/p port, ret -62
[  322.497703] bcm2835-codec bcm2835-codec: bcm2835_codec_flush_buffers: Timeout waiting for buffers to be returned - 8 outstanding
[  322.497787] ------------[ cut here ]------------
[  322.497790] WARNING: CPU: 1 PID: 3340 at drivers/media/common/videobuf2/videobuf2-core.c:2024 __vb2_queue_cancel+0x220/0x2a0 [videobuf2_common]
[  322.497818] Modules linked in: ntfs3 rfcomm snd_seq_dummy snd_hrtimer snd_seq cmac algif_hash aes_arm64 aes_generic algif_skcipher af_alg bnep brcmfmac_wcc brcmfmac v3d gpu_sched binfmt_misc drm_shmem_helper brcmutil joydev vc4 snd_usb_audio hci_uart cfg80211 btbcm snd_hwdep bluetooth snd_usbmidi_lib rpivid_hevc(C) snd_soc_hdmi_codec drm_display_helper bcm2835_codec(C) bcm2835_isp(C) bcm2835_v4l2(C) cec raspberrypi_hwmon v4l2_mem2mem drm_dma_helper bcm2835_mmal_vchiq(C) sg drm_kms_helper snd_rawmidi videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops snd_soc_core snd_seq_device i2c_brcmstb videobuf2_v4l2 ecdh_generic ecc snd_bcm2835(C) snd_compress i2c_bcm2835 snd_pcm_dmaengine rfkill snd_pcm videodev vc_sm_cma(C) libaes snd_timer videobuf2_common raspberrypi_gpiomem snd mc nvmem_rmem uio_pdrv_genirq uio i2c_dev drm fuse drm_panel_orientation_quirks dm_mod backlight ip_tables x_tables ipv6
[  322.497940] CPU: 1 PID: 3340 Comm: ThreadPoolSingl Tainted: G         C         6.6.32-v8+ #1
[  322.497945] Hardware name: Raspberry Pi 4 Model B Rev 1.5 (DT)
[  322.497948] pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  322.497952] pc : __vb2_queue_cancel+0x220/0x2a0 [videobuf2_common]
[  322.497965] lr : __vb2_queue_cancel+0x38/0x2a0 [videobuf2_common]
[  322.497977] sp : ffffffc082f23ae0
[  322.497979] x29: ffffffc082f23ae0 x28: ffffffe8ceb4ce18 x27: ffffff80b23e4000
[  322.497987] x26: ffffff8041e2e530 x25: 0000000000000000 x24: 0000000040045613
[  322.497994] x23: 0000000000000000 x22: ffffff80b23c3298 x21: ffffff80b23c3298
[  322.498000] x20: 000000000000000a x19: ffffff80b23c3298 x18: ffffffffffffffff
[  322.498006] x17: 756f656d6954203a x16: ffffffe941ed4350 x15: 6873756c665f6365
[  322.498013] x14: 0000000000000002 x13: ffffff8042e00028 x12: 0000000000000000
[  322.498019] x11: ffffff80a89cf4b0 x10: ffffff80a89cf478 x9 : ffffffe941f2b1a0
[  322.498026] x8 : ffffff80a89cf4a0 x7 : 0000000000000001 x6 : fffffffffffffff8
[  322.498032] x5 : 0000000041001000 x4 : 000000000015000b x3 : ffffff80c292ea80
[... 85 lines cut ...]
[  346.082112] bcm2835-codec bcm2835-codec: bcm2835_codec_create_component: failed to create component ril.video_decode
[  349.154079] bcm2835_mmal_vchiq: timed out waiting for sync completion
[  349.154096] bcm2835_mmal_vchiq: vchiq_mmal_component_init: failed to create component -62 (Not enough GPU mem?)
[  349.154102] bcm2835-codec bcm2835-codec: bcm2835_codec_create_component: failed to create component ril.video_decode
[  834.979375] swiotlb_tbl_map_single: 2 callbacks suppressed
[  834.979388] vc4-drm gpu: swiotlb buffer is full (sz: 794624 bytes), total 32768 (slots), used 0 (slots)
[  835.397248] vc4-drm gpu: swiotlb buffer is full (sz: 794624 bytes), total 32768 (slots), used 0 (slots)
```

vclog -m contains thousands of the same errors in a row, here is short cut:
```
1152287.687: alloc_compact_internal(space=3291328, in_low_region=0, mode=7) failed
1152287.719: mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)
1152287.897: alloc_compact_internal(space=3291328, in_low_region=0, mode=7) failed
1152287.927: mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)
1152298.706: alloc_compact_internal(space=3291328, in_low_region=0, mode=7) failed
1152298.751: mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)
1152307.720: alloc_compact_internal(space=3291328, in_low_region=0, mode=7) failed
1152307.754: mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)
1152307.947: alloc_compact_internal(space=3291328, in_low_region=0, mode=7) failed
1152307.981: mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)
```

**Additional context**
The issue is reproducible.

pelwell, 2024-06-10:
Is this replacing https://github.com/raspberrypi/linux/issues/6170? You seem to have found a page (or class of pages) that causes Chromium to exhaust the firmware's resources, and are now mining a rich seam of similar but non-identical problems.

qrp73, 2024-06-10:
I'm not sure, but at a glance yes - it seems that this issue leads to issue https://github.com/raspberrypi/linux/issues/6170. I discovered it because I'm tired system freeze issue which started to happens very often since kernel 6.6.28 or something like that.

Note, that when this firmware crash happens for a first time it don't leads to a system freeze, like it described in #6170, but I believe that this firmware crash is a root of cause for system freeze which happens later when this firmware crash happens (maybe several times). Also after this firmware crash something is going broken in firmware, because it leads to problems with some video playback.

qrp73, 2024-06-11:
looks like today update for ```chromium-codecs-ffmpeg-extra:arm64 (124.0.6367.73-rpt1, 125.0.6422.133-rpt1), chromium-browser:arm64 (124.0.6367.73-rpt1, 125.0.6422.133-rpt1), chromium-browser-l10n:arm64 (124.0.6367.73-rpt1, 125.0.6422.133-rpt1)``` solved the issue. 

At least with original steps to reproduce I can play videos with no crash.

qrp73, 2024-06-15:
After some runtime, I confirm that system crash issue due to playing video on browser disappears.

Now there is no crash, but sometimes this error appears in the log, but now it happens rare and don't lead to system crash:
```
[14696.596892] vc4-drm gpu: swiotlb buffer is full (sz: 491520 bytes), total 32768 (slots), used 418 (slots)
[14696.606252] vc4-drm gpu: swiotlb buffer is full (sz: 524288 bytes), total 32768 (slots), used 934 (slots)
[14698.303095] vc4-drm gpu: swiotlb buffer is full (sz: 491520 bytes), total 32768 (slots), used 418 (slots)
[14698.908934] vc4-drm gpu: swiotlb buffer is full (sz: 524288 bytes), total 32768 (slots), used 934 (slots)
[14708.127736] v3d fec00000.v3d: MMU error from client CLE (4) at 0x5201000, pte invalid
[14747.703863] vc4-drm gpu: swiotlb buffer is full (sz: 524288 bytes), total 32768 (slots), used 552 (slots)
[14747.748102] vc4-drm gpu: swiotlb buffer is full (sz: 524288 bytes), total 32768 (slots), used 552 (slots)
[31434.235549] ------------[ cut here ]------------
[31434.235570] WARNING: CPU: 0 PID: 40051 at drivers/media/common/videobuf2/videobuf2-core.c:2024 __vb2_queue_cancel+0x220/0x2a0 [videobuf2_common]
[31434.235633] Modules linked in: rfcomm snd_seq_dummy snd_hrtimer snd_seq cmac algif_hash aes_arm64 aes_generic algif_skcipher af_alg bnep brcmfmac_wcc vc4 brcmfmac snd_soc_hdmi_codec joydev drm_display_helper snd_usb_audio cec binfmt_misc snd_hwdep drm_dma_helper bcm2835_codec(C) snd_usbmidi_lib drm_kms_helper rpivid_hevc(C) snd_rawmidi snd_seq_device bcm2835_isp(C) hci_uart brcmutil v4l2_mem2mem snd_soc_core btbcm cfg80211 bluetooth v3d bcm2835_v4l2(C) gpu_sched bcm2835_mmal_vchiq(C) videobuf2_vmalloc ecdh_generic snd_compress videobuf2_dma_contig ecc drm_shmem_helper videobuf2_memops snd_bcm2835(C) snd_pcm_dmaengine rfkill videobuf2_v4l2 snd_pcm videodev raspberrypi_hwmon libaes snd_timer i2c_bcm2835 i2c_brcmstb vc_sm_cma(C) videobuf2_common snd raspberrypi_gpiomem mc nvmem_rmem uio_pdrv_genirq uio drm i2c_dev fuse dm_mod drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
[31434.235966] CPU: 0 PID: 40051 Comm: ThreadPoolSingl Tainted: G         C         6.6.32-v8+ #1
[31434.235979] Hardware name: Raspberry Pi 4 Model B Rev 1.5 (DT)
[31434.235986] pstate: 80000005 (Nzcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[31434.235999] pc : __vb2_queue_cancel+0x220/0x2a0 [videobuf2_common]
[31434.236038] lr : __vb2_queue_cancel+0x38/0x2a0 [videobuf2_common]
[31434.236074] sp : ffffffc081dc3ae0
[31434.236080] x29: ffffffc081dc3ae0 x28: ffffffd729731e18 x27: ffffff8042158000
[31434.236102] x26: ffffff804264ad30 x25: 0000000000000000 x24: 0000000040045613
[31434.236122] x23: 0000000000000000 x22: ffffff809a4ac028 x21: ffffff809a4ac028
[31434.236141] x20: 0000000000000009 x19: ffffff809a4ac028 x18: ffffffffffffffff
[31434.236160] x17: 0000000000000005 x16: ffffffd7944f8df8 x15: ffffffc101dc3787
[31434.236178] x14: 000000000000000c x13: ffffff8043b80028 x12: 0000000000000000
[31434.236197] x11: ffffff808b44b990 x10: ffffff808b44b908 x9 : ffffffd7297909f4
[31434.236216] x8 : ffffff808b44b930 x7 : 0000000000000001 x6 : ffffc00000000000
[31434.236234] x5 : ffffff804264b01c x4 : 0000000000000000 x3 : 0000000000000000
[31434.236252] x2 : ffffff80eb888000 x1 : 0000000000000000 x0 : 0000000000000002
[31434.236271] Call trace:
[31434.236278]  __vb2_queue_cancel+0x220/0x2a0 [videobuf2_common]
[31434.236315]  vb2_core_streamoff+0x2c/0xe0 [videobuf2_common]
[31434.236350]  vb2_streamoff+0x24/0x80 [videobuf2_v4l2]
[31434.236383]  v4l2_m2m_streamoff+0x80/0x160 [v4l2_mem2mem]
[31434.236419]  v4l2_m2m_ioctl_streamoff+0x20/0x38 [v4l2_mem2mem]
[31434.236452]  v4l_streamoff+0x2c/0x40 [videodev]
[31434.236552]  __video_do_ioctl+0x194/0x400 [videodev]
[31434.236636]  video_usercopy+0x1e4/0x780 [videodev]
[31434.236719]  video_ioctl2+0x20/0x40 [videodev]
[31434.236801]  v4l2_ioctl+0x48/0x70 [videodev]
[31434.236883]  __arm64_sys_ioctl+0xb0/0x100
[31434.236906]  invoke_syscall+0x50/0x128
[31434.236922]  el0_svc_common.constprop.0+0x48/0xf8
[31434.236937]  do_el0_svc+0x28/0x40
[31434.236950]  el0_svc+0x40/0xf8
[31434.236967]  el0t_64_sync_handler+0x13c/0x158
[31434.236976]  el0t_64_sync+0x190/0x198
[31434.236987] ---[ end trace 0000000000000000 ]---
[31434.236998] videobuf2_common: driver bug: stop_streaming operation is leaving buf 000000000b086321 in active state
[31434.237009] videobuf2_common: driver bug: stop_streaming operation is leaving buf 000000009f9af9a2 in active state
[31434.237279] ------------[ cut here ]------------
[31434.237293] WARNING: CPU: 0 PID: 57920 at drivers/media/common/videobuf2/videobuf2-core.c:1049 vb2_buffer_done+0x29c/0x2c0 [videobuf2_common]
[31434.237348] Modules linked in: rfcomm snd_seq_dummy snd_hrtimer snd_seq cmac algif_hash aes_arm64 aes_generic algif_skcipher af_alg bnep brcmfmac_wcc vc4 brcmfmac snd_soc_hdmi_codec joydev drm_display_helper snd_usb_audio cec binfmt_misc snd_hwdep drm_dma_helper bcm2835_codec(C) snd_usbmidi_lib drm_kms_helper rpivid_hevc(C) snd_rawmidi snd_seq_device bcm2835_isp(C) hci_uart brcmutil v4l2_mem2mem snd_soc_core btbcm cfg80211 bluetooth v3d bcm2835_v4l2(C) gpu_sched bcm2835_mmal_vchiq(C) videobuf2_vmalloc ecdh_generic snd_compress videobuf2_dma_contig ecc drm_shmem_helper videobuf2_memops snd_bcm2835(C) snd_pcm_dmaengine rfkill videobuf2_v4l2 snd_pcm videodev raspberrypi_hwmon libaes snd_timer i2c_bcm2835 i2c_brcmstb vc_sm_cma(C) videobuf2_common snd raspberrypi_gpiomem mc nvmem_rmem uio_pdrv_genirq uio  [... cut ...]
[49387.693334]  video_usercopy+0x1e4/0x780 [videodev]
[49387.693424]  video_ioctl2+0x20/0x40 [videodev]
[49387.693507]  v4l2_ioctl+0x48/0x70 [videodev]
[49387.693595]  __arm64_sys_ioctl+0xb0/0x100
[49387.693623]  invoke_syscall+0x50/0x128
[49387.693639]  el0_svc_common.constprop.0+0x48/0xf8
[49387.693655]  do_el0_svc+0x28/0x40
[49387.693669]  el0_svc+0x40/0xf8
[49387.693687]  el0t_64_sync_handler+0x13c/0x158
[49387.693696]  el0t_64_sync+0x190/0x198
[49387.693708] ---[ end trace 0000000000000000 ]---
```

qrp73, 2024-06-20:
it looks like issue is fixed with updates, running for several days with no issue. So, I close the issue

closed 2024-06-20 by qrp73