raspberrypi/firmware #1902: UE200 USB to Ethernet Adapter not initializing in boot

Eldho1416, 2024-06-10:
**Describe the bug**
I have USB to Lan adapter from TP link  [https://tinyurl.com/kptfvtrp](link) which used to work perfectly fine around 6-8 months ago in Ubuntu Server 20.04 running Raspberry PI 4B (4GB RAM) and I recently tried to connect it to same hardware and OS and it fails to show the interface. It doesn't show up in `lsusb` but if unplug and plug it it works fine what could be the problem?

**To reproduce**
Connect UE200 from TPlink before boot and then boot

**Expected behaviour**
Supposed to be initailzed during boot

**Actual behaviour**
Not initializing during boot

**System**

* RPi 4B (4GB)
* Tried on both Ubuntu 20.04 and Raspbian OS lite (64Bit)

**Logs**
`dmesg`
```
[    2.535403] usb 1-1.3: new full-speed USB device number 5 using xhci_hcd
[    2.615603] usb 1-1.3: device descriptor read/64, error -32
[    2.807586] usb 1-1.3: device descriptor read/64, error -32
[    2.999391] usb 1-1.3: new full-speed USB device number 6 using xhci_hcd
[    3.079588] usb 1-1.3: device descriptor read/64, error -32
[    3.271585] usb 1-1.3: device descriptor read/64, error -32
[    3.991400] usb 1-1.3: new full-speed USB device number 7 using xhci_hcd
[    3.991629] usb 1-1.3: Device not responding to setup address.
[    4.227594] usb 1-1.3: Device not responding to setup address.
[    4.435387] usb 1-1.3: device not accepting address 7, error -71
[    4.519396] usb 1-1.3: new full-speed USB device number 8 using xhci_hcd
[    4.519629] usb 1-1.3: Device not responding to setup address.
[    4.727670] usb 1-1.3: Device not responding to setup address.
[    4.935901] usb 1-1.3: device not accepting address 8, error -71
```

`lsusb` (While connected during boot)
```
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
**Additional context**
I tried this in Ubuntu 20.04, 22.04 and Raspbian Lite (64Bit) and also downgraded the kernel version but the issue still persists while it is running perfectly in the OS i installed during Nov/Dec 2023

Eldho1416, 2024-06-11:
Update: I tried connecting with Jetson AGX and it is initializing with `tegra-xusb`

```
05:10:21 nvidia-desktop kernel: usb 1-2: new high-speed USB device number 2 using tegra-xusb
05:10:21 nvidia-desktop kernel: usb 1-2: New USB device found, idVendor=2357, idProduct=0602, bcdDevice=20.00
05:10:21 nvidia-desktop kernel: usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
05:10:21 nvidia-desktop kernel: usb 1-2: Product: USB 10/100 LAN
05:10:21 nvidia-desktop kernel: usb 1-2: Manufacturer: TP-LINK
05:10:21 nvidia-desktop kernel: usb 1-2: SerialNumber: 3C52A1B7DA8C
05:10:21 nvidia-desktop kernel: cdc_ether 1-2:2.0 eth0: register 'cdc_ether' at usb-3610000.xhci-2, CDC Ethernet Device, 3c:52:a1:b7:da:8c
```

Does this mean `xhci_hcd` is not supporting the Ethernet module?

pelwell, 2024-06-11:
> Does this mean xhci_hcd is not supporting the Ethernet module?

No, I don't think that's the case.

>  if unplug and plug it it works fine 

That sounds like some sort of power-on glitch - it could be due to timing of reset signals, or perhaps the system is running out of power.
* Which power supply and cable are you using?
* Do any of those devices have independent power supplies, or is the Pi supply having to power them all?

Eldho1416, 2024-06-12:
Hi @pelwell,

I'm using the official rasberry pi adapter with rating 5.1V, 3A [Link](https://robu.in/product/official-usb-type-c-15-3w-power-supply-for-raspberry-pi-4/?gad_source=1&gclid=CjwKCAjw65-zBhBkEiwAjrqRMMKmrDIyov9Nut1dQvzuJ749v1mnPV--AnmjJEt04SMNQbYGn1RJ3RoCABQQAvD_BwE).

I had the same doubt regrading the power issues so i disconnected all the other connection from rasberry pi but that didn't solve the problem.

pelwell, 2024-06-14:
Do you have a USB hub? If so, try putting it between the LAN adaptor and the Pi 4 - sometimes that can help with devices that are not particularly standards-compliant, and it would be interesting to know if this issue might belong to that class of problems.

Eldho1416, 2024-06-15:
Sure i'll try with USB hub and let you know the results

Eldho1416, 2024-06-18:
Hello @pelwell,

I tried with [this](https://zebronics.com/products/zeb-100hb) USB hub without and with external power of 5V/3A still the issue persists.

Eldho1416, 2024-07-04:
Hi @pelwell,

I'm still facing same issue with new set of raspberry pi, is there any way to mock uplugging and plug again since i cannot physically do it all the time or any was to intialize USB after booting?

pelwell, 2024-07-04:
Have you tried `uhubctl`?
```
$ sudo apt install uhubctl
$ lsusb -t
# Note the bus number for the device you want to power-cycle - let's call it <bus>
$ sudo uhubctl -a cycle -l <bus>
```

pelwell, 2024-07-04:
I should make it clear that Pi 4B doesn't have per-port VBUS control. However, uhubctl will issue USB resets to individual ports, and you may find that is sufficient.

Eldho1416, 2024-07-04:
Thanks @pelwell,

output of `lsusb -t` without when driver not initialized is 
```
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=dwc2/1p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
```

so i tried `sudo uhubctl -a cycle -l 1` but that didn't help.

I tried again without specifying  bus number `sudo uhubctl -a cycle` and not the driver is working fine.

```
lsusb -t
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=dwc2/1p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 8, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 2: Dev 9, If 0, Class=Communications, Driver=cdc_ether, 480M
        |__ Port 2: Dev 9, If 1, Class=CDC Data, Driver=cdc_ether, 480M
```

Eldho1416, 2024-07-04:
uhh nvm, It doesn't work when other USB devices are connected

anttiryt, 2024-07-29:
There seems another work on the same problem at rpi-eeprom

https://github.com/raspberrypi/rpi-eeprom/issues/472

OP has some success with  NET_INSTALL_ENABLED=0 in config.txt but my bookworm config.txt doesn't look like a place to put  NET_INSTALL_ENABLED=0 in it. Any advice?

```
...
# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

```

timg236, 2024-07-29:
The bootloader doesn't do anything with USB Ethernet devices. At best NET_INSTALL_ENABLED=0 stops the bootloader enumerating USB once which might help with a flaky USB device but that's essentially luck rather than defined behavior.

hstr0100, 2025-01-15:
I am experiencing the same issue on the Pi Zero hardware.

```
[    8.394713] Indeed it is in host mode hprt0 = 00001101
[    8.553554] fuse: init (API version 7.37)
[    8.890795] usb 1-1: device descriptor read/64, error -110
[    9.011154] Indeed it is in host mode hprt0 = 00001101
[    9.520813] usb 1-1: device descriptor read/64, error -71
[    9.640944] Indeed it is in host mode hprt0 = 00001101
[   10.110922] usb 1-1: new high-speed USB device number 3 using dwc_otg
[   10.121017] Indeed it is in host mode hprt0 = 00001101
[   10.570929] usb 1-1: device descriptor read/64, error -71
[   10.691123] Indeed it is in host mode hprt0 = 00001101
[   11.260871] usb 1-1: device descriptor read/64, error -71
[   11.380953] usb usb1-port1: attempt power cycle
```

This issue appears to be exclusive to newer kernel releases and affects at least Ubuntu, RaspberryPI OS, and DietPi.

Running a Buster build from late 2021 completely mitigates the problem, and the adapter works perfectly. However, as soon as an `apt-get dist-upgrade` is performed, the adapter becomes unreliable and only occasionally works at boot. The adapter also functions perfectly with motionEyeOS, which is based on an older Buildroot release.

In my case, the UE200 adapter is externally powered, and the power supply gives the PI plenty of headroom as well, so the issue is not related to power.