raspberrypi/firmware #1905: Questions: About power domain firmware tags

lategoodbye, 2024-06-17:
Hi,
recently i took a closer look at the firmware driven power domain driver ( drivers/pmdomain/bcm/raspberrypi-power.c ), which raised some questions:

Is it intended that RPI_FIRMWARE_SET_DOMAIN_STATE is not documented in the Wiki?
Is RPI_FIRMWARE_SET_POWER_STATE tag behavior (e.g. state bits) identical to RPI_FIRMWARE_SET_DOMAIN_STATE except of the used device ids?
Is the wait bit still relevant? From my observations RPI_FIRMWARE_SET_POWER_STATE blocks even without the wait bit set.
Does the "device doesn't exist" bit really work as documented, because it's at the same place like the wait bit for the request?

Thanks

pelwell, 2024-06-18:
> Is it intended that RPI_FIRMWARE_SET_DOMAIN_STATE is not documented in the Wiki?

Not really, it just hasn't been done - AFAIK you're the first to ask.

> Is RPI_FIRMWARE_SET_POWER_STATE tag behavior (e.g. state bits) identical to RPI_FIRMWARE_SET_DOMAIN_STATE except of the used device ids?

I wouldn't say they are identical. The older SET_POWER_STATE interfaces mostly uses the same underlying firmware API, but operates at a slightly higher level and allows for additional, per-device configuration. @popcornmix may be able to shed some more light on this.

> Is the wait bit still relevant? From my observations RPI_FIRMWARE_SET_POWER_STATE blocks even without the wait bit set.

I don't see any support for the wait bit - if anything it looks like it would make an "off-with-wait" look like an "on".

> Does the "device doesn't exist" bit really work as documented, because it's at the same place like the wait bit for the request?

Yes - if the command is received correctly but with an invalid ID, the response is `1<<1`.

lategoodbye, 2024-06-18:
So the following part of the Mailbox documentation contains errors?

#### Set power state
 * Tag: 0x00028001
 * Request:
   * Length: 8
   * Value:
     * u32: device id
     * u32: state
   * State:
     * Bit 0: 0=off, 1=on
     * Bit 1: 0=do not wait, 1=wait
     * Bits 2-31: reserved for future use (set to 0)

Response indicates new state, with/without waiting for the power to become stable.

popcornmix, 2024-06-18:
Yes, the calls inside the firmware are all blocking, so there is no way to "not wait".
That means bit 1 is not considered.

I think technically we follow the API described, but the API description isn't optimal for the implementation.

pelwell, 2024-06-18:
I think "wait + off" is treated as an "on".

lategoodbye, 2024-06-18:
Thanks for the clarification, so i won't start to implement the wait bit for the raspberry-power driver.

popcornmix, 2024-06-18:
Yes. The next firmware update will stop "wait + off" being interpreted as "on", but with the interface we have, I don't believe setting wait will ever do anything useful, so best just to ignore it.

lategoodbye, 2024-06-18:
Great, i don't like the wait bit at all because it's at the same place as the "id don't exist" bit in the response.

closed 2024-06-19 by lategoodbye