raspberrypi/firmware #1906: gpio-poweroff for RPi 0W, bullseye, 32-bit

seamusdemora, 2024-06-23:
I've got a RPi 0W I'm trying to press into service for a "garden irrigation" project. As it's in the garden, it's also battery/solar powered. I have a "fixture" I've used for a couple of years that controls power application to the RPi.  This "fixture" uses a Realtime clock (RTC), and a ["latching relay"](https://en.wikipedia.org/wiki/Relay#Latching_relay) to control power to the RPi. I use the RTC's `#INT` output to CLOSE the latching relay contacts and supply Power, and `gpio-poweroff` to OPEN the latching relay contacts and remove Power. This fixture has always worked well, it's fairly simple, and I've checked and double-checked it for this project. 

I have the following entries in `boot/config.txt`: 
```
# ?????????????????? : 
# dtoverlay=dwc-otg

# seamus: set up RTC on i2c0
dtparam=i2c_vc=on
dtoverlay=i2c-rtc,ds3231,i2c0,wakeup-source
dtoverlay=gpio-poweroff 
# dtoverlay=gpio-poweroff,active_delay_ms=250,inactive_delay_ms=500,timeout_ms=8000
# dtoverlay=gpio-poweroff,gpiopin=17

```

The problem I seem to be having is that the `gpio-poweroff` overlay is not _"firing"_ the GPIO pin! I've tried it on the default pin (GPIO 26), and on GPIO 17, but get nothing from either. I've got an Agilent oscilloscope, and have tried to capture the transition; I DO NOT SEE A TRANSITION on GPIO 26 (or on GPIO 17). 

I researched the problem: The only **relevant** item I could find was [this post in the forums](https://forums.raspberrypi.com/viewtopic.php?t=292633). The OP appeared to have researched it well - he actually came up with the solution that worked for him.  Unfortunately, it **did not work** for me. I tried powering in the PWR (first) and the OTG (second) connectors, with and without the `dwc-otg` overlay in both cases - ***without success***. 

And so I'm posting here as a last resort. Any ideas or solutions for this?? 

P.S.: I would have tried posting in the forums, but was "banned for life" several years ago (~2016). It was an innocuous post, but now deleted by the admins apparently so I can't even reference it any longer.

pelwell, 2024-06-23:
Which kernel version is this?

seamusdemora, 2024-06-23:
Sorry - I don't know, but It was kept updated (to oldstable). I have just fried the RPi 0W while troubleshooting... accidentally connected the 12V battery to the RPi's power input  :)   I'm not giving up, just waiting for delivery of a new RPi 0W.

seamusdemora, 2024-06-23:
Your question re kernel version piques my curiosity... is there a kernel version(s) that is known to not support `gpio-poweroff`?

seamusdemora, 2024-06-23:
If it helps any, I got this from the SD card I removed from the (now fried) Rpi 0W; no version #'s but do dates help any? 

```
-rwxr-xr-x 1 root root 6810592 Apr 16  2023 kernel7.img
-rwxr-xr-x 1 root root 7225232 Apr 16  2023 kernel7l.img
-rwxr-xr-x 1 root root 8219600 Apr 16  2023 kernel8.img
-rwxr-xr-x 1 root root 6466528 Apr 16  2023 kernel.img
```

seamusdemora, 2024-06-23:
6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023

pelwell, 2024-06-25:
gpio-poweroff works just fine on `6.1.21+ #1642  Mon Apr 3 17:19:14 BST 2023`. What doesn't work is shutting down with the desktop enabled. If you switch to "boot to console" (in raspi-config) you'll find that all three gpio-poweroff configurations in your config.txt work as expected.

I suggest upgrading to a 6.6 kernel, or booting to console.

pelwell, 2024-06-25:
If you wait long enough, something times out and the shutdown will complete, wiggling the poweroff GPIO as it exits.

seamusdemora, 2024-06-26:
> gpio-poweroff works just fine on `6.1.21+ #1642 Mon Apr 3 17:19:14 BST 2023`. What doesn't work is shutting down with the desktop enabled. If you switch to "boot to console" (in raspi-config) you'll [...]

Desktop??  I'm using the "Lite" version of the OS. What am I missing? 

I've only ever used the "Lite" version of the OS on virtually every hardware model made, and I've never had a problem with the `gpio-poweroff` overlay. This **is** the first time I've tried it on my (recently departed) Pi 0W, and I concluded that the issue I'm seeing must be related to this model. 

Also - I connected my o'scope (first to GPIO 26, later to GPIO 17) on the Zero W, but it was *never triggered* - even once when I left the scope connected for maybe an hour after the `gpio-poweroff`. I know what it looks like... below is a screenshot I made some time ago on a 4B :  
[image] 

Anyway - I appreciate your suggestion, but remain confused & without a clear path ahead. I assume my use of the "Lite" version of the OS rules out your explanation of the "desktop" being the cause??

pelwell, 2024-06-26:
It's working fine for me on a Zero W running the current Lite image downgraded to 6.1.21. You should try the same (`sudo rpi-update 0b15297` to install that specific, nearly-15-month-old kernel).

If you want to debug your problem, add `ignore_level` to cmdline.txt and see what appears on the display or UART. If you have neither display nor UART, you're going to find it difficult.

seamusdemora, 2024-06-27:
My Zero W is scheduled for delivery on Monday, July 1. I was planning on using the SD card from the dead Zero W ('bullseye' w/ 6.1.21)... would that not be functionally equivalent to a fresh install of 'bookworm' & downgrading the kernel to 6.1.21?  Be happy to do whatever you suggest - just trying to save steps where it makes sense. 

wrt `ignore_level`: 
I can connect either a display & kbd -OR- wire up the UART to another RPi.  It may be easier for me to connect the display & kbd, add `video=HDMI-A-1:1920x1080M@60` to `cmdline.txt` ... if there's no other diffs of which I'm unaware.

pelwell, 2024-06-27:
I got the option name wrong - it should be `ignore_loglevel` (corrected above).

seamusdemora, 2024-06-27:
Is there any sort of documentation on that option: `ignore_loglevel`?

seamusdemora, 2024-06-29:
My replacement RPi 0W arrived a day early!  :0   

I've re-installed the SD card (from the now-departed RPi 0W), and re-ran my test. The `gpio-poweroff` overlay works perfectly! I don't know what caused the malfunction on the first RPi 0W, and I won't be learning anything further as it's been consigned to the bin. 

Thank you again for your help, and my sincere apologies for the false alarm.

closed 2024-06-30 by pelwell