raspberrypi/firmware #1907: btuart fails on Raspberry Pi 5 due to UART changes (alpha8)

a10kiloham, 2024-07-02:
**Describe the bug**
btuart service fails consistently with error about rpi3 being found

**To reproduce**
systemctl restart btuart

**System**

In /opt/btspeaker/btuart.sh the line ``` /proc/device-tree/soc/gpio@7e200000/uart0_pins/brcm\,pins``` is no longer valid as there is no folder of that name in the soc directory nor is there a uart0_pins folder in any of the two new directories gpio@7d508500 and gpio@7d517c00

Similarly in the same file the stty line does not work as there is no ttyAMA0 anymore. I believe the hci attach isn't needed even anymore. For RPI5, the correct line to attach is below which works.
```
/usr/bin/btattach -B /dev/ttyAMA10 -P bcm -S 115200 -N
```
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

Kernel is: Linux hifiberry 6.6.31-v8 #1 SMP PREEMPT Thu May 23 07:43:46 UTC 2024 aarch64 GNU/Linux

timg236, 2024-07-02:
This issue should be raised against the Github repo for btuart.sh script  or possibly Linux if a DT alias is required to resolve the BT UART name.

pelwell, 2024-07-02:
/opt/btspeaker/btuart.sh is not one of ours...

timg236, 2024-07-02:
Ah, ok. In that case the bug should be raised against whatever owns btuart.sh to update accordingly

The device-tree paths here are correct.

pelwell, 2024-07-02:
In RPiOS the `hciuart` systemd service runs `/usr/bin/btuart`, but only if `/proc/device-tree/soc/gpio@7e200000/bt_pins/brcm,pins` is non-empty. On Pi 5 it is not non-empty because it is absent, so we rely on the kernel having initialised the BT UART as an HCI interface.

Having Pi 5 Bluetooth initialised by `hciattach` (`dtparam=krnbt=off`) is unsupported.

closed 2025-02-14 by popcornmix