raspberrypi/firmware #1909: After firmware update i couldn't boot up my rpi5 with usb hard drive

thidasi, 2024-07-12:
**Describe the bug**
I did an rpi-update to 6.6.39-v8-16k+. After that i couldn't boot up my raspberry pi 5. I figured out if i disconnect one of my usb hard drive, it solve the problem. With the previous firmware it worked fine.

[image]

**To reproduce**
1. connect an old usb hard drive?
2. Update firmware with rpi-update to 6.6.39-v8-16k+

**Expected behaviour**
To use my existing usb hard drive.

**Actual behaviour**
Without the usb hard drive it works fine.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

System Information
------------------

Raspberry Pi 5 Model B Rev 1.0
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"

Raspberry Pi reference 2024-03-15
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, f19ee211ddafcae300827f953d143de92a5c6624, stage2

Linux hidasipi 6.6.39-v8-16k+ #1783 SMP PREEMPT Thu Jul 11 18:58:24 BST 2024 aarch64 GNU/Linux
Revision        : d04170
Serial          : 6de10619e9da575e
Model           : Raspberry Pi 5 Model B Rev 1.0
Throttled flag  : throttled=0x0
Camera          : vc_gencmd_read_response returned -1 error=1 error_msg="Command not registered"

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Raspberry Pi 5
* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2024-03-15
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, f19ee211ddafcae300827f953d143de92a5c6624, stage2
* Which firmware version (`vcgencmd version`)?
2024/06/05 16:41:49
Copyright (c) 2012 Broadcom
version 6fe0b091 (release) (embedded)
* Which kernel version (`uname -a`)?
Linux hidasipi 6.6.39-v8-16k+ #1783 SMP PREEMPT Thu Jul 11 18:58:24 BST 2024 aarch64 GNU/Linux

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
I couldn't make a firmware downgrade. I got a commit hash from https://github.com/raspberrypi/firmware/commits/stable/ 
but with "sudo rpi-update 3590de0c181d433af368a95f15bc480bdaff8b47"
i got "invalid hash given" error
I wanted to rollback to 6.6.31

thidasi, 2024-07-12:
@pelwell sadly it didn't work, got a different error: "invalid git hash specified"

i tried this one: 0ccee173faa4f34a77befe752e0187a2678b2823
https://github.com/raspberrypi/rpi-firmware/commits/stable/

pelwell, 2024-07-12:
That hash works for me. Try a shortened version, so you are sure to get it right:
```
$ sudo rpi-update 0ccee
```

thidasi, 2024-07-12:
@pelwell it finds the package, but don't start the download after the Y (proceed)
[image]

popcornmix, 2024-07-12:
Can you run:
```wget https://github.com/raspberrypi/rpi-firmware/tarball/0ccee173faa4f34a77befe752e0187a2678b2823```
and confirm if that works?

thidasi, 2024-07-12:
i found the problem it was in my /etc/hosts

my mistake 

thanks for your kindness!

closed 2024-07-12 by thidasi