raspberrypi/firmware #1916: Pi1541 sd cards working on Pi3A never run on a 3B+ 2GB Pi!

Alectronics, 2024-08-19:
Tried invain to make a working SD card for my Raspberry Pi3B+ (2017). Never boots. Pi is okay running with original Raspberry Pi boot image. 

Gave up on trying because it never works on my Pi3B+ with the too limited given instructions that only work for Pi3A.

JamesH65, 2024-08-21:
It might be worth posting your issue on the forums, you are likely to get a more prompt response.

Alectronics, 2024-08-21:
Hi James. I already did and Stephen White helped me to solve the issue.

It was the wrong pre-compile 1.2x Raspberry Pi file that was presented online with the 3 to copy files to overwrite if needed any of those files in the root of the SD-card. Stephen gave the right and only working link to the working for both Pi3A and Pi3B+ file.
On : https://github.com/raspberrypi/firmware

After that my 3B+ kept rebooting after the in only seconds the picture of the Commodore disk drive on my Oled showed. Which never happened on my 3A Pi's.
After I used a 3Amp 5V usb power adapter it worked excellent on my Pi3B+ too.

Before the 3B+ Pi1541 never worked because the wrong RaspberryPi file was given that only worked on my two 3A Pi1541 Pi's and never on my 3B+ 1541 Pi. Why I wrote to Stephen wo was a great help! (I already donated a small amount for his great work before I even had made a working Pi1541!)

Best regards.
Albert van Bemmelen,
Weert, The Netherlands.