raspberrypi/firmware #1917: brcmfmac firmware error `kernel: brcmfmac: brcmf_sdio_htclk: HT Avail request error: -5`

samveen, 2024-08-22:
My Zero2W  had been up and running for close to 9 days, when the wifi firrmware gave errors and broke my connection to it today. With wifi connectivity shot, I triggered it's graceful shutdown to capture logs, and then booted it up again. I do not have more details than as below, as the log volume was large enough to exceed my log storage (`log2ram`  `lzo` compressed ramdisk of 100M), and previous logs were rotated out.

Details available:

The error messages I found (first, duplicate error lines, last; each kernel message misses the same error message multiple times, so actual number is much larger):
```
samveen@zero2w:~$ sudo journalctl -n 50000000 |grep -E 'brcmfmac: brcmf_sdio_htclk:|Missed [0-9]+ kernel messages' |head -n2
Aug 22 12:16:56 zero2w.cluster.samveen.in systemd-journald[454]: Missed 10 kernel messages
Aug 22 12:16:56 zero2w.cluster.samveen.in kernel: brcmfmac: brcmf_sdio_htclk: HT Avail request error: -5
```
```
samveen@zero2w:~$ sudo journalctl -n 50000000 |grep -E 'brcmfmac: brcmf_sdio_htclk:|Missed [0-9]+ kernel messages' |wc -l
28973
```
```
samveen@zero2w:~$ sudo journalctl -n 50000000 |grep -E 'brcmfmac: brcmf_sdio_htclk:|Missed [0-9]+ kernel messages' |tail -n2
Aug 22 12:37:07 zero2w.cluster.samveen.in kernel: brcmfmac: brcmf_sdio_htclk: HT Avail request error: -5
Aug 22 12:37:07 zero2w.cluster.samveen.in kernel: brcmfmac: brcmf_sdio_htclk: HT Avail request error: -5
```

Uptime:
```
samveen@zero2w:~$ last |grep reboot | head -n2 
reboot   system boot  6.6.31+rpt-rpi-v Thu Aug 22 14:03   still running
reboot   system boot  6.6.31+rpt-rpi-v Tue Aug 13 14:12 - 12:36 (8+22:24)
```

OS details:
```
samveen@zero2w:~$ uname -a && cat /etc/os-release
Linux zero2w.cluster.samveen.in 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

Firmware version:
```
samveen@zero2w:~$ apt list firmware-brcm80211 --installed && apt show firmware-brcm80211=1:20230625-2+rpt2  2>/dev/null |grep -E '^Package:|^Version:|^Status:|^Section:|^Source:|^Description:'
firmware-brcm80211/now 1:20230625-2+rpt2 all [installed,upgradable to: 1:20230625-2+rpt3]
^Description:'
Package: firmware-brcm80211
Version: 1:20230625-2+rpt2
Status: install ok installed
Section: non-free-firmware/kernel
Source: firmware-nonfree
Description: Binary firmware for Broadcom/Cypress 802.11 wireless cards
```
A newer update is available. I'll be installing it:
```
samveen@zero2w:~$ apt list firmware-brcm80211 --upgradable && apt show firmware-brcm80211  2>/dev/null |grep -E '^Package:|^Version:|^Status:|^Section:|^Source:|^Description:'
firmware-brcm80211/stable,stable 1:20230625-2+rpt3 all [upgradable from: 1:20230625-2+rpt2]
Package: firmware-brcm80211
Version: 1:20230625-2+rpt3
Section: non-free-firmware/kernel
Source: firmware-nonfree
Description: Binary firmware for Broadcom/Cypress 802.11 wireless cards
```

`raspinfo` output

```
samveen@zero2w:~$ raspinfo
System Information
[... 638 lines cut ...]
[   13.413875] zram0: detected capacity change from 0 to 204800
[   13.512344] EXT4-fs (zram0): mounted filesystem 638139b9-930e-4d7d-8ccb-72e1b87616c3 r/w with ordered data mode. Quota mode: none.
[   13.681117] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[   13.700124] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   13.700869] usbcore: registered new interface driver brcmfmac
[   13.813626] EXT4-fs (zram0): unmounting filesystem 638139b9-930e-4d7d-8ccb-72e1b87616c3.
[   13.851396] systemd[1]: log2ram.service: Main process exited, code=exited, status=1/FAILURE
[   13.852532] systemd[1]: log2ram.service: Failed with result 'exit-code'.
[   13.853541] systemd[1]: Failed to start log2ram.service - Log2Ram.
[   13.941463] systemd[1]: Starting systemd-journald.service - Journal Service...
[   13.953604] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[   13.954386] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Jun 14 2023 07:27:45 version 7.45.96.s1 (gf031a129) FWID 01-70bd2af7 es7
[   13.963124] systemd[1]: var-log.mount: Deactivated successfully.
[   13.968404] systemd[1]: var-hdd.log.mount: Deactivated successfully.
[   14.140420] systemd[1]: Listening on systemd-rfkill.socket - Load/Save RF Kill Switch Status /dev/rfkill Watch.
[   14.213052] systemd[1]: Starting systemd-rfkill.service - Load/Save RF Kill Switch Status...
[   14.268084] systemd[1]: Started systemd-journald.service - Journal Service.
[   14.408905] systemd-journald[429]: Received client request to flush runtime journal.
[   19.663123] Adding 970748k swap on /var/swap.  Priority:-2 extents:64 across:33091584k SS
[   21.848493] systemd[589]: memfd_create() called without MFD_EXEC or MFD_NOEXEC_SEAL set
[   22.197154] enc28j60 spi0.0 eth0: link down
[   22.197196] enc28j60 spi0.0 eth0: multicast mode
[   22.228940] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   23.294391] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
```

Shahn32125, 2024-10-08:
insightfull [192.168.1.1](https://19216811.cam/)  [192.168.100.1](https://1921681001.id/)

maxxsyntax, 2024-11-02:
Experiencing the same problem  The pi slows to a halt. 

```
sudo journalctl -n 50000000 |grep -E 'brcmfmac: brcmf_sdio_htclk:|Missed [0-9]+ kernel messages' |head -n2

Oct 23 14:12:34 raspberrypi systemd-journald[34643]: Missed 8 kernel messages
Oct 23 14:12:34 raspberrypi kernel: brcmfmac: brcmf_sdio_htclk: HT Avail request error: -5
```

```
pi@node1:~ $ sudo journalctl -n 50000000 |grep -E 'brcmfmac: brcmf_sdio_htclk:|Missed [0-9]+ kernel messages' |wc -l
10571674
```

```
sudo journalctl -n 50000000 |grep -E 'brcmfmac: brcmf_sdio_htclk:|Missed [0-9]+ kernel messages' |tail -n2
Oct 23 14:40:15 raspberrypi kernel: brcmfmac: brcmf_sdio_htclk: HT Avail request error: -5]
Oct 23 14:40:15 raspberrypi kernel: brcmfmac: brcmf_sdio_htclk: HT Avail request error: -5
```

```
pi@node1:~ $ uname -a && cat /etc/os-release
Linux node1 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

```
apt list firmware-brcm80211 --installed && apt show firmware-brcm80211=1:20230625-2+rpt2  2>/dev/null |grep -E '^Package:|^Version:|^Status:|^Section:|^Source:|^Description:'
Listing... Done
firmware-brcm80211/stable,stable,now 1:20230625-2+rpt3 all [installed]
N: There is 1 additional version. Please use the '-a' switch to see it
Package: firmware-brcm80211
```

```
pi@node1:~ $ raspinfo 
System Information
------------------

Raspberry Pi Zero 2 W Rev 1.0
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"

Raspberry Pi reference 2024-07-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 48efb5fc5485fafdc9de8ad481eb5c09e1182656, stage2

Linux node1 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
Revision	: 902120
Serial		: 000000003da64fdd
Model		: Raspberry Pi Zero 2 W Rev 1.0
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0, libcamera interfaces=0

Videocore information
---------------------

Aug 30 2024 19:19:11 
Copyright (c) 2012 Broadcom
version 2808975b80149bbfe86844655fe45c7de66fc078 (clean) (release) (start)

[... 1021 lines cut ...]
[ 4161.758961] ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[ 4161.758995] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST failed, err=-110
[ 4269.538940] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[ 4269.539306] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST failed, err=-110
[ 8421.598964] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[ 8424.162954] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[ 8424.163333] ieee80211 phy0: brcmf_cfg80211_get_station: GET STA INFO failed, -110
[ 8427.746958] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[ 8427.747512] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST failed, err=-110
[ 8433.630979] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[ 8433.631347] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST failed, err=-110
[ 8439.518974] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[ 8439.519352] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST failed, err=-110
[ 8445.662977] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[ 8445.663348] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST failed, err=-110
[ 8449.000012] brcmfmac: brcmf_sdio_read_control: last control frame is being processed.
[ 8449.000189] brcmfmac: brcmf_sdio_read_control: last control frame is being processed.
[ 8449.000332] brcmfmac: brcmf_sdio_read_control: last control frame is being processed.
[ 8449.000470] brcmfmac: brcmf_sdio_read_control: last control frame is being processed.
[ 8449.000608] brcmfmac: brcmf_sdio_read_control: last control frame is being processed.
[ 8853.726958] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[ 8853.727318] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST failed, err=-110
[ 8961.758980] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[ 8961.759353] ieee80211 phy0: brcmf_fill_bss_param: Failed to get bss info (-110)
```