raspberrypi/firmware #1918: Add tag on raspberrypi/linux for this release

Bajec, 2024-09-03:
Please add stable release tag on raspberrypi/linux repo branch on which firmware 1.20240902 was build! 
We have hard time tracking releases and source code without tags!

popcornmix, 2024-09-03:
Each commit in the firmware contains the hash of the linux tree it was built from.

e.g. top of stable branch contains:
https://github.com/raspberrypi/firmware/blob/stable/extra/git_hash

which points at https://github.com/raspberrypi/linux/commits/cf64a1dfecc2dc418efdd61701c1a4b185ab4761

Bajec, 2024-09-03:
> Each commit in the firmware contains the hash of the linux tree it was built from.
> 
> e.g. top of stable branch contains: https://github.com/raspberrypi/firmware/blob/stable/extra/git_hash
> 
> which points at https://github.com/raspberrypi/linux/commits/cf64a1dfecc2dc418efdd61701c1a4b185ab4761

Thank you, this will help. But, it would be nice to provide tags for stable releases like it was for last few releases.