raspberrypi/firmware #1921: config_hdmi_boost below 5 is not working

mrittler, 2024-10-07:
**Describe the bug**
I would like to reduce the signal strength of the HDMI interface via `config_hdmi_boost` to reduce EMC emissions. According to the documentation the minimum value for `config_hdmi_boost` is 0 and the maximum is 11. But values below 5 seem to be ignored and instead the default value of 5 is used.

**To reproduce**
- add `config_hdmi_boost=0` to  `/boot/config.txt`
- reboot
- run `vcgencmd get_config int` and check the value of `config_hdmi_boost`

**Expected behaviour**
`config_hdmi_boost=0`

**Actual behaviour**
`config_hdmi_boost=5`

**System**
* Raspberry Pi CM3+
* Custom Linux distribution
* firmware version: Aug 26 2022 14:04:36 102f1e848393c2112206fadffaaf86db04e98326
* Linux kernel: 5.15.34-v7

popcornmix, 2024-10-08:
I've had a look at the code. On CM3+ the default `config_hdmi_boost` is 5.

You can increase it by specifying a new positive `config_hdmi_boost` value.

But reducing it requires a negative setting.
`config_hdmi_boost=-1`
should get you 4.

mrittler, 2024-10-24:
Thanks a lot for the hint. Using negative values works indeed. 
What is the lowest value that can be used?

I know this is declared as a legacy option but I think it would be nice to update this in the following documentation accordingly: https://www.raspberrypi.com/documentation/computers/legacy_config_txt.html#config_hdmi_boost

popcornmix, 2024-10-24:
The code will accept values down to 0 after the subtraction (so `config_hdmi_boost=-5` is the lowest).
But this is not supported operation. The output will almost certainly not pass hdmi compliance (as the signal will be too low), and it no doubt won't work on some displays.

To fix interference issues you should focus on the shielding of your board and hdmi cable.

closed 2025-02-14 by popcornmix