raspberrypi/firmware #1922: Raspberry pi 5 se občas nepřipojí k LAN síti.

radekjuthner, 2024-10-18:
I started RPI5, pressed SHIFT. RPI loaded Network installer and then lost network connection. Could not use Network installer. Replugging the network cable solved the problem. 
Losing LAN connection happens very randomly - about once in 7 to 10 boots. 
If I continue to boot linux in a bad state, then it is not possible to ping the router from the RPI, but it is possible to ping the RPI from the router.
I originally suspected the Linux driver ([#6420](https://github.com/raspberrypi/linux/issues/6420)), but the error occurs before it even boots.

Device (s)
Raspberry Pi 5

System
cat /etc/rpi-issue
Raspberry Pi reference 2024-03-15
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, f19ee211ddafcae300827f953d143de92a5c6624, stage2

vcgencmd version
2024/10/10 00:36:47
Copyright (c) 2012 Broadcom
version 0e63dd5e (release) (embedded)

uname -a
Linux raspi5 6.6.44-v8+ https://github.com/raspberrypi/linux/issues/1789 SMP PREEMPT Mon Aug 5 15:22:52 BST 2024 aarch64 GNU/Linux

radekjuthner, 2024-11-28:
Really no idea? I know of multiple cases with this problem.

peterharperuk, 2024-11-28:
Are you saying the network installer is somehow breaking the lan connection when Linux boots?

peterharperuk, 2024-11-28:
Sorry, still confused. You were in the network installer you weren't getting an I address. How did you then boot Linux?

peterharperuk, 2024-11-28:
Right so it's not a problem "caused" by the network installer. Obviously I'm not seeing this problem. Can you post the output of "ip addr" before and after it works?

> In my opinion, the problem is before the operating system boots - probably in the bootloader firmware

Why do you think that? Could it not be an interoperability problem with your router?

radekjuthner, 2024-11-28:
I published the outputs here: [#6420](https://github.com/raspberrypi/linux/issues/6420 )

peterharperuk, 2024-11-28:
> I have a Raspberry pi 5 with SSD

I assume you actually mean an NVMe disk. Once you're connected you could use the imager on the OS to flash a blank SD card with an image. It would be interesting to know if you only have the problem when booting from this NVMe.

Your ethernet adapter seems to have a funny name. How did you setup this disk? Your ethernet adapter seems to have renamed itself end0 which might have some bearing on the problem?

radekjuthner, 2024-11-28:
I also have a problem with booting from the SD card. I don't think the problem is in the OS, but before the OS boots. When I get to the Network installer (shift key when turning on the RPI), no OS is booted yet and the network is sometimes unavailable.

peterharperuk, 2024-11-28:
The network installer that gets downloaded IS a linux image that's booted onto your device.

radekjuthner, 2024-11-28:
Below I attach the output when the error occurs. It's from a fresh OS install.

`ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 2c:cf:67:5a:36:5b brd ff:ff:ff:ff:ff:ff
    inet 192.168.100.180/24 brd 192.168.100.255 scope global dynamic noprefixroute eth0
       valid_lft 43170sec preferred_lft 43170sec
3: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 2c:cf:67:5a:36:5c brd ff:ff:ff:ff:ff:ff

dmesg | grep eth0
[    0.550764] macb 1f00100000.ethernet eth0: Cadence GEM rev 0x00070109 at 0x1f00100000 irq 112 (2c:cf:67:5a:36:5b)
[    3.748085] macb 1f00100000.ethernet eth0: PHY [1f00100000.ethernet-ffffffff:01] driver [Broadcom BCM54213PE] (irq=POLL)
[    3.748094] macb 1f00100000.ethernet eth0: configuring for phy/rgmii-id link mode
[    7.843823] macb 1f00100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control t

ip link show eth0
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP mode DEFAULT group default qlen 1000
    link/ether 2c:cf:67:5a:36:5b brd ff:ff:ff:ff:ff:ff

ip link show eth0
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP mode DEFAULT group default qlen 1000
    link/ether 2c:cf:67:5a:36:5b brd ff:ff:ff:ff:ff:ff

ip neigh
192.168.100.130 dev eth0 INCOMPLETE 
192.168.100.1 dev eth0 INCOMPLETE 

 lspci | grep Ethernet
0001:01:00.0 Ethernet controller: Raspberry Pi Ltd RP1 PCIe 2.0 South Bridge

ping from router to RPI:
 ping 192.168.100.180
PING 192.168.100.180 (192.168.100.180) 56(84) bytes of data.
64 bytes from 192.168.100.180: icmp_seq=42 ttl=64 time=1920 ms
64 bytes from 192.168.100.180: icmp_seq=43 ttl=64 time=880 ms
64 bytes from 192.168.100.180: icmp_seq=86 ttl=64 time=880 ms
64 bytes from 192.168.100.180: icmp_seq=117 ttl=64 time=880 ms
64 bytes from 192.168.100.180: icmp_seq=148 ttl=64 time=880 ms
64 bytes from 192.168.100.180: icmp_seq=179 ttl=64 time=880 ms
64 bytes from 192.168.100.180: icmp_seq=210 ttl=64 time=880 ms
64 bytes from 192.168.100.180: icmp_seq=241 ttl=64 time=880 ms
64 bytes from 192.168.100.180: icmp_seq=272 ttl=64 time=880 ms
64 bytes from 192.168.100.180: icmp_seq=303 ttl=64 time=880 ms
64 bytes from 192.168.100.180: icmp_seq=334 ttl=64 time=890 ms
64 bytes from 192.168.100.180: icmp_seq=377 ttl=64 time=1920 ms
64 bytes from 192.168.100.180: icmp_seq=378 ttl=64 time=880 ms
64 bytes from 192.168.100.180: icmp_seq=421 ttl=64 time=1920 ms
64 bytes from 192.168.100.180: icmp_seq=422 ttl=64 time=880 ms
64 bytes from 192.168.100.180: icmp_seq=465 ttl=64 time=1920 ms
64 bytes from 192.168.100.180: icmp_seq=466 ttl=64 time=870 ms
64 bytes from 192.168.100.180: icmp_seq=509 ttl=64 time=1920 ms
64 bytes from 192.168.100.180: icmp_seq=510 ttl=64 time=880 ms`

radekjuthner, 2024-11-28:
ARP did not show anything interesting.
My router is Banana BPI-R3 with Arch linux with ISC DHCPd. The RPI has a dedicated IP address based on the MAC address. Raspberry pi 4 and Raspberry pi 400 run on the same network without problems. Of course also other devices Smart TV, Smart Phones, printer. Only RPI 5 has a problem.

peterharperuk, 2024-11-28:
> The RPI has a dedicated IP address based on the MAC address

Does it behave better if you remove the static IP address from the router? I can't really imagine why it would make a difference but it might be worth a try.

radekjuthner, 2024-11-29:
I tried running things, but the result was the same. I let the IP address be assigned via DHCP without filtering by MAC address, assigning the IP address depending on the MAC address, I set a static IP address on the RPI5. It did not solve the problem. As I wrote, the problem must be somewhere at the firmware level, or I have faulty hardware.

peterharperuk, 2024-11-29:
I wouldn't rule it out but I doubt it's a firmware problem.

>I set a static IP address on the RPI5

I didn't quite follow all this. Are you still using a static IP address or not? If not, is the DHCP server the router or another machine?

Can you run the following two commands and post the output just for my curiosity?

```
sudo vcgencmd otp_dump | grep "5[1-2]:"
cat /sys/firmware/devicetree/base/chosen/rpi-duid
```

radekjuthner, 2024-11-29:
sudo vcgencmd otp_dump | grep "5[1-2]:"
cat /sys/firmware/devicetree/base/chosen/rpi-duid

Returns:
51:2ccf675a
52:365c0000

It doesn't know if you wanted an output when the network connection is faulty. The output provided is from a problem-free network connection.

David-H-Bolton, 2025-03-21:
I can confirm I also have this intermittent problem with a Raspberry Pi 5. It boots from the SD Card that came with it in a starter pack and I keep it updated every few days. At first I thought it was a faulty ethernet cable and swapped it with a cable to my work laptop (I work from home). It just happened today and I've used the work laptop with the swapped cable for several days without incident.  Unplugging and replugging the Pi cable fixed it.

Treah, 2025-04-04:
I have the same/very similar issue with my pi5. It used to be intermittent but today it just stopped working completely. When a cable is plugged in I get lights on the jack on the PI and it looks like its getting traffic but I wont communicate at all on the network. Even manual IP address assignment fails to transmit any packets.  Linux sees the device and the wireless works fine with DHCP and everything else but wired is a bust. I suspect its the Ethernet chip that is on these things is either faulty or there is some problem with firmware but I cant be sure.

afullagar, 2025-04-18:
I have the same / similar problem. I purchased 2 x RPI 5s. One I wanted to run using SSD etc and the other normal SDcard. I replaced my RPI4 running as my internet gateway. has inbuilt ethernet and two USB 1G dongels. Rpi 4 works great but running buster. Created an RPI 5 gateway on bookworm lite and used same config / setup. Just had continual network issues. Packets dropping and then no ip network connectivity at all. I ripped it out and went back to RPI 4 - working great. There is definately something wrong with teh RPI5s. Not sure if hardware, software or firmwae but tehy are not reliable as a router.