raspberrypi/firmware #1923: Feature Request:  Add support for TCS3725 ahd VMEML6040 RGB Color Sensors

Jibun-no-Kage, 2024-10-29:
Feature Request:  Add support for TCS34725 ahd VMEML6040 RGB Color Sensors as device tree overlays.

Example of TCS34725 RGB Color Sensor...
https://www.adafruit.com/product/1334

VMEML6040 RGB Color Sensor...
https://www.taydaelectronics.com/veml6040-color-sensor-rgbw-i2c-module.html

pelwell, 2024-10-29:
TCS34725 will need it's driver enabling - `CONFIG_TCS3472=m`.  It makes no use of Device Tree configuration, so it should just be a matter of adding the name to the `i2c-sensor` overlay.

The VEML6040 (not VMEML6040) driver doesn't exist in 6.6, but it is there in 6.12. It also doesn't seem to require any configuration.

If I have a moment I might throw together the necessary config and overlay updates against 6.12. What's your application?

pelwell, 2024-10-29:
And which I2C addresses can the devices appear on - don't make me read the datasheets.