raspberrypi/firmware #1924: Commit 0a3f48f2 makes RPi 3B+ unlikely to boot up

tomty89, 2024-11-05:
/cc @graysky2

**Describe the bug**
I have an RPi 3B+ running ArchLinuxARM with upstream kernel (linux-aarch64 6.11.6-2) and (uboot-raspberrypi 2024.07-4). After the [update](https://github.com/archlinuxarm/PKGBUILDs/commit/a435a47e4fa39c86451d53014a48c2a82d088c89) of the bootloader package (from c3a48014 to 0a3f48f2), getting the Pi to boot up becomes very difficult.

Unfortunately I don't have any display device available, so I cannot tell if any error would show up when it fails to boot (or if it just stuck at some point without giving any error). All I can tell is, the chance that I get to successfully boot it up (i.e. being able to ping and ssh it) is like 1 time out of 10~20 reboot (by reboot I mean like power cycle).

When I succeed, everything works fine. I can install and uninstall large system package and use it without any problem, so I don't think it has anything to do with the microSD card. And there's nothing special I need to succeed. Just power cycle it until I have enough luck to "got through" the problem.

After rolling back to c3a48014 the problem is at least apparently gone. (I mean, the chance that it fails to boot up at least seems to drastically decrease. On and off (I mean like among booloader updates) the Pi have occasionally had shown similar problem, but the chance is not high, like the successful rate is at least 1/2.)

**To reproduce**
Update bootloader files from those of c3a48014 to those of 0a3f48f2

**Expected behaviour**
Boot up successfully every time or at least most of the times

**Actual behaviour**
Unacceptably low successful rate of booting up

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? Pi3B+
* Which OS and version (`cat /etc/rpi-issue`)? ArchLinuxARM
* Which firmware version (`vcgencmd version`)? 0a3f48f2
* Which kernel version (`uname -a`)? `Linux archlinux 6.11.6-arch1-1 #1 SMP PREEMPT_DYNAMIC Fri, 01 Nov 2024 03:30:41 +0000 x86_64 GNU/Linux`

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
Add any other relevant context for the problem.

pelwell, 2024-11-05:
I may have misunderstood you, but something doesn't add up. You say you are running an upstream kernel, and that the most recent bootloader change makes boot unreliable, but the only changes from your "good" state to your "bad" state are kernel change - the firmware is unaltered.

Which kernel are you actually running - the upstream one from Arch or the one in the bootloader package?

tomty89, 2024-11-05:
Hmm, okay. I didn't realize the raspberry-bootloader package update was actually a "no-op".

The whole story was like, I updated both the kernel package (upstream, not rpi kernel; from `linux-aarch64-6.11.5-1` to `linux-aarch64-6.11.6-2`) and the raspberry-bootloader package (different package build on ArchLinuxARM, but same content; I quick checked their extractions with `rsync -rvc` as well).

The problem occurred, and after it persists for like 40 power cycles (with only 2 boots succeeded), I rolled back both packages, and rebooted a few times without problem. I figured the problem was the bootloader files instead of the upstream kernel, because the latter was a minor update, and the history (as far as I'm concerned) of the former were "smellier". So I re-update (only) the kernel package again, problem doesn't seem to have come back.

TL;DR I don't really know what happened. Maybe a solar flare or something. Or maybe my Pi itself is dying in a certain way and happened to have a serious "episode". Or maybe some "nondeterministic" long-time bug happened to get enough luck to put on a show. Closing for now.

closed 2024-11-05 by tomty89