raspberrypi/firmware #1925: RPI 5 fatal firmware error

nadir23, 2024-11-12:
I'm experiencing a boot issue on RPI5 as follows:

1) When I try to boot it with uSD card having bootloader update firmware, it flashes 4 long, 5 short and red screen on display.

2) When the uSD card is loaded with OS the status led is constant red.

3) When I try uSD with OS and serial debug no serial infos are given.

4) When I try uSD with bootloader update and serial debug the following log is displayed:

`1.85 pieeprom.sig`
`1.85 hash: 6fafdd66f0bb1f9ad5eb68ba7ba1fb537a6182567e9be83e64bf9a2d2591820a`
`1.91 ts: 1727272381`
`2.28 Unexpected error @ 0x00010164`
`2.28 USB-OTG disconnect`
`2.31 BOOT ERROR: code 45 - ''`

Thank you in advance for your help.

peterharperuk, 2024-11-12:
It's probably as assertion failure. You seem to be suggesting "normal" boot is not working? Do you see anything on a screen if you don't have any uSD card inserted?

pelwell, 2024-11-12:
Also:
* How long have you had this Pi 5? 
* Has it ever worked? 
* Can you associate the start of the failures with any significant event?

popcornmix, 2024-11-12:
Can you try with a different sdcard?

nadir23, 2024-11-12:
> Also:
> 
> * How long have you had this Pi 5?
> * Has it ever worked?
> * Can you associate the start of the failures with any significant event?

I had it for two months, started it using a 2A capable PSU. Installed a Raspbian OS on uSD and working as it should.
When trying to use an SSD drive using a dedicated hat the OS didn't detect the drive and alerted of a limited current capability. 
I hadn't access to the original RPI5 psu, so I bought a 5V3A capable PD charger and set max_usb_current=1.
The RPI5 had difficulty starting sometimes it starts and other time not. After, it totally stopped.

nadir23, 2024-11-12:
> Can you try with a different sdcard?

I tried different uSD cards without success. the same uSD cards work on an other RPI4B.

pelwell, 2024-11-12:
3A is not sufficient for a Pi 5 with an SSD - we require 5A to automatically enable `usb_max_current_enable` (which is the name of the config setting, not `max_usb_current`).

popcornmix, 2024-11-12:
The `0x00010164` address is a failure in communicating with sdcard.
If a different sdcard doesn't work then it's a hardware failure (possibly damage to the sdcard socket, or sdcard related components).

Do you get anything out of uart when booting without an sdcard? (you should if you have a valid bootloader).
If bootloader is functional it may be possible to boot from an sdcard in a usb adaptor.

nadir23, 2024-11-13:
> 3A is not sufficient for a Pi 5 with an SSD - we require 5A to automatically enable `usb_max_current_enable` (which is the name of the config setting, not `max_usb_current`).

I'v seen some tests on the web that made them work using RPI4 PSU. I'v bought the original 27W one and will test it when delivered.

nadir23, 2024-11-13:
> The `0x00010164` address is a failure in communicating with sdcard. If a different sdcard doesn't work then it's a hardware failure (possibly damage to the sdcard socket, or sdcard related [...]

Thank you very much for your reply, I was looking for a diagnosis that may lead to the identification of the origin of the malfunction. 

When I tried booting the RPI5 without uSD, the UART didn't show any info.

When I try the uSD with bootloader update using RPI Imager using the 5V3A PD charger, another error has emerged:

  `1.83 pieeprom.sig`
  `1.83 hash: 6fafdd66f0bb1f9ad5eb68ba7ba1fb537a6182567e9be83e64bf9a2d2591820a`
  1.89 ts: 1731065899
  `2.08 Unexpected error @ 0x0000f9a0`
  `2.09 USB-OTG disconnect`
  `2.11 BOOT ERROR: code 45 - ''`

The previous one `0x00010164` was displayed when the RPI was powered using a bench power supply from 5V pins. The current one `0x0000f9a0` when using the PD charger.

nadir23, 2024-11-17:
I have tried the original 27W PSU, the same issue persists.

glanduin, 2025-11-07:
I am seeing the exact same error on my Pi5 but only occasionally. It happens on the very first power-up. And if i turn the power off and on again, it boots fine. It never does this during the reboots. I would expect the system to "reset" itself after such an error, but unfortunately it doesn't do it. I have to manually reset it. And like i said, after a reset, it works. This started to happen after a firmware update (i can't pinpoint which one) but i am 100% sure i have never seen this error like let's say untill the last 3 months or so.

timg236, 2025-11-07:
There is no automatic reset by default to avoid boot loops.

However, you can set a watchdog to do a reset if you think this is an intermittent error
https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#BOOT_WATCHDOG_TIMEOUT

tobiasBora, 2025-12-03:
I also experience fatal firmware error but when booting from NVMe: https://github.com/raspberrypi/firmware/issues/2006

timg236, 2025-12-03:
From the link-register the failure appears to be a PCIe init issue, possibly this device as an unusual set of memory regions. 

Has this ever worked?

P33M, 2025-12-05:
> > This is due to the fact that the controller is not backward compatible with PCIe Gen 2.
> 
> Wow - I'm surprised that's even allowed.

It's not, Table 4-5 in PCIe base specification 3.1a - data rate identifiers are a thermometer code. But that doesn't preclude an interop issue where there's marginal link quality at gen 2.

The crash is an assert which fires if an attempt to read past the end of the assigned PCI bus segment is made - in this case 32MB from the base of the outbound window. It could also fire if the computed address is bad - which may happen if at any point the readback for a BAR is all 1s - i.e. link went away.

Firmware fixes aren't going to fix that, but to get a better chance of booting we could limit the RC to only advertise gen 1, so no autonomous speed uplift happens.

tobiasBora, 2025-12-05:
Thanks!

> Firmware fixes aren't going to fix that

I'm curious: why is there no errors when accessing the disk from the kernel? What can we do in kernel that can't be done here?

> we could limit the RC to only advertise gen 1, so no autonomous speed uplift happens.

Happy to try once implemented, thanks ^^ But does it mean that I'm stuck with gen 1 after or can the kernel upgrade it to gen 2 or 3?