raspberrypi/firmware #1926: Pi 5 RTC: 32-bit timestamp and potential Y2038 error?

dsseng, 2024-11-22:
**Is this the right place for my bug report?**
Yes, this is about VPU firmware interface.

**Describe the bug**
Apparently VPU firmware interface for RTC operations on Pi 5 expects all register operations to be 32-bit wide. This also applies to the `RTC_TIME` register holding Unix timestamp in form of seconds. Limiting one to 32 bits and casting it to 64 bits like done in the Linux driver could cause 2038 error when the timestamp exceeds 32-bit integer max value.

**To reproduce**
- [function to read time](https://github.com/raspberrypi/linux/blob/ab9bb48bcf5bebc8a75df84fc992096ed9d1c57b/drivers/rtc/rtc-rpi.c#L42-L47)
- [function to set time](https://github.com/raspberrypi/linux/blob/ab9bb48bcf5bebc8a75df84fc992096ed9d1c57b/drivers/rtc/rtc-rpi.c#L54-L57)

**Expected behaviour**
`RTC_TIME` register (perhaps a virtual one) should have 64-bit width (well, in case hardware supports that) to support the same timestamp boundaries as modern software expects.

From what I've seen in mainline Linux drivers for DA9063 and similar chips they store RTC data in separate registers for day, month, year, hour etc. If that's the case for the undocumented DA9090 as well it should be able to provide 64-bit timestamps.

**Actual behaviour**
Timestamp stored in `RTC_TIME` is 32-bit and driver upcasts that value to 64 bits.

**System**

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
* Which OS and version (`cat /etc/rpi-issue`)? Noticed in a kernel from `rpi-6.12.y` branch while learning the interface to write a U-Boot driver
* Which firmware version (`vcgencmd version`)? Perhaps applies to any version

```
2024/11/05 12:38:12 
Copyright (c) 2012 Broadcom
version 3c4fc886 (release) (embedded)
```

* Which kernel version (`uname -a`)? `Linux helios 6.6.59-v8-16k+ #1811 SMP PREEMPT Tue Nov  5 20:06:38 GMT 2024 aarch64 GNU/Linux`

**Logs**

**Additional context**

pelwell, 2024-11-22:
I'll start the ball rolling with the observation that the RTC (a DA9091, not DA9090) contains a 32-bit seconds counter, so that's the largest value it can cope with. Presenting a 64-bit API is therefore misleading. The Linux driver could be changed to spot times older than some threshold and set bit 32 appropriately, but I think that's the best we can do.

pelwell, 2024-11-22:
Plus, your initial premise is wrong. Time conversion is done by `rtc_time64_to_tm`:
```
/**
 * rtc_time64_to_tm - converts time64_t to rtc_time.
 *
 * @time:	The number of seconds since 01-01-1970 00:00:00.
 *		(Must be positive.)
 * @tm:		Pointer to the struct rtc_time.
 */
void [rtc_time64_to_tm](https://elixir.bootlin.com/linux/v6.11/C/ident/rtc_time64_to_tm)([time64_t](https://elixir.bootlin.com/linux/v6.11/C/ident/time64_t) [time](https://elixir.bootlin.com/linux/v6.11/C/ident/time), struct [rtc_time](https://elixir.bootlin.com/linux/v6.11/C/ident/rtc_time) *[tm](https://elixir.bootlin.com/linux/v6.11/C/ident/tm))
{
...
```
The 32-bit value read over the firmware mailbox interface is interpreted as an unsigned 32-bit number. When passed as a parameter to the above function it is promoted to an unsigned 64-bit number, and then treated as an offset from 1st Jan 1970. This gives a rollover date of (According to Wikipedia - I couldn't be bothered) 7 February 2106 at 06:28:15 UTC.

dsseng, 2024-11-22:
Ah, alright, if the register is unsigned it's pretty much okay. Thank you for clarification!

closed 2024-11-22 by dsseng

pelwell, 2024-11-22:
> if the register is unsigned 

Signedness, like beauty, is in the eye of the beholder - it's a matter of interpretation. Here, the interpretation is done in the kernel driver, where it cruelly prevents one from winding the clock back to the 19th century.

dsseng, 2024-11-22:
Well said! :slightly_smiling_face: 

Done it in the same way, !remindme 2106 :wink: