raspberrypi/firmware #1934: RPi4: Kernel clock_gettime64() vDSO disabled on 32-bit kernel due to firmware not initializing timer registers

rsaxvc, 2025-01-08:
**Is this the right place for my bug report?**

I think so. But, it's an integration issue between firmware and kernel, so it might belong there.

**Describe the bug**
Currently, the Raspberry-Pi Linux kernel disables clock_gettime64 vDSO due to BCM2711's timer registers being marked as not configured by firmware. This is a request that if the RaspberryPi isn't configuring this timer, to do so.

**To reproduce**

1. strace the following program

```
#include <time.h>

int main(){
struct timespec tp;
return clock_gettime(CLOCK_MONOTONIC, &tp);
}
```
2.strace shows clock-gettime64 systemcall is in use instead of vDSO.

**Expected behaviour**
I expect that the firmware would configure the timer registers, then Linux would not need to mark them with `cpu-registers-not-fw-configured`, then the clock_gettime64() vDSO should work.

**Actual behaviour**
The Raspberry Pi foundation's kernel believes the firmware has not configured the timer registers. Did the firmware configure them? If not, could it?

**System**

```
$ cat /etc/issue
Raspbian GNU/Linux 11 \n \l
$ vcgencmd version 
Jan  5 2023 10:46:54 
Copyright (c) 2012 Broadcom
version 8ba17717fbcedd4c3b6d4bce7e50c7af4155cba9 (clean) (release) (start)
$ uname -a
Linux raspberrypi 6.1.21-v7l+ #1642 SMP Mon Apr  3 17:22:30 BST 2023 armv7l GNU/Linux
$ cat /proc/cpuinfo | tail -4
Hardware	: BCM2711
Revision	: d03114
Serial		: 10000000701809ed
Model		: Raspberry Pi 4 Model B Rev 1.4
```

**Additional context**
My autopilot spends a lot of time doing this syscall.

pelwell, 2025-01-08:
Is this bug report completely accurate? I agree that a 64-bit userland (which implies a 64-bit kernel) _does not_ require clock_gettime64 calls, and that a 32-bit kernel (and therefore 32-bit userland) _does_ require them, but I also find that a 32-bit userland with a 64-bit kernel _does_ need them. I therefore think that the problem correlates with the flavour of userland in use.

rsaxvc, 2025-01-09:
> Is this bug report completely accurate?

I believe so, and I believe it's kernel-related, not userland-related, but these are the facts I'm certain of:

- Applications on my Pi4B with raspberrypi kernel and device tree doesn't use the clock_gettime64() vDSO, and use the syscall instead.
- If I remove cpu-registers-not-fw-configured from the timer in the device tree, then applications use the vDSO, and it appears to work now.
- `cpu-registers-not-fw-configured` is [Only supported for 32-bit systems which follow the ARMv7 architected reset values.](https://www.kernel.org/doc/Documentation/devicetree/bindings/timer/arm%2Carch_timer.yaml)
- As of kernel commit 688837ba49c9e, `cpu-registers-not-fw-configured` is not mentioned in arch/arm64, only in `arch/arm/kernel/vdso.c`, where its presence triggers calling `vdso_nullpatch_one()` on "__vdso_gettimeofday", "__vdso_clock_gettime", and "__vdso_clock_gettime64" - I think that's what disables the vDSO.