raspberrypi/firmware #1939: Devicetree root has no `chassis-type` property

septatrix, 2025-01-29:
**Describe the bug**
The devicetree does not have a `chassis-type` property indicating the form factor of the system.

**To reproduce**
`cat /proc/device-tree/chassis-type`

**Expected behaviour**
The cat command should output `"embedded"`

**Actual behaviour**
The cat command fails because the file/property does not exist

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? RPi 4B
* Which OS and version (`cat /etc/rpi-issue`)? `2024-11-19, 891df1e21ed2b6099a2e6a13e26c91dea44b34d4, stage4`
* Which firmware version (`vcgencmd version`)? `version 2ae30f53898ae2f1ba77ff570a92991bedfb0398 (clean) (release) (start)`
* Which kernel version (`uname -a`)? `6.6.62+rpt-rpi-v8`

**Additional context**

> Specifies a string that identifies the form-factor of the system. The property value can be one of:
> * "desktop"
> * "laptop"
> * "convertible"
> * "server"
> * "tablet"
> * "handset"
> * "watch"
> * "embedded"

From https://devicetree-specification.readthedocs.io/en/latest/chapter3-devicenodes.html#root-node. Labelled as OR=Optional but Recommended

6by9, 2025-01-29:
Not a firmware bug - those type of DT properties come from the DT files in the kernel tree. https://github.com/raspberrypi/linux/tree/rpi-6.6.y/arch/arm/boot/dts/broadcom/

Not all Pi systems would be classified as "embedded".
Pi400 and Pi500 would be closer to "desktop". There are a number of CM based NAS solutions that would be "server". [CutiePi](https://cutiepi.io/) would be "tablet", as would a few other CM products. 

Of the DT files in mainline Linux:
- for arch/arm, only Qualcomm and Samsung have added it to any of their DT files.
- for arch/arm64 there are a few more - some boards from Allwinner, Amlogic, Apple, Exynos, Freescale, Mediatek, Qualcomm, Rockchip, and TI.

So it's far from ubiquitous.

I know of nothing making use of this property - what benefit do you see from adding this property?

septatrix, 2025-01-29:
> Not a firmware bug - those type of DT properties come from the DT files in the kernel tree. https://github.com/raspberrypi/linux/tree/rpi-6.6.y/arch/arm/boot/dts/broadcom/

I thought they belonged here because they are also stored in this repo under boot/ but I can also raise this in the linux repo (or someone please move it over to there).

> Not all Pi systems would be classified as "embedded". Pi400 and Pi500 would be closer to "desktop". There are a number of CM based NAS solutions that would be "server". [CutiePi](https://cutiepi.io/) would be "tablet", as would a few other CM products.

That is correct. Luckily e.g. the Pi400/500 have separate device trees so they can be set to desktop without a problem. And if I recall correctly some of the products using a compute module ship their own devicetree so changing the type for them would also not be a problem.

> Of the DT files in mainline Linux:
> 
>     * for arch/arm, only Qualcomm and Samsung have added it to any of their DT files.
> 
>     * for arch/arm64 there are a few more - some boards from Allwinner, Amlogic, Apple, Exynos, Freescale, Mediatek, Qualcomm, Rockchip, and TI.
> 
> So it's far from ubiquitous.

Many of those DTs are not for specific devices but only chip classes which, like the CM, could be used for all kinds of stuff. The chassis-type is then often set by downstream vendor patches (i.e. OEMs)

> I know of nothing making use of this property - what benefit do you see from adding this property?

It can be used as a fallback for `/sys/class/dmi/id/chassis_type` (which exposes equivalent information for UEFI machines). This information is e.g. used to show machine icons (tower PC, laptop, tablet etc.) on settings pages.

pelwell, 2025-01-29:
That sounds very cosmetic. I could move it to the linux repo, but it would only delay the inevitable.

closed 2025-01-29 by pelwell

6by9, 2025-01-29:
If someone fancied submitting a sensible PR to the linux repo of the basic "desktop" for Pi400 and Pi500, and "embedded" for everything else, then it would be considered for merging.
Ideally it'd be submitted upstream via the linux-arm-kernel mailing list.