raspberrypi/firmware #1944: raspberrypi-firmware soc:firmware: Request 0x00030002 returned status 0x00000000

randomMesh, 2025-02-10:
Hi, 

i am running Void Linux on my Raspberry Pi 5 and i see a lot of errors in the kernel log,

`raspberrypi-firmware soc:firmware: Request 0x00030002 returned status 0x00000000`

I don't know if Void Linux is supported here but i want to ask if you can assist getting rid of these errors.

I've seen [this thread](https://forums.raspberrypi.com/viewtopic.php?t=366030) on the official Raspberry forums in which a Raspberry Pi engineer says the something has killed the firmware and the error can occur if

- Too high an overclock
- Inadequate power supply
- Sending invalid messages (less likely on the pi5 where firmware does a lot less).
- Corrupting firmware's memory (would likely need sudo/mmap to be allowed to do this outside of kernel)
- Faulty hardware

I do not overclock my PI.
I use the official PSU.
I don't have faulty hardware since if i boot Raspberry Pi OS, the errors are gone.

So it must either be "Sending invalid messages" or "Corrupting firmware's memory".

```
[root@xxx ~]$ strings /boot/start.elf | grep VC_BUILD
VC_BUILD_ID_USER: dom
VC_BUILD_ID_TIME: 11:47:03
VC_BUILD_ID_VARIANT: start
VC_BUILD_ID_TIME: Dec  5 2024
VC_BUILD_ID_BRANCH: bcm2711_2
VC_BUILD_ID_HOSTNAME: buildbot
VC_BUILD_ID_PLATFORM: raspberrypi_linux
VC_BUILD_ID_VERSION: 03554ca336a03ace164f36755144e0d8c060062d (clean)
```

Thanks!

randomMesh, 2025-02-11:
I am not sure but i think these errors are somehow connected to the inbuilt WiFi. If i disable wifi by adding this to config.txt, there errors are gone:

`dtoverlay=disable-wifi-pi5`

pelwell, 2025-02-11:
After some discussions here, I think the problem may be symptomatic of a busy system rather than a crashed firmware. If I'm right, part of the response from the firmware has not reached memory in time for the ARM cores to see it. This can be avoided by the use of a memory barrier.

I've attached a trial build of the Pi 5 EEPROM with an added memory barrier.

1. Download the file, unzip it and copy the resulting pieeprom_250211.bin to the Pi 5.
2. Run `sudo rpi-eeprom-update -d -f pieeprom_250211.bin`.
3. Reboot.

Then see if the problem remains.

[pieeprom_250211.zip](https://github.com/user-attachments/files/18750557/pieeprom_250211.zip)

randomMesh, 2025-02-11:
Thanks for looking into this.

I installed the EEPROM and report back if the problem remains.

```
BOOTLOADER: up to date
   CURRENT: Tue 11 Feb 13:09:56 UTC 2025 (1739279396)
    LATEST: Wed 22 Jan 00:16:51 UTC 2025 (1737505011)
   RELEASE: latest (/usr/lib/firmware/raspberrypi/bootloader-2712/latest)
            Use raspi-config to change the release.
```

randomMesh, 2025-02-11:
This seems to have fixed the problem, the errors don't show up in the kernel log anymore.

Thanks again.

pelwell, 2025-02-11:
Cool. The problem has been there since approximately day 1 (back to the Broadcom days), but the fix has now been merged and will be in future EEPROM builds.

randomMesh, 2025-02-11:
Aaaand the errors are back after some hours.

```
[ 1362.611672] raspberrypi-firmware soc:firmware: Request 0x00030002 returned status 0x00000000
[ 1929.483519] raspberrypi-firmware soc:firmware: Request 0x00030002 returned status 0x00000000
[ 2676.419304] raspberrypi-firmware soc:firmware: Request 0x00030002 returned status 0x00000000
[ 3464.499826] raspberrypi-firmware soc:firmware: Request 0x00030002 returned status 0x00000000
[ 3473.299627] raspberrypi-firmware soc:firmware: Request 0x00030002 returned status 0x00000000
[ 3516.949766] raspberrypi-firmware soc:firmware: Request 0x00030002 returned status 0x00000000
[ 3527.149406] raspberrypi-firmware soc:firmware: Request 0x00030002 returned status 0x00000000
[ 3579.949806] raspberrypi-firmware soc:firmware: Request 0x00030002 returned status 0x00000000
[ 3620.349507] raspberrypi-firmware soc:firmware: Request 0x00030002 returned status 0x00000000
[ 3624.616628] raspberrypi-firmware soc:firmware: Request 0x00030002 returned status 0x00000000
[ 3674.617100] raspberrypi-firmware soc:firmware: Request 0x00030002 returned status 0x00000000
[ 3792.900262] raspberrypi-firmware soc:firmware: Request 0x00030002 returned status 0x00000000
[ 3882.300193] raspberrypi-firmware soc:firmware: Request 0x00030002 returned status 0x00000000
[ 4032.616947] raspberrypi-firmware soc:firmware: Request 0x00030002 returned status 0x00000000
[ 4069.583695] raspberrypi-firmware soc:firmware: Request 0x00030002 returned status 0x00000000
[ 4153.700459] raspberrypi-firmware soc:firmware: Request 0x00030002 returned status 0x00000000
[ 4386.551242] raspberrypi-firmware soc:firmware: Request 0x00030002 returned status 0x00000000
[ 4406.818251] raspberrypi-firmware soc:firmware: Request 0x00030002 returned status 0x00000000
[ 4505.301339] raspberrypi-firmware soc:firmware: Request 0x00030002 returned status 0x00000000
[ 4563.818042] raspberrypi-firmware soc:firmware: Request 0x00030002 returned status 0x00000000
[ 4667.617766] raspberrypi-firmware soc:firmware: Request 0x00030002 returned status 0x00000000
[ 4757.585116] raspberrypi-firmware soc:firmware: Request 0x00030002 returned status 0x00000000
[ 5290.668543] raspberrypi-firmware soc:firmware: Request 0x00038002 returned status 0x00000000
[ 5290.668567] raspberrypi-clk soc:firmware:clocks: Failed to change fw-clk-arm frequency: -22
[ 5377.935923] raspberrypi-firmware soc:firmware: Request 0x00030002 returned status 0x00000000
[ 5557.115177] raspberrypi-firmware soc:firmware: Request 0x00030002 returned status 0x00000000
[ 5829.537197] raspberrypi-firmware soc:firmware: Request 0x00030002 returned status 0x00000000
```

pelwell, 2025-02-12:
I received a notification of a comment saying that the problem was back. The fact that the comment is not here suggests it was wrong in some respect. Is this something I still need to look into or can I close this issue?

randomMesh, 2025-02-12:
You can close the issue, the problem is not back.

I wanted to configure the bootloader you gave me via `rpi-eeprom-config -e` and i didn't know this command would install the old eeprom afterwards.

My fault, sorry for the noise.

pelwell, 2025-02-12:
No problem. Thanks.

closed 2025-02-12 by pelwell

flatsiedatsie, 2026-01-09:
I just saw this error in my logs on a Pi 4 with the latest Raspberry Pi OS 64 bit.