raspberrypi/firmware #1945: Mini UART broken somewhere between 13691cee95902d76bc88a3f658abeb37b3c90b03 and 8a8adaf2f1d6836facdc1c66f4e838b2025ec851 on rpi3b+

petemoore, 2025-02-13:
**Describe the bug**
After upgrading my project's firmware from 13691cee95902d76bc88a3f658abeb37b3c90b03 to 8a8adaf2f1d6836facdc1c66f4e838b2025ec851 the mini uart interface seems to have a timing issue on Raspberry Pi 3B+, whereby I receive garbled output coming from the pi (suspect clock timing issue). Reverting the firmware version resolves the issue.

**To reproduce**
Download and run https://github.com/spectrum4/spectrum4/releases/tag/v0.0.3 with mini uart connected (which ships with firmware version 13691cee95902d76bc88a3f658abeb37b3c90b03), and see everything working fine. Now update firmware to version 8a8adaf2f1d6836facdc1c66f4e838b2025ec851 and see the garbled output from the kernel.

**Expected behaviour**
Expected firmware upgrade to not affect mini uart logging.

**Actual behaviour**
Add a clear and concise description of what actually happened.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi?

3B+

**Additional context**

UART code (which worked up until this firmware upgrade):
  * https://github.com/spectrum4/spectrum4/blob/main/src/spectrum4/kernel/uart.s

Config.txt:
  * https://github.com/spectrum4/spectrum4/blob/main/src/spectrum4/kernel/config.txt

I'm mostly curious if there are any known changes to the mini uart between these two firmware versions. If required, I could try to bisect and find the exact commit it broke, but that will be quite a labour intensive process, so preferred to ask the question first! Many thanks! :-)

pelwell, 2025-02-14:
The mini0-UART isn't broken, exactly, but the baud rate is 1.6 times what it should be, which would correspond to a core clock of 400MHz, not the more usual 250MHz. This may be as a result of the recent change to init_turbo - @popcornmix can tell you more.

Putting core_freq=250 should fix it for now, but that may have implications for the Pi 4 which I see you give a different divisor.

petemoore, 2025-02-14:
Many thanks Phil!

I've just finished bisecting and discovered it was introduced in 4f43ba2f54dc6afddac54971a44f8ebb0c2e74d4, which leads me to think it might be the change to default value of `initial_turbo`.... I'm just testing now to see if setting it to 0 in `config.txt` resolves the issue... Unfortunately the tftp booting on my rpi3B+ is a little bit flaky right now which is hindering the process, but hoping to report back shortly. If that doesn't work, I'll also try the `core_freq` fix you suggested. Many thanks!

petemoore, 2025-02-14:
I'm happy to report setting `initial_turbo=0` using 8a8adaf2f1d6836facdc1c66f4e838b2025ec851 (as well as the very latest firmware 7db5aa85c9644cf3c41469a578e320ae2c550178) resolves the issue. :-)

Many thanks!

closed 2025-02-14 by petemoore

popcornmix, 2025-02-14:
@pelwell is this an issue on RPiOS?

Just tried on a Pi02W with `enable_uart=1`, and core_freq is locked at 400MHz, and uart is fine.
I guess without enable_uart=1, the core frequency won't get pinned, but then I see no uart output anyway.

pelwell, 2025-02-14:
No - the problem here is that it is bare metal code that hardcodes the divisor.