raspberrypi/firmware #1946: [request] Remove 1.0 volt PMIC firmware limitation for Raspberry pi 5

joyel24, 2025-02-15:
RPI5's PMIC is capable of doing 1.1V output to CPU, there is a hack that give us a way to remove this limitation while OS is up.
https://github.com/geerlingguy/pi-overvolt

Problem is, to experiment overclock, we need to reboot after editing /boot/fimware/config.txt

And when it restart, PMIC CPU Max voltage go back to 1.0V and causes boot crashes as the voltage is to low for >~3.2GHz CPU Frequency

joyel24, 2025-02-16:
You can also add a new pamater opposite to initial_turbo.
for example, a way to force to boot at lowest frequecy will fix that issue also and is more wise..

If we can set a very high freq arm_freq and be sure that this freq is not going to be achieved before our system is booted. It'll be perfect

timg236, 2025-02-17:
Closing as NGTF. The voltage settings are designed to avoid damage to the chip.

timg236, 2025-02-17:
You can already set initial_turbo=0

joyel24, 2025-02-17:
@timg236 It doesn't work in this case, what I'm asking for is a way to lock CPU to the minimal frequency.

initial_turbo=0 only disable forced max frequency during startup but doesn't prevent the cpu to go higher than minimal freq.

It seems that event with a powersave linux kernel it's not working as max frequency is probably set by GPU firmware before launching kernel.

pelwell, 2025-02-17:
You don't seem to understand: any support for users trying potentially unsafe configurations could be considered encouragement to do so. It's not going to happen.

joyel24, 2025-02-17:
@pelwell You're joking... https://github.com/raspberrypi/firmware/issues/1876

pelwell, 2025-02-17:
Run too fast -> become unstable.
Too many volts -> become paperweight.

joyel24, 2025-02-17:
I'm not asking overvolt anymore. just a way to lock cpu minimal freq, it's not the same thing at all !

joyel24, 2025-02-17:
@pelwell I'm not asking overvolt anymore. just a way to lock cpu minimal freq, it's not the same thing at all !

popcornmix, 2025-02-17:
> It seems that even with a powersave linux kernel it's not working as max frequency is probably set by GPU eeprom firmware before launching kernel.

With initial_turbo=0, I don't believe the bootloader will increase arm frequency above idle frequency without an explicit request from the arm (typically though the ondemand cpufreq governor).

Make sure you have `CONFIG_CPU_FREQ_DEFAULT_GOV_POWERSAVE=y` in your kernel config.

joyel24, 2025-02-17:
@popcornmix Thanks for this more precised answer :)

I set default governor to powersave during `make menuconfig` so it is supposed to be the case. I also found an init scripts that it changing governor to ondemand and forced this one to set powersave also.

I'll try with the an old lite raspbian release to see if it changes.

popcornmix, 2025-02-17:
We used to default to powersave governor and switch to ondemand during boot (specifically in `/etc/init.d/raspi-config`), but in recent bookworm images it is just ondemand in kernel by default with no change to governor.

joyel24, 2025-02-17:
Thanks, I did not noticed the changes to init script because it's still present in the actual latest image given with pi imager.