raspberrypi/firmware #1948: 2xCSI + 1xDSI Running issue since kernel 6.1.50-v8+ on CM4-IO-BASE-A / CM4

Trackhe, 2025-02-28:
---
name: 2xCSI + 1xDSI Running issue since kernel 6.1.50-v8+ on CM4-IO-BASE-A
about: I would like to use 2 cameras via CSI and a display via DSI.

---

**Describe the bug**
Different bugs on different versions.
But mainly, no display output (black) after boot log and or not useable csi cams.
I would update to the latest kernel and give all the information you need when i missed something out please tell me.

**To reproduce**
Use CM4-IO-BASE-A or CM4-IO-BASE-B

follow:
CSI-DSI Section
https://www.waveshare.com/wiki/CM4-IO-BASE-B#CSI_DSI
or
https://www.waveshare.com/wiki/CM4-IO-BASE-A#CSI_DSI

connect mal 2
https://www.raspberrypi.com/products/camera-module-v2/ 
+
https://biqu.equipment/de/products/bigtreetech-pi-tft43-v2-0-screen-board?srsltid=AfmBOorUQp1h9p76o24QtknTC1sFe78lm4TA6RS61Ztf9qlumNEMsxO5
v2.0 should be exactly like the rpi tft dsi display work

change config.txt in /boot or /boot/firmware when you are on bookworm to
```
camera_auto_detect=0
display_auto_detect=0
disable_fw_kms_setup=1

#dtparam=i2c_arm=on
#dtparam=spi=on

auto_initramfs=1
max_framebuffers=2

dtoverlay=vc4-kms-v3d,cma-256
dtoverlay=vc4-kms-dsi-7inch

dtoverlay=dwc2,dr_mode=host
dtoverlay=imx219,cam1
dtoverlay=imx219,cam0

arm_64bit=1
arm_boost=1

gpu_mem=256
```

And one of the Kernels (Debian 11 or 12 seems not make the difference)

`6.1.29-v8+` works full
`6.1.32-v8+` works full
`6.1.41-v8+` works full
`6.1.47-v8+` works full
`6.1.50-v8+` one cam dosn't work
`6.1.51-v8+` one cam dosn't work
`6.1.53-v8+` one cam dosn't work
`6.1.54-v8+` one cam dosn't work

`6.6.30-v8+` one cam dosn't work but is shown by  libcamera-hello but without the right crops and fps `[0:01:13.797117432] [1077] ERROR V4L2 v4l2_device.cpp:353 'imx219 0-0010': Unable to set controls: Device or resource busy`
`6.6.39-v8+`  one cam dosn't work but is shown by  libcamera-hello but without the right crops and fps `[0:01:13.797117432] [1077] ERROR V4L2 v4l2_device.cpp:353 'imx219 0-0010': Unable to set controls: Device or resource busy`

`6.6.78-v8+` resolution report fixed, display isnt working anymore, only one cam is working

i guess the only option is to use `6.1.47-v8+` `b2b3c05f6e9944c2e7eab8648a0cde932e25a31e` for fully working
I try it now in Bookworm

Bookworm (Debian 12)
`6.6.51+rpt-rpi-v8` shipped with | both cams seems fully work | display stops working after boot log / no cli
`6.6.74+rpt-rpi-v8` both cams seems fully work | display stops working after boot log / no cli

`6.6.78-v8+` not work with the reporting of one cam but not useable, and no display

**System**
Normal (Working Output), The Display Works but when the cams are working it is not granting that the Display Works.
raspinfo with `Bookworm Debian 12` and the working old Kernel `6.1.47-v8+` https://pastebin.com/nMtPiVAU

6by9, 2025-03-03:
Support for non-Raspberry Pi products (eg Waveshare CM4-IO and third party display) should come from your vendor.
Vendors who cloned the original Pi 7" display have done better or worse jobs over it, and that fact that your vendor says that the display has a GT911 touch controller when the Pi is trying to talk to an FT5406 drops it into the same category as Waveshare's mess in https://github.com/raspberrypi/linux/issues/6538

```
[   11.124087] rpi_touchscreen_attiny 10-0045: Failed to read REG_ID reg: -5
[   11.124121] rpi_touchscreen_attiny: probe of 10-0045 failed with error -5
```
Was resolved via https://github.com/raspberrypi/linux/pull/6382 in Sept 2024.

I will dig out the bits to test dual cameras plus display on the Raspberry Pi CM4IO with the latest kernels, but if that works then there's little I can suggest other than approach your vendors.

6by9, 2025-03-04:
I've just tested a CM4IO with a Pi 7" DSI display, imx708 (v3), and imx462 camera modules (it seems my imx219's have disappeared).
Both cameras and the display work fine on 6.6.78 and 6.12.17.
Additional config.txt entries of 
```
dtoverlay=vc4-kms-dsi-7inch
dtoverlay=imx708,cam0
dtoverlay=imx462,clock-frequency=74250000
```

Rereading your report and the Waveshare wiki, anything that mentions installing a dt-blob.bin file is out of date and erroneous on Bookworm. That is only applicable if you are using either the legacy display or camera stacks, and you're using vc4 display and libcamera for camera.

On a 6.6.78 kernel, post the full output of `dmesg` (use pastebin or code markup), and also the output from `i2cdetect -l` and `sudo pinctrl get`

Trackhe, 2025-03-06:
Actually i m so sorry that i don't closed this ticket.

On the last Kernel, i get this working in Debian 11 but only the reporting of libcamera-hello --list-cam is buggy.
Also the Touch works great.

Thank you for your support. Btw i reported it to Waveshare too but there is no answer at this time. 

"Rereading your report and the Waveshare wiki, anything that mentions installing a dt-blob.bin file is out of date and erroneous on Bookworm." 
I will have this in my mind now thank you for that.

closed 2025-03-06 by Trackhe