raspberrypi/firmware #1950: Device tree support failed after update?  Or some type of odd hardware fault?

Jibun-no-Kage, 2025-03-02:
Not sure if this a bug or there was a design change that seems to have broken device tree support for some sensors?  Or some type of odd hardware fault?  Did a typical update/full-upgrade and lost sensor support apparently if not more.

```
apt update
apt full-upgrade
reboot
```

OS/Kernel information...
```
Linux phobia.dachshund-digital.org 6.12.16-v8+ #1859 SMP PREEMPT Mon Feb 24 13:14:16 GMT 2025 aarch64 GNU/Linux
```
By the way, happened under kernel 6.6.x as well on same hardware, I updated firmware just to see if it fixed the issue, it did not.

My config.txt for example, for bmp280...
```
dtoverlay=i2c-sensor,bmp280,addr=0x76
```
Query of devices on i2c bus, kernel recognization of sensor fails., register address should show UU if kernel recognized sensoer..
```
# i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- 76 --
```
Output from dmesg, given reference to 'hwmon sysfs' thought this might be related, not sure how throttling is applicable to a i2c sensor...
```
[  105.596908] deprecated, use hwmon sysfs instead
[  105.596954] WARNING: CPU: 1 PID: 1569 at drivers/firmware/raspberrypi.c:239 get_throttled_show+0x68/0x78
[  105.596975] Modules linked in: binfmt_misc bmp280_spi bmp280_i2c bmp280 industrialio_triggered_buffer kfifo_buf industrialio regmap_i2c brcmfmac_wcc joydev sr_mod cdrom rpivid_hevc(C) bcm2835_codec(C) bcm2835_v4l2(C) bcm2835_isp(C) sg bcm2835_mmal_vchiq(C) vc4 brcmfmac vc_sm_cma(C) v4l2_mem2mem videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops snd_soc_hdmi_codec brcmutil drm_display_helper cec videobuf2_v4l2 cfg80211 drm_dma_helper snd_soc_core v3d videodev snd_bcm2835(C) snd_compress rfkill snd_pcm_dmaengine snd_pcm videobuf2_common gpu_sched mc drm_shmem_helper snd_timer snd drm_kms_helper i2c_bcm2835 i2c_brcmstb raspberrypi_hwmon raspberrypi_gpiomem nvmem_rmem uio_pdrv_genirq uio i2c_dev drm fuse drm_panel_orientation_quirks backlight dm_mod ip_tables x_tables ipv6
[  105.597111] CPU: 1 UID: 0 PID: 1569 Comm: python Tainted: G         C         6.12.16-v8+ #1859
[  105.597119] Tainted: [C]=CRAP
[  105.597122] Hardware name: Raspberry Pi 4 Model B Rev 1.4 (DT)
[  105.597125] pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  105.597130] pc : get_throttled_show+0x68/0x78
[  105.597136] lr : get_throttled_show+0x68/0x78
[  105.597140] sp : ffffffc08484bbf0
[  105.597143] x29: ffffffc08484bbf0 x28: 0000000000000001 x27: 0000000000400cc0
[  105.597152] x26: 000000007ffff000 x25: ffffffa0455e83f8 x24: ffffffa0455e83e8
[  105.597161] x23: 0000000000000000 x22: ffffffa040ed1810 x21: ffffff9f839cf540
[  105.597170] x20: ffffffa044f56200 x19: ffffffa0463f9000 x18: 0000000000000000
[  105.597178] x17: 0000000000000000 x16: 0000000000000000 x15: 0000000000000000
[  105.597186] x14: 0000000000000000 x13: 64616574736e6920 x12: 7366737973206e6f
[  105.597194] x11: 6d77682065737520 x10: ffffffd629de5380 x9 : ffffffd62871bf28
[  105.597202] x8 : 00000000ffffefff x7 : ffffffd629de5380 x6 : 0000000000000000
[  105.597211] x5 : ffffffa043f993c8 x4 : 0000000000000000 x3 : 0000000000000027
[  105.597219] x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffffffa041f58000
[  105.597227] Call trace:
[  105.597231]  get_throttled_show+0x68/0x78
[  105.597236]  dev_attr_show+0x28/0x80
[  105.597244]  sysfs_kf_seq_show+0xb4/0x138
[  105.597252]  kernfs_seq_show+0x34/0x48
[  105.597257]  seq_read_iter+0x1bc/0x4b8
[  105.597262]  kernfs_fop_read_iter+0x148/0x1c8
[  105.597267]  vfs_read+0x268/0x318
[  105.597274]  ksys_read+0x70/0x108
[  105.597280]  __arm64_sys_read+0x24/0x38
[  105.597286]  invoke_syscall+0x50/0x120
[  105.597294]  el0_svc_common.constprop.0+0x48/0xf8
[  105.597300]  do_el0_svc+0x28/0x40
[  105.597306]  el0_svc+0x30/0xd0
[  105.597314]  el0t_64_sync_handler+0x13c/0x158
[  105.597318]  el0t_64_sync+0x190/0x198
[  105.597323] ---[ end trace 0000000000000000 ]---
```

Jibun-no-Kage, 2025-03-02:
Pi seems to be running a warmer than I recall... but not throttled?
```
# vcgencmd measure_temp
temp=75.4'C
#  vcgencmd get_throttled
throttled=0x0
root@phobia:~#
```

Jibun-no-Kage, 2025-03-02:
OK... Did a couple of reboots... and the sensor discovery issue disappeared so closing this... But the get throttled issue seems to persist.  So not related to the original sensor issue I had but now seems to have disappeared..

closed 2025-03-02 by Jibun-no-Kage

popcornmix, 2025-03-03:
The "get_throttled" warning is deliberate. See https://github.com/raspberrypi/linux/pull/2706/commits/7401d7d96c8c7b7b48eb15ab21e234ab62597f6f

It is harmless it just warns that you are using the deprecated interface to query get_throttled. 
I can provoke that with:
```cat /sys/devices/platform/soc@107c000000/soc@107c000000:firmware/get_throttled```
and the fix is to not do that (we should probably remove the sysfs file for newer kernels.