raspberrypi/firmware #1953: rpi4 does not boot from SD card with firmware upgraded to 20250305

arrjay, 2025-03-14:
---
name: rpi4 boot failure using SD card with firmware 20250305

---

**Describe the bug**
After upgrading raspios to use raspi-firmware 20250305 the system no longer boots. it comes up to a rainbow screen and the SD access card light blinks in "unable to find kernel" pattern.

**To reproduce**
Upgrade to raspi-firmware package 1:1.20250305-1 - tested on a system that uses SD cards as rootfs.

**Expected behaviour**
the system loads a kernel and initrd

**Actual behaviour**
the system is not sure what an SD card is, fails to boot.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Raspberry Pi 4 Model B 2018
* Which OS and version (`cat /etc/rpi-issue`)?
Raspbian GNU/Linux 12 (bookworm)
* Which firmware version (`vcgencmd version`)?
I don't know when the system doesn't boot.
* Which kernel version (`uname -a`)?
6.6.74+rpt-rpi-v7l

**Logs**
from adding enable_uart=1 and uart_2ndstage=1 I was able to get the following logs on boot failure:
```
Read start4.elf bytes  2263328 hnd 0x51b7
Read fixup4.dat bytes     5459 hnd 0x503f
0x00b03112 0x00000000 0x00001fff
MEM GPU: 32 ARM: 992 TOTAL: 1024
Starting start4.elf @ 0xfec00200 partition 0
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 18
+

MESS:00:00:04.187860:0: arasan: arasan_emmc_open
MESS:00:00:04.189520:0: arasan: arasan_emmc_set_clock C0: 0x00800000 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 400000 max: 400000 delay: 5
MESS:00:00:04.309753:0: arasan: arasan_emmc_set_clock C0: 0x00800000 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 400000 max: 400000 delay: 5
MESS:00:00:04.322623:0: arasan: arasan_emmc_set_clock C0: 0x00800f00 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 390000 max: 400000 delay: 5
MESS:00:00:04.360402:0: arasan: arasan_emmc_set_clock C0: 0x00800f06 C1: 0x000e0207 emmc: 200000000 actual: 50000000 div: 0x00000002 target: 50000000 min: 0 max: 50000000 delay: 1
MESS:00:00:04.477992:0: arasan: arasan_emmc_set_clock C0: 0x00800000 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 400000 max: 400000 delay: 5
MESS:00:00:04.490874:0: arasan: arasan_emmc_set_clock C0: 0x00800f00 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 390000 max: 400000 delay: 5
MESS:00:00:04.528230:0: arasan: arasan_emmc_set_clock C0: 0x00800f06 C1: 0x000e0207 emmc: 200000000 actual: 50000000 div: 0x00000002 target: 50000000 min: 0 max: 50000000 delay: 1
MESS:00:00:04.541835:0: gpioman: gpioman_get_pin_num: attempting to get pin number for SDCARD_CONTROL_POWER before we're ready
MESS:00:00:04.652243:0: gpioman: gpioman_get_pin_num: attempting to get pin number for SDCARD_CONTROL_POWER before we're ready
MESS:00:00:04.664675:0: arasan: arasan_emmc_set_clock C0: 0x00800000 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 400000 max: 400000 delay: 5
MESS:00:00:04.677540:0: arasan: arasan_emmc_set_clock C0: 0x00800f00 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 390000 max: 400000 delay: 5
MESS:00:00:04.708175:0: gpioman: gpioman_get_pin_num: attempting to get pin number for SDCARD_CONTROL_WRITE_PROTECTED before we're ready
MESS:00:00:04.724040:0: arasan: arasan_emmc_set_clock C0: 0x00800f06 C1: 0x000e0207 emmc: 200000000 actual: 50000000 div: 0x00000002 target: 50000000 min: 0 max: 50000000 delay: 1
MESS:00:00:04.737387:0: gpioman: gpioman_get_pin_num: attempting to get pin number for SDCARD_CONTROL_POWER before we're ready
MESS:00:00:04.828910:0: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:04.833420:0: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:04.838655:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:04.946485:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:04.955791:0: arasan: arasan_emmc_set_clock C0: 0x00800000 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 400000 max: 400000 delay: 5
MESS:00:00:04.968655:0: arasan: arasan_emmc_set_clock C0: 0x00800f00 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 390000 max: 400000 delay: 5
MESS:00:00:04.999297:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_WRITE_PROTECTED not defined
MESS:00:00:05.012038:0: arasan: arasan_emmc_set_clock C0: 0x00800f06 C1: 0x000e0207 emmc: 200000000 actual: 50000000 div: 0x00000002 target: 50000000 min: 0 max: 50000000 delay: 1
MESS:00:00:05.025391:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:05.032987:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:05.140898:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:05.150209:0: arasan: arasan_emmc_set_clock C0: 0x00800000 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 400000 max: 400000 delay: 5
MESS:00:00:05.163091:0: arasan: arasan_emmc_set_clock C0: 0x00800f00 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 390000 max: 400000 delay: 5
[... 102 lines cut ...]
MESS:00:00:08.325641:0: arasan: arasan_emmc_set_clock C0: 0x00800f00 C1: 0x000e8007 emmc: 100000000 actual: 390625 div: 0x00000080 target: 400000 min: 390000 max: 400000 delay: 5
MESS:00:00:08.356341:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_WRITE_PROTECTED not defined
MESS:00:00:08.369333:0: arasan: arasan_emmc_set_clock C0: 0x00800f06 C1: 0x000e0107 emmc: 100000000 actual: 50000000 div: 0x00000001 target: 50000000 min: 0 max: 50000000 delay: 1
MESS:00:00:08.382804:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:08.390253:0: dterror: Failed to load Device Tree file '?'
MESS:00:00:08.396433:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:08.504318:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:08.513739:0: arasan: arasan_emmc_set_clock C0: 0x00800000 C1: 0x000e8007 emmc: 100000000 actual: 390625 div: 0x00000080 target: 400000 min: 400000 max: 400000 delay: 5
MESS:00:00:08.526720:0: arasan: arasan_emmc_set_clock C0: 0x00800f00 C1: 0x000e8007 emmc: 100000000 actual: 390625 div: 0x00000080 target: 400000 min: 390000 max: 400000 delay: 5
MESS:00:00:08.557423:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_WRITE_PROTECTED not defined
MESS:00:00:08.570418:0: arasan: arasan_emmc_set_clock C0: 0x00800f06 C1: 0x000e0107 emmc: 100000000 actual: 50000000 div: 0x00000001 target: 50000000 min: 0 max: 50000000 delay: 1
MESS:00:00:08.583918:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:08.591356:0: Failed to open command line file 'cmdline.txt'
MESS:00:00:08.598247:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:08.705604:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:08.715013:0: arasan: arasan_emmc_set_clock C0: 0x00800000 C1: 0x000e8007 emmc: 100000000 actual: 390625 div: 0x00000080 target: 400000 min: 400000 max: 400000 delay: 5
MESS:00:00:08.727972:0: arasan: arasan_emmc_set_clock C0: 0x00800f00 C1: 0x000e8007 emmc: 100000000 actual: 390625 div: 0x00000080 target: 400000 min: 390000 max: 400000 delay: 5
MESS:00:00:08.758672:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_WRITE_PROTECTED not defined
MESS:00:00:08.771652:0: arasan: arasan_emmc_set_clock C0: 0x00800f06 C1: 0x000e0107 emmc: 100000000 actual: 50000000 div: 0x00000001 target: 50000000 min: 0 max: 50000000 delay: 1
MESS:00:00:08.785123:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:08.792582:0: No compatible kernel found
```

**Additional context**
Add any other relevant context for the problem.

timg236, 2025-03-14:
You haven't added any information about the OS or bootloader release

Try upgrading to the latest bootloader
```
sudo apt update
sudo apt full-upgrade
sudo rpi-eeprom-update -a
```

arrjay, 2025-03-14:
that's going to make it _real hard_ to boot newly built os releases on old hardware, isn't it?

but sure, give me a second and I'll check that too.

arrjay, 2025-03-14:
fair, let's do that. this is a SD card with raspios from debootstrap, and, as I had noted, was booting with previous firmware version.

partition table (GPT)
```
# sfdisk -l /dev/mmcblk0
Disk /dev/mmcblk0: 14.56 GiB, 15634268160 bytes, 30535680 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: A6F46730-9A78-4F80-91FC-AC448C1E76A7

Device           Start      End  Sectors  Size Type
/dev/mmcblk0p1    4096   552959   548864  268M Linux filesystem
/dev/mmcblk0p2  557056  1060863   503808  246M Linux filesystem
/dev/mmcblk0p3 1064960  1073151     8192    4M Linux filesystem
/dev/mmcblk0p4 1077248  1175551    98304   48M Linux filesystem
/dev/mmcblk0p5 1179648 30531584 29351937   14G Linux filesystem
```

partition table (MBR)
```
Disk /dev/mmcblk0: 14.56 GiB, 15634268160 bytes, 30535680 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x00000000

Device         Boot   Start      End  Sectors  Size Id Type
/dev/mmcblk0p1         4096   552959   548864  268M  e W95 FAT16 (LBA)
/dev/mmcblk0p2       557056  1060863   503808  246M 83 Linux
/dev/mmcblk0p3            1     4095     4095    2M ee GPT
/dev/mmcblk0p4      1060864 30535679 29474816 14.1G ee GPT

Partition table entries are not in disk order.
```

contents of /boot/firmware:
```
# find /boot/firmware/ -type f
/boot/firmware/bcm2708-rpi-b-plus.dtb
/boot/firmware/bcm2708-rpi-b-rev1.dtb
/boot/firmware/bcm2708-rpi-b.dtb
/boot/firmware/bcm2708-rpi-cm.dtb
/boot/firmware/bcm2708-rpi-zero-w.dtb
/boot/firmware/bcm2708-rpi-zero.dtb
/boot/firmware/bcm2709-rpi-2-b.dtb
/boot/firmware/bcm2709-rpi-cm2.dtb
/boot/firmware/bcm2710-rpi-2-b.dtb
/boot/firmware/bcm2710-rpi-3-b-plus.dtb
/boot/firmware/bcm2710-rpi-3-b.dtb
/boot/firmware/bcm2710-rpi-cm3.dtb
/boot/firmware/bcm2710-rpi-zero-2-w.dtb
/boot/firmware/bcm2710-rpi-zero-2.dtb
/boot/firmware/bcm2711-rpi-4-b.dtb
/boot/firmware/bcm2711-rpi-400.dtb
/boot/firmware/bcm2711-rpi-cm4-io.dtb
/boot/firmware/bcm2711-rpi-cm4.dtb
/boot/firmware/bcm2711-rpi-cm4s.dtb
/boot/firmware/bcm2835-rpi-a-plus.dtb
/boot/firmware/bcm2835-rpi-a.dtb
/boot/firmware/bcm2835-rpi-b-plus.dtb
/boot/firmware/bcm2835-rpi-b-rev2.dtb
/boot/firmware/bcm2835-rpi-b.dtb
/boot/firmware/bcm2835-rpi-cm1-io1.dtb
/boot/firmware/bcm2835-rpi-zero-w.dtb
/boot/firmware/bcm2835-rpi-zero.dtb
/boot/firmware/bcm2836-rpi-2-b.dtb
/boot/firmware/bcm2837-rpi-3-a-plus.dtb
/boot/firmware/bcm2837-rpi-3-b-plus.dtb
/boot/firmware/bcm2837-rpi-3-b.dtb
/boot/firmware/bcm2837-rpi-cm3-io3.dtb
/boot/firmware/bcm2837-rpi-zero-2-w.dtb
/boot/firmware/bootcode.bin
/boot/firmware/cmdline.txt
/boot/firmware/config.txt
/boot/firmware/fixup.dat
/boot/firmware/fixup4.dat
/boot/firmware/fixup4cd.dat
/boot/firmware/fixup4db.dat
/boot/firmware/fixup4x.dat
/boot/firmware/fixup_cd.dat
/boot/firmware/fixup_db.dat
/boot/firmware/fixup_x.dat
/boot/firmware/initramfs
/boot/firmware/initramfs7
/boot/firmware/initramfs7l
/boot/firmware/kernel.img
/boot/firmware/kernel7.img
/boot/firmware/kernel7l.img
/boot/firmware/overlays/pijuice.dtbo
/boot/firmware/start.elf
/boot/firmware/start4.elf
/boot/firmware/start4cd.elf
/boot/firmware/start4db.elf
/boot/firmware/start4x.elf
/boot/firmware/start_cd.elf
/boot/firmware/start_db.elf
/boot/firmware/start_x.elf
```

current /boot/firmware/config.txt
```
# cat /boot/firmware/config.txt 
enable_uart=1
dtoverlay=vc4-kms-v3d
gpu_mem=16
[pi1]
kernel kernel.img
initramfs initramfs followkernel
[pi2]
kernel kernel7.img
initramfs initramfs7 followkernel
[pi3]
kernel kernel7.img
initramfs initramfs7 followkernel
[pi4]
kernel kernel7l.img
initramfs initramfs7l followkernel
[pi0]
kernel kernel.img
initramfs initramfs followkernel
[pi02]
kernel kernel7.img
initramfs initramfs7 followkernel
[all]
```

arrjay, 2025-03-14:
rpi-eeprom does show an update, so I will try that now. this was running `Mon Apr 15 13:12:14 UTC 2024 (1713186734)
`

arrjay, 2025-03-14:
rebooted into new firmware and `rpi-eeprom-update` shows it is current `Tue Feb 11 17:00:13 UTC 2025 (1739293213)`
modified `config.txt` before reboot to set gpu_mem to 32, add uart_2ndstage=1
dropped apt preferences that were pinning old firmware and updated. using full-upgrade

rebooted and I'm still not resolving a kernel, though the new logs are _slightly_ different:

```
  3.33 [sdcard] vl805.bin not found
  3.34 [sdcard] pieeprom.upd not found
  3.37 [sdcard] recover4.elf not found
  3.41 [sdcard] recovery.elf not found
  3.71 Read start4.elf bytes  2263328 hnd 0x4709
  3.74 Read fixup4.dat bytes     5459 hnd 0x4591
  3.77 0x00b03112 0x00000000 0x00001fff
  3.80 MEM GPU: 32 ARM: 992 TOTAL: 1024
  4.14 Starting start4.elf @ 0xfec00200 partition 1
  4.16 XHCI-STOP
  4.17 xHC0 ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
  4.23 USBSTS 18
  4.27 PCI0 reset
  4.37 +

MESS:00:00:04.138034:0: arasan: arasan_emmc_open
MESS:00:00:04.139692:0: arasan: arasan_emmc_set_clock C0: 0x00800000 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 400000 max: 400000 delay: 5
MESS:00:00:04.259935:0: arasan: arasan_emmc_set_clock C0: 0x00800000 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 400000 max: 400000 delay: 5
MESS:00:00:04.272805:0: arasan: arasan_emmc_set_clock C0: 0x00800f00 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 390000 max: 400000 delay: 5
MESS:00:00:04.310576:0: arasan: arasan_emmc_set_clock C0: 0x00800f06 C1: 0x000e0207 emmc: 200000000 actual: 50000000 div: 0x00000002 target: 50000000 min: 0 max: 50000000 delay: 1
MESS:00:00:04.428161:0: arasan: arasan_emmc_set_clock C0: 0x00800000 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 400000 max: 400000 delay: 5
MESS:00:00:04.441043:0: arasan: arasan_emmc_set_clock C0: 0x00800f00 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 390000 max: 400000 delay: 5
MESS:00:00:04.478806:0: arasan: arasan_emmc_set_clock C0: 0x00800f06 C1: 0x000e0207 emmc: 200000000 actual: 50000000 div: 0x00000002 target: 50000000 min: 0 max: 50000000 delay: 1
MESS:00:00:04.492411:0: gpioman: gpioman_get_pin_num: attempting to get pin number for SDCARD_CONTROL_POWER before we're ready
MESS:00:00:04.602819:0: gpioman: gpioman_get_pin_num: attempting to get pin number for SDCARD_CONTROL_POWER before we're ready
MESS:00:00:04.615251:0: arasan: arasan_emmc_set_clock C0: 0x00800000 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 400000 max: 400000 delay: 5
MESS:00:00:04.628115:0: arasan: arasan_emmc_set_clock C0: 0x00800f00 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 390000 max: 400000 delay: 5
MESS:00:00:04.658756:0: gpioman: gpioman_get_pin_num: attempting to get pin number for SDCARD_CONTROL_WRITE_PROTECTED before we're ready
MESS:00:00:04.675022:0: arasan: arasan_emmc_set_clock C0: 0x00800f06 C1: 0x000e0207 emmc: 200000000 actual: 50000000 div: 0x00000002 target: 50000000 min: 0 max: 50000000 delay: 1
MESS:00:00:04.688365:0: gpioman: gpioman_get_pin_num: attempting to get pin number for SDCARD_CONTROL_POWER before we're ready
MESS:00:00:04.780102:0: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:04.784611:0: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:04.789845:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:04.897676:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:04.906981:0: arasan: arasan_emmc_set_clock C0: 0x00800000 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 400000 max: 400000 delay: 5
MESS:00:00:04.919845:0: arasan: arasan_emmc_set_clock C0: 0x00800f00 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 390000 max: 400000 delay: 5
MESS:00:00:04.950488:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_WRITE_PROTECTED not defined
MESS:00:00:04.963630:0: arasan: arasan_emmc_set_clock C0: 0x00800f06 C1: 0x000e0207 emmc: 200000000 actual: 50000000 div: 0x00000002 target: 50000000 min: 0 max: 50000000 delay: 1
MESS:00:00:04.976972:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:04.984572:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:05.092485:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:05.101794:0: arasan: arasan_emmc_set_clock C0: 0x00800000 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 400000 max: 400000 delay: 5
MESS:00:00:05.114675:0: arasan: arasan_emmc_set_clock C0: 0x00800f00 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 390000 max: 400000 delay: 5
MESS:00:00:05.145311:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_WRITE_PROTECTED not defined
MESS:00:00:05.158461:0: arasan: arasan_emmc_set_clock C0: 0x00800f06 C1: 0x000e0207 emmc: 200000000 actual: 50000000 div: 0x00000002 target: 50000000 min: 0 max: 50000000 delay: 1
MESS:00:00:05.171821:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:05.661520:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
MESS:00:00:05.667070:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
MESS:00:00:05.676104:0: *** Restart logging
[... cut ...]
MESS:00:00:08.726153:0: arasan: arasan_emmc_set_clock C0: 0x00800f06 C1: 0x000e0107 emmc: 100000000 actual: 50000000 div: 0x00000001 target: 50000000 min: 0 max: 50000000 delay: 1
MESS:00:00:08.739619:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:08.747082:0: No compatible kernel found
```

the `Failed to load Device Tree file '?'` is certainly one outstanding here. I'm somewhat confused though as `dtb_file 'bcm2711-rpi-4-b.dtb'` appears earlier in this same message set.

arrjay, 2025-03-14:
that one works! here's the serial log:

```
  4.07 [sdcard] vl805.bin not found
  4.08 [sdcard] pieeprom.upd not found
  4.12 [sdcard] recover4.elf not found
  4.15 [sdcard] recovery.elf not found
  4.46 Read start4.elf bytes  2263776 hnd 0x8a
  4.50 Read fixup4.dat bytes     5459 hnd 0x89
  4.52 0x00b03112 0x00000000 0x00001fff
  4.55 MEM GPU: 32 ARM: 992 TOTAL: 1024
  4.89 Starting start4.elf @ 0xfec00200 partition 1
  4.91 XHCI-STOP
  4.92 xHC0 ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
  4.98 USBSTS 18
  4.02 PCI0 reset
  4.12 +

MESS:00:00:04.313087:0: arasan: arasan_emmc_open
MESS:00:00:04.314748:0: arasan: arasan_emmc_set_clock C0: 0x00800000 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 400000 max: 400000 delay: 5
MESS:00:00:04.434988:0: arasan: arasan_emmc_set_clock C0: 0x00800000 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 400000 max: 400000 delay: 5
MESS:00:00:04.447861:0: arasan: arasan_emmc_set_clock C0: 0x00800f00 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 390000 max: 400000 delay: 5
MESS:00:00:04.481784:0: boot-part: 1 fs-type: 0
MESS:00:00:04.483197:0: boot-part: 1 fs-type: 2
MESS:00:00:04.487576:0: arasan: arasan_emmc_set_clock C0: 0x00800f06 C1: 0x000e0207 emmc: 200000000 actual: 50000000 div: 0x00000002 target: 50000000 min: 0 max: 50000000 delay: 1
MESS:00:00:04.532343:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:04.535351:0: brfs: File read: 422 bytes
MESS:00:00:04.622423:0: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:04.626924:0: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:04.646140:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:05.097984:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
MESS:00:00:05.103535:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
MESS:00:00:05.112566:0: *** Restart logging
MESS:00:00:05.114342:0: brfs: File read: 422 bytes
MESS:00:00:05.185094:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:05.190126:0: hdmi: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:05.207263:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:05.212338:0: hdmi: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:05.217937:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:05.226695:0: HDMI0: hdmi_pixel_encoding: 300000000
MESS:00:00:05.232187:0: HDMI1: hdmi_pixel_encoding: 300000000
MESS:00:00:12.111758:0: brfs: File read: /mfs/sd/initramfs7l
MESS:00:00:12.114353:0: Loaded 'initramfs7l' to 0x0 size 0x2027f23
MESS:00:00:12.176055:0: initramfs loaded to 0x2cfd8000 (size 0x2027f23)
MESS:00:00:12.189366:0: dtb_file 'bcm2711-rpi-4-b.dtb'
MESS:00:00:12.191442:0: brfs: File read: 33718051 bytes
MESS:00:00:12.211087:0: brfs: File read: /mfs/sd/bcm2711-rpi-4-b.dtb
MESS:00:00:12.214376:0: Loaded 'bcm2711-rpi-4-b.dtb' to 0x100 size 0xdb2c
MESS:00:00:12.255251:0: brfs: File read: 56108 bytes
MESS:00:00:12.306083:0: brfs: File read: /mfs/sd/overlays/overlay_map.dtb
MESS:00:00:12.371772:0: brfs: File read: 5451 bytes
MESS:00:00:12.381208:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:12.386256:0: brfs: File read: 422 bytes
MESS:00:00:12.466355:0: brfs: File read: /mfs/sd/overlays/vc4-kms-v3d-pi4.dtbo
MESS:00:00:12.626502:0: Loaded overlay 'vc4-kms-v3d-pi4'
MESS:00:00:13.039471:0: brfs: File read: 3913 bytes
MESS:00:00:13.048683:0: brfs: File read: /mfs/sd/cmdline.txt
MESS:00:00:13.051313:0: Read command line from file 'cmdline.txt':
MESS:00:00:13.057174:0: 'rd.lockslot=UUID=D138-81E0:AUTO rd.growpart=1 root=LABEL=a8e239c930c581be console=tty0  quiet  splash'
MESS:00:00:13.249730:0: brfs: File read: 103 bytes
MESS:00:00:13.929013:0: brfs: File read: /mfs/sd/kernel7l.img
MESS:00:00:13.931655:0: Loaded 'kernel7l.img' to 0x8000 size 0x785ab0
MESS:00:00:13.937833:0: Device tree loaded to 0x2cfc9e00 (size 0xe11f)
MESS:00:00:13.945910:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:13.953141:0: uart: Baud rate change done...
MESS:00:00:13.955159:0:
```

arrjay, 2025-03-14:
I also have pi1, 0w, 2 and 3 I can try any firmwares with in this partition scenario if you like. I'm assuming the fixup4/start4 is really just the pi4 loader

timg236, 2025-03-14:
Thanks for testing the patch. The fix is merged to the internal firmware branch and will appear in the next rpi-update firmware release

srcshelton, 2025-03-16:
Same on RPi2 (900MHz model) - after changing nothing other than the fixup*, start* binaries to the 1.20250305 release, the system goes from booting correctly to a 7-flash sequence with a rainbow display.

timg236, 2025-03-17:
Fix is now in the rpi-update release

closed 2025-07-05 by arrjay