raspberrypi/firmware #1963: apt full-upgrade fails - [1.20250326] - /etc/kernel/postinst.d/z50-raspi-firmware: 78: rsync: not found

Gyakusetsu, 2025-04-09:
**Describe the bug**
Updating from kernel 6.6.51 to latest 6.12.20
And installing rsync solves it, I wonder why rsync is needed at post kernel install?

**To reproduce**
```bash
$ sudo apt update
$ sudo apt upgrade # full-upgrade
```

**Expected behaviour**
No errors

**Actual behaviour**
Errors

This is the info after installing rsync `sudo apt install rsync`:

**System**
[raspinfo.txt](https://github.com/user-attachments/files/19662080/raspinfo.txt)

* Which model of Raspberry Pi? 
Raspberry Pi 5
* Which OS and version (`cat /etc/rpi-issue`)? 
Raspberry Pi reference 2024-11-19
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 891df1e21ed2b6099a2e6a13e26c91dea44b34d4, stage2
* Which firmware version (`vcgencmd version`)?
2025/03/10 17:10:37
Copyright (c) 2012 Broadcom
version 2bb2ae64 (release) (embedded)
* Which kernel version (`uname -a`)?
Linux raspberrypi 6.12.18-v8-16k+ #1862 SMP PREEMPT Wed Mar 12 12:33:09 GMT 2025 aarch64 GNU/Linux

**Logs**
```bash
sudo apt upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following NEW packages will be installed:
  linux-headers-6.12.20+rpt-common-rpi linux-headers-6.12.20+rpt-rpi-2712 linux-headers-6.12.20+rpt-rpi-v8
  linux-image-6.12.20+rpt-rpi-2712 linux-image-6.12.20+rpt-rpi-v8 linux-kbuild-6.12.20+rpt
The following packages will be upgraded:
  base-files curl dns-root-data ghostscript libc-bin libc-dev-bin libc-devtools libc-l10n libc6 libc6-dbg libc6-dev
  libcurl3-gnutls libcurl4 libfreetype6 libgs-common libgs10 libgs10-common liblzma5 libnm0 libnuma1 libpam-systemd
  libpixman-1-0 libsystemd-shared libsystemd0 libudev1 libxslt1.1 linux-headers-rpi-2712 linux-headers-rpi-v8
  linux-image-rpi-2712 linux-image-rpi-v8 linux-libc-dev locales network-manager raspi-config raspi-utils
  raspi-utils-core raspi-utils-dt raspi-utils-eeprom raspi-utils-otp raspinfo rpi-eeprom systemd systemd-sysv tzdata
  udev vim-common vim-tiny wget xz-utils
49 upgraded, 6 newly installed, 0 to remove and 0 not upgraded.
Need to get 124 MB of archives.
After this operation, 110 MB of additional disk space will be used.
Do you want to continue? [Y/n]
Get:1 http://deb.debian.org/debian bookworm/main arm64 base-files arm64 12.4+deb12u10 [70.9 kB]
Get:2 http://deb.debian.org/debian bookworm/main arm64 systemd-sysv arm64 252.36-1~deb12u1 [42.6 kB]
Get:3 http://deb.debian.org/debian bookworm/main arm64 libpam-systemd arm64 252.36-1~deb12u1 [209 kB]
Get:4 http://deb.debian.org/debian-security bookworm-security/main arm64 liblzma5 arm64 5.4.1-1 [199 kB]
Get:5 http://deb.debian.org/debian bookworm/main arm64 systemd arm64 252.36-1~deb12u1 [2,837 kB]
Get:6 http://deb.debian.org/debian bookworm/main arm64 libsystemd-shared arm64 252.36-1~deb12u1 [1,511 kB]
Get:7 http://deb.debian.org/debian bookworm/main arm64 libsystemd0 arm64 252.36-1~deb12u1 [308 kB]
Get:8 http://deb.debian.org/debian bookworm/main arm64 udev arm64 252.36-1~deb12u1 [1,640 kB]
Get:9 http://deb.debian.org/debian bookworm/main arm64 libudev1 arm64 252.36-1~deb12u1 [103 kB]
Get:10 http://deb.debian.org/debian bookworm-updates/main arm64 tzdata all 2025b-0+deb12u1 [299 kB]
Get:11 http://deb.debian.org/debian bookworm/main arm64 vim-tiny arm64 2:9.0.1378-2+deb12u2 [652 kB]
Get:12 http://deb.debian.org/debian bookworm/main arm64 vim-common all 2:9.0.1378-2+deb12u2 [125 kB]
Get:13 http://deb.debian.org/debian bookworm/main arm64 wget arm64 1.21.3-1+deb12u1 [922 kB]
Get:14 http://deb.debian.org/debian-security bookworm-security/main arm64 xz-utils arm64 5.4.1-1 [469 kB]
Get:15 http://deb.debian.org/debian bookworm/main arm64 curl arm64 7.88.1-10+deb12u12 [309 kB]
Get:16 http://deb.debian.org/debian bookworm/main arm64 libcurl4 arm64 7.88.1-10+deb12u12 [367 kB]
[... 278 lines cut ...]
Setting up raspinfo (20250314-1) ...
Processing triggers for libc-bin (2.36-9+rpt2+deb12u10) ...
Processing triggers for man-db (2.11.2-2) ...
Processing triggers for dbus (1.14.10-1~deb12u1) ...
Processing triggers for ntpsec (1.2.2+dfsg1-1+deb12u1) ...
Processing triggers for initramfs-tools (0.142+rpt3+deb12u1) ...
update-initramfs: Generating /boot/initrd.img-6.12.20+rpt-rpi-v8
'/boot/initrd.img-6.12.20+rpt-rpi-v8' -> '/boot/firmware/initramfs8'
update-initramfs: Generating /boot/initrd.img-6.12.20+rpt-rpi-2712
'/boot/initrd.img-6.12.20+rpt-rpi-2712' -> '/boot/firmware/initramfs_2712'
Processing triggers for hicolor-icon-theme (0.17-2) ...
Errors were encountered while processing:
 linux-image-6.12.20+rpt-rpi-2712
 linux-image-rpi-2712
 linux-image-6.12.20+rpt-rpi-v8
 linux-headers-6.12.20+rpt-rpi-2712
 linux-headers-6.12.20+rpt-rpi-v8
 linux-headers-rpi-2712
 linux-image-rpi-v8
 linux-headers-rpi-v8
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

**Additional context**
Add any other relevant context for the problem.

XECDesign, 2025-04-10:
What's the output of `apt policy raspi-firmware; apt depends raspi-firmware`?

rei-vilo, 2025-04-12:
```
$ apt policy raspi-firmware
raspi-firmware:
  Installed: 1:1.20250326-1
  Candidate: 1:1.20250326-1
  Version table:
 *** 1:1.20250326-1 500
        500 http://archive.raspberrypi.com/debian bookworm/main arm64 Packages
        500 http://archive.raspberrypi.com/debian bookworm/main armhf Packages
        100 /var/lib/dpkg/status
     1.20220830+ds-1 500
        500 http://deb.debian.org/debian bookworm/non-free-firmware arm64 Packages
        500 http://deb.debian.org/debian bookworm/non-free-firmware armhf Packages
```

and

```
$ apt depends raspi-firmware
raspi-firmware
  PreDepends: dpkg (>= 1.17.14)
    dpkg:armhf
  Depends: dosfstools
  Depends: rsync
    rsync:armhf
  Breaks: <linux-firmware-raspi3> (<< 1.20161123-2)
  Breaks: <raspi3-firmware> (<< 1.20190718-1~)
  Suggests: firmware-brcm80211
  Suggests: firmware-misc-nonfree
  Suggests: bluez-firmware
  Replaces: <linux-firmware-raspi3> (<< 1.20161123-2)
  Replaces: <raspi3-firmware> (<< 1.20190718-1~)
```

XECDesign, 2025-04-14:
Okay, you seem to have the right version of raspi-firmware which lists rsync as a dependency. I'm not sure how you could've ended up without rsync.

Gyakusetsu, 2025-04-14:
@XECDesign, note: this is after I already installed rsync and rerun the `apt full-upgrade`

```bash
raspi-firmware:
  Installed: (none)
  Candidate: 1:1.20250326-1
  Version table:
     1:1.20250326-1 500
        500 http://archive.raspberrypi.com/debian bookworm/main arm64 Packages
        500 http://archive.raspberrypi.com/debian bookworm/main armhf Packages
     1:1.20241126-1 -1
        100 /var/lib/dpkg/status
     1.20220830+ds-1 500
        500 http://deb.debian.org/debian bookworm/non-free-firmware arm64 Packages
        500 http://deb.debian.org/debian bookworm/non-free-firmware armhf Packages
raspi-firmware
  PreDepends: dpkg (>= 1.17.14)
    dpkg:armhf
  Depends: dosfstools
  Depends: rsync
    rsync:armhf
  Breaks: <linux-firmware-raspi3> (<< 1.20161123-2)
  Breaks: <raspi3-firmware> (<< 1.20190718-1~)
  Suggests: firmware-brcm80211
  Suggests: firmware-misc-nonfree
  Suggests: bluez-firmware
  Replaces: <linux-firmware-raspi3> (<< 1.20161123-2)
  Replaces: <raspi3-firmware> (<< 1.20190718-1~)
```

XECDesign, 2025-04-14:
Ah, just saw that the previous reply wasn't from OP...

@Gyakusetsu  If I had to guess, you removed raspi-firmware from your system (which is required for kernel updates to work) and kept the conffiles. so, the kernel hooks from that package still exist but won't work properly.

rei-vilo, 2025-04-14:
On my side, I did with the Raspberry Pi what most helpdesks ask first: reboot! Surprise, the issues were gone.

However, the Hailo was complaining about a version mismatch between the drive and the application, so I did with the Raspberry Pi what most helpdesks ask in second place: perform a fresh installation.

XECDesign, 2025-04-14:
@rei-vilo I don't think your issue is related to this one.

Kariss83, 2025-04-18:
I got the exact same issue on a fresh install of Raspberry Pi OS Lite (64-BIT) today on an old RPi3, I feel like a lot of people are experimenting the same but I cannot find what's wrong on my side. I'm interested in a solution if you guys ever come up with one.

Let me know if there is any info that might help debugging.

Nordsoft91, 2025-06-08:
Same problem, pi 5

XECDesign, 2025-06-09:
> Same problem, pi 5

Same explanation as before.

I may make the script fail more gracefully in this situation make it warn the user about what happened, but it doesn't address the core issue.