raspberrypi/firmware #1965: initramfs selection broken when kernel=kernel8.img

gitbls, 2025-04-20:
### Describe the bug

Something is broken in initramfs loading with `kernel=kernel8.img` on a Pi5, such that the wrong initramfs is loaded.

### Steps to reproduce the behaviour

* Boot a Pi5 with the normal kernel. Add `break=premount` to /boot/firmware/cmdline.txt
* Reboot and the system will stop in initramfs
* Observe that the output from `lsmod` displays a few modules that have already been loaded
* `exit` initramfs and the system will continue booting
* Edit /boot/firmware/config.txt
  * Add `kernel=kernel8.img`
* Shutdown and reboot into initramfs
* Observe that the output from `lsmod` displays only the listing header and NO modules
* `exit` initramfs and the system will continue booting
* Edit /boot/firmware/config.txt
  * Comment out `auto_initramfs=1`
  * Add `initramfs initramfs8 followkernel`
* Shutdown and reboot init initramfs
* Observe that the output from `lsmod` works again, showing loaded modules

While this is not a problem in the normal use case, trying to use `cryptsetup` in initramfs fails due to the incorrect initramfs being loaded.

See https://gitlab.com/cryptsetup/cryptsetup/-/issues/940 for additional details and comments from the cryptsetup developer, Milan Broz.

### Device (s)

Raspberry Pi 5

### System

https://pastebin.com/UEh3iz2h

pelwell, 2025-04-21:
Thanks for reporting this - it's clearly a bug, and an easy one to fix. https://drive.google.com/file/d/1OxmVXmUHBH5woHupzSoWrubbax0SXfdg/view?usp=sharing contains an EEPROM image with a patch that should fix the problem. Once you've extracted the .bin file from the zip, install it with:
```
$ sudo rpi-eeprom-update -f pieeprom_is1965.bin
```

gitbls, 2025-04-21:
Tested and confirmed fix. 

Will it be in the next eeprom update?

Thx!

pelwell, 2025-04-21:
The change hasn't been approved and merged yet, but I think it's unlikely we'll make a release before it gets merged, so yes.

gitbls, 2025-04-21:
Perfect. Thanks again!

mbroz, 2025-04-21:
Thanks! Maybe you can also update [auto_initramfs](https://www.raspberrypi.com/documentation/computers/config_txt.html#auto_initramfs) documentation.

While debugging this issue, the description was confusing. It should probably say that it selects initramfs based on the autodetected kernel name. But how does it behave when an explicit kernel is selected (but no initramfs)?

pelwell, 2025-04-23:
There's another auto_initramfs implementation that works for custom kernel names. The only requirements are that the kernel image name starts with `kernel`, the initramfs with `initramfs`, and that the rest matches (except that initramfs doesn't have `.img` on the end). Put another way, the names are of the form `kernel<variant>.img` and `initramfs<variant>`, where `<variant>` can be any reasonable string (don't go looking for exceptions).

This scheme obviously works with the names currently used with RPiOS, but also allows for other variations.

The new image is available here: https://drive.google.com/file/d/1RL4HNb-5AyyIOc_JyNqqAumqc2SZxJwE/view?usp=drive_link

pelwell, 2025-05-08:
A new Pi 5 EEPROM containing the initramfs selection fix is now available in the "latest" channel for rpi-eeprom-update.

pelwell, 2025-05-08:
Yes, that's the idea - is that not what I said?

pelwell, 2025-05-08:
I can confirm:
```
$ sudo apt update
...
$ sudo apt install rpi-eeprom
...
$ sudo rpi-eeprom-update
*** UPDATE AVAILABLE ***

Run "sudo rpi-eeprom-update -a" to install this update now.

To configure the bootloader update policy run "sudo raspi-config"

BOOTLOADER: update available
   CURRENT: Thu  8 May 11:12:23 UTC 2025 (1746702743)
    LATEST: Thu  8 May 14:13:17 UTC 2025 (1746713597)
   RELEASE: latest (/usr/lib/firmware/raspberrypi/bootloader-2712/latest)
            Use raspi-config to change the release.
```

gitbls, 2025-05-08:
Thx! Testing now, will report back

pelwell, 2025-05-08:
I spoke too hastily. I had already run rpi-update, which downloaed the newest image. It hasn't actually appeared in the apt repo yet - my apologies.

gitbls, 2025-05-08:
> I spoke too hastily. I had already run rpi-update, which downloaed the newest image. It hasn't actually appeared in the apt repo yet - my apologies.

Maybe not. I took a fully-updated (as of 30 mins ago) system, change /etc/default/rpi-eeprom-update to 'latest', set kernel=kernel8.img and did the sdm-cryptconfig conversion.

It worked perfectly.

I can quickly/easily do it again with rpi-update if that would be helpful for your testing. LMK.

Thanks again for chasing this down and correcting it!

pelwell, 2025-05-08:
What is ` vcgencmd bootloader_version` showing? The latest release is:
```
2025/05/08 15:13:17
version 69471177ba7e4cb7597cb2496f2a0b23f19c1113 (release)
timestamp 1746713597
update-time 1746722270
capabilities 0x0000007f
```
If what you have is older than that then it would be good if you could rpi-update to get the newest.

gitbls, 2025-05-08:
> What is ` vcgencmd bootloader_version` showing? The latest release is:
>
> ```
> 2025/05/08 15:13:17
> version 69471177ba7e4cb7597cb2496f2a0b23f19c1113 (release)
> timestamp 1746713597
> update-time 1746722270 [...]

Oh crap. I just wiped that disk to go ahead and test an rpi-updated system. Sorry! Will rerun the first scenario in a few mins

pelwell, 2025-05-08:
Only if you really want to. It's more important that the rpi-update release gets its tyres kicked a bit.

gitbls, 2025-05-08:
Interesting. I took the same customized IMG I used for the first (non-rpi-updated test...that worked), modified config.txt and /etc/default/rpi-eeprom-update, and did an rpi-update. 

This fails in initramfs as it did when the bug was initially reported.

So:
1) Fully-updated system works correctly
2) rpi-updated system fails

pelwell, 2025-05-08:
Please post the output of `vcgencmd bootloader_version` and `sudo vclog -m | grep initramfs` for your rpi-updated system.

gitbls, 2025-05-08:
> Please post the output of `vcgencmd bootloader_version` and `sudo vclog -m | grep initramfs` for your rpi-updated system

I need to redo the test since the disk has been hashed and is in the middle of a LUKS conversion, so not recoverable.

Will redo both tests, recording the output from both the vanilla updated system and the rpi-updated system.

gitbls, 2025-05-08:
Confirmed. Works with a fully apt-updated system, but fails with an rpi-updated system.

Both tests show vcgnecmd bootloader-version:
2025/05/08 15:13:17
version 69471177ba7e4cb7597cb2496f2a0b23f19c1113 (release)
timestamp 1746713597
update-time 1746723487
capabilities 0x0000007f

Here's the output from rpi-update:
```
You appear to be using a custom kernel file.
Skipping installation of new kernel, as bundled dtb files may be incompatible with your kernel.
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
FW_REV:f54e67fef6e726725d3a8f56d232194497bd247c
BOOTLOADER_REV:cd4048df1d55be89bf84879754a4acf9c92e1f7a
 *** We're running for the first time
 *** Backing up files (this will take a few minutes)
 *** Backing up firmware
 *** Backing up modules 6.12.25+rpt-rpi-v8
WANT_32BIT:0 WANT_64BIT:1 WANT_64BIT_RT:0 WANT_PI4:1 WANT_PI5:1
##############################################################
WARNING: This update bumps to rpi-6.12.y linux tree
See discussions at:
https://forums.raspberrypi.com/viewtopic.php?t=379745
##############################################################

Downloading bootloader tools
Downloading bootloader images
 *** Downloading specific firmware revision (this will take a few minutes)
BOOTLOADER: up to date
   CURRENT: Thu May  8 02:13:17 PM UTC 2025 (1746713597)
    LATEST: Thu May  8 02:13:17 PM UTC 2025 (1746713597)
   RELEASE: latest (/usr/lib/firmware/raspberrypi/bootloader-2712/latest)
            Use raspi-config to change the release.
 *** Updating firmware
 *** Updating kernel modules
 *** depmod 6.12.27-v8+
 *** depmod 6.12.27-v8-16k+
 *** Updating VideoCore libraries
 *** Running ldconfig
 *** Storing current firmware revision
 *** Deleting downloaded files
 *** Syncing changes to disk
 *** If no errors appeared, your firmware was successfully updated to f54e67fef6e726725d3a8f56d232194497bd247c
 *** A reboot is needed to activate the new firmware
```
`vclog -m | grep initramfs` from the rpi-updated system after rpi-update and reboot:
```
006409.646: initramfs (initramfs8) loaded to 0x2e408000 (size 0xbf7c0a)
```
I was doing initial testing using the full sdm-cryptconfig. I just now did another test using the scenario described in the OP for this issue. When it got to the break in premount for the rpi-updated system, `lsmod` showed no modules.

Presumably it was correct in the "fully apt-updated" system, since the rootfs encryption process worked.

My highly unscientific conclusion from 8 time zones away is that whatever is in apt works fine now, but something either didn't make it into the rpi-update version, or something else was added that broke the rpi-update version. 

Happy to do further testing as needed.

gitbls, 2025-05-08:
> If, while running the rpi-update bootloader, you were now to reinstall the apt kernel you would find that your initramfs is working at it should.

Is `sudo apt install --reinstall raspberrypi-kernel` the way to do that?

Thx

gitbls, 2025-05-08:
hmmm...apparently not. I tried that. It updated raspberrypi-kernel, but when the system reboots, it heads into initramfs (bc of my `break=premount`), gets to the initramfs prompt and spews about 20 lines of `usb 3-2.2: device descriptor read/64, error -32` errors.

LMK how I should reinstall the apt kernel correctly and I'll give it another go.

timg236, 2025-05-08:
The rpi-eeprom firmware updates is now in APT. 
The changes are all pretty safe and have been tested individually so this doesn't need multiple days of testing in rpi-update
```
sudo apt reinstall rpi-eeprom
Package: rpi-eeprom
Version: 27.8-1
Priority: optional
Section: misc
Maintainer: Tim Gover <tim.gover@raspberrypi.com>
Installed-Size: 35.4 MB
Provides: rpi-eeprom-images
Depends: raspi-utils, python3, binutils, pciutils, python3-pycryptodome
Recommends: flashrom
Breaks: rpi-eeprom-images (<< 7.2)
Replaces: rpi-eeprom-images (<< 7.2)
Homepage: https://github.com/raspberrypi/rpi-eeprom/
Download-Size: 2,965 kB
APT-Manual-Installed: yes
APT-Sources: http://archive.raspberrypi.com/debian bookworm/main arm64 Packages
Description: Raspberry Pi 4/5 boot EEPROM updater
 Checks whether the Raspberry Pi bootloader EEPROM is up-to-date and updates
 the EEPROM.
```

gitbls, 2025-05-09:
> The rpi-eeprom firmware updates is now in APT. The changes are all pretty safe and have been tested individually so this doesn't need multiple days of testing in rpi-update
> 

Thanks for the update on the update! 😁

This version works correctly in the mixed pagesize scenario.

closed 2025-05-09 by gitbls