raspberrypi/firmware #1966: Upgrade from 1.20250305 to 1.20250326, display no longer works

op1407, 2025-04-27:
Firmware upgrade caused the touchscreen display to stop working. No image, just scrambled lines.
Tried to manually downgrade the firmware, went back too far to version 6.6.xx, display worked but there were other unrelated issues.

It has been confirmed that the display works, Raspi 5 works, ports (USB,HDMI) works.

```
pi@LucyPi5:~ $ uname -a
Linux LucyPi5 6.12.22-v8-16k+ #1871 SMP PREEMPT Mon Apr  7 13:45:56 BST 2025 aarch64 GNU/Linux
```

```
pi@LucyPi5:~ $ vcgencmd version
2025/04/08 00:16:47
Copyright (c) 2012 Broadcom
version 50beecf0 (release) (embedded)
```

There was a bug fix for a another type of display, that why the firmware was updated to 6.12.22. It didn't help my case.

/boot/firmware/config.txt

```
# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
#display_auto_detect=1

# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
#dtoverlay=vc4-kms-v3d
#max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

![Image](https://github.com/user-attachments/assets/3f8ce10f-72ea-4255-8b3c-cf87340882bc)

![Image](https://github.com/user-attachments/assets/c157566e-a79f-41f5-90ee-01067b7ec9ae)

[cm5]
dtoverlay=dwc2,dr_mode=host

[all]

dtoverlay=vc4-fkms-v3d
max_framebuffers=2

lcd_rotate=2
start_x=1
gpu_mem=256

hdmi_blanking=1

hdmi_group=2
hdmi_mode=87
hdmi_cvt=1024 600 60 0 0 0
display_hdmi_rotate=3

hdmi_force_hotplug=1

hdmi_driver=2
#usb_max_current_enable=1

#device_tree_address=0x03000000
#enable_tvout=0
```

Display:
RPI DIS10180P
Elecrow RC101S 10 1024x600 IPS

pelwell, 2025-04-28:
That display has an HDMI interface, so it shouldn't need any special configuration unless the EDID is dodgy in some way. I'm bothered by all the old `hdmi_...` config.txt settings - most of them have no effect on a Pi 5 (and never have), only `hdmi_force_hotplug`.

Things to try (separately):
1. Comment out `disable_fw_kms_setup=1`.
2. Change `vc4-fkms-v3d` to `vc4-kms-v3d`.

op1407, 2025-04-29:
I've had some issues with the display from the start, that is why the config.txt is what it is. It used to work perfectly before the upgrade.

I've tried your suggestions, 

> That display has an HDMI interface, so it shouldn't need any special configuration unless the EDID is dodgy in some way. I'm bothered by all the old `hdmi_...` config.txt settings - most of them [...]

The number 2 disabled the display completely, now it just says "No Signal", the number 1 didn't do anything.

pelwell, 2025-04-29:
What desktop/app are you running?

JamesH65, 2025-05-30:
For the bootloader screen the Pi simply chooses a standard video mode and uses that - it does not ask the monitor what it supports. Past the bootloader, the system interrogates the monitor and will then select a mode from the list of supported modes, which are contained in an EDID. However, the monitor is NOT returning a valid EDID, hence the system is unable to choose a video mode. This is usually down to bad cables, bad adapters or bad monitors.

pelwell, 2025-05-30:
> Is there a way to downgrade the firmware to 1.20250305 version ?

The available images are stored in /usr/lib/firmware/raspberrypi/bootloader-2712/default. The one which was active on 20250305 looks like pieeprom-2025-02-12.bin. Try this:
```
$ sudo rpi-eeprom-update -f /usr/lib/firmware/raspberrypi/bootloader-2712/default/pieeprom-2025-02-12.bin
$ sudo reboot
```