raspberrypi/firmware #1968: VCIO tag RPI_FIRMWARE_GET_FIRMWARE_HASH returns invalid response size (8 instead of 20)

zyga, 2025-05-05:
**Describe the bug**

VCIO tag `RPI_FIRMWARE_GET_FIRMWARE_HASH` (00x00000003) seems to mishandle return buffer size, reporting 8 bytes instead of 20. The Linux kernel ignores the size encoded in the response status/size word, so the error is unnoticed there.

Hexdump of the property request:

```
addr _0 _1 _2 _3 | _4 _5 _6 _7 | _8 _9 _A _B | _C _D _E _F
000_ 2C .. .. .. | .. .. .. .. | 03 .. .. .. | 14 .. .. ..
....
002_ .. .. .. .. | .. .. .. .. | .. .. .. ..
```

Hexdump of the property response:
```
addr _0 _1 _2 _3 | _4 _5 _6 _7 | _8 _9 _A _B | _C _D _E _F
000_ 2C .. .. .. | .. .. .. 80 | 03 .. .. .. | 14 .. .. ..
001_ 08 .. .. 80 | 5B 97 08 28 | BF 9B 14 80 | 65 44 68 E8
002_ 7D 5C E4 5F | 78 C0 6F E6 | .. .. .. ..
```

**To reproduce**

```
$ vcmailbox 0x00000003 0x14 0 0 0 0 0 0
0x0000002c 0x80000000 0x00000003 0x00000014 0x80000008 0x2808975b 0x80149bbf 0xe8684465 0x5fe45c7d 0xe66fc078 0x00000000 
```

**Expected behaviour**

The tag status/response word should IMO equal 0x8000_0014 as the firmware hash is five 32bit words long.

**Actual behaviour**

The tag status/response word is 0x8000_0008 which indicates that the response used 8 bytes of the value buffer.

**System**

https://paste.debian.net/hidden/69ae0d85/

**Logs**

N/A

**Additional context**

N/A

zyga, 2025-05-05:
This also seems to affect tag 0x30080 (what is the official name?)

pelwell, 2025-05-07:
0x30080 would be RPI_FIRMWARE_GET_GENCMD_RESULT. There's a potential fix for both of those in the pipeline.

pelwell, 2025-05-08:
There is a trial firmware with the suggested fix available to download: https://drive.google.com/file/d/161RDNSHAISTtv0cCOpBSsq1GY8tfx7jU/view?usp=sharing

Unpack the .zip archive, copy the contents to /boot/firmware, and reboot.

zyga, 2025-05-08:
@pelwell hash looks good now, thank you:

```
zyga@pi2-2:~$ vcmailbox 0x00000003 0x14 0 0 0 0 0 0
0x0000002c 0x80000000 0x00000003 0x00000014 0x80000014 0x53e4b446 0x260a1033 0x259c90e6 0x82857d20 0x7c91d579 0x00000000 
zyga@pi2-2:~$ ls
firmware_is1968.zip  pipropinsight
zyga@pi2-2:~$ ./pipropinsight --help
Usage of ./pipropinsight:
  -c string
    	vcgencmd command
  -hexdump
    	hexdump mailbox request/response messages
zyga@pi2-2:~$ ./pipropinsight -hexdump -c version
2025/05/08 11:55:59 property request
addr _0 _1 _2 _3 | _4 _5 _6 _7 | _8 _9 _A _B | _C _D _E _F
000_ 0C 04 .. .. | .. .. .. .. | 80 .. 03 .. | F4 03 .. ..
001_ .. .. .. .. | .. .. .. .. | 76 65 72 73 | 69 6F 6E ..
....
040_ .. .. .. .. | .. .. .. .. | .. .. .. ..
2025/05/08 11:55:59 property response
addr _0 _1 _2 _3 | _4 _5 _6 _7 | _8 _9 _A _B | _C _D _E _F
000_ 0C 04 .. .. | .. .. .. 80 | 80 .. 03 .. | F4 03 .. ..
001_ F4 03 .. 80 | .. .. .. .. | 4D 61 79 20 | 20 37 20 32
002_ 30 32 35 20 | 31 37 3A 34 | 39 3A 33 31 | 20 0A 43 6F
003_ 70 79 72 69 | 67 68 74 20 | 28 63 29 20 | 32 30 31 32
004_ 20 42 72 6F | 61 64 63 6F | 6D 0A 76 65 | 72 73 69 6F
005_ 6E 20 35 33 | 65 34 62 34 | 34 36 32 36 | 30 61 31 30
006_ 33 33 32 35 | 39 63 39 30 | 65 36 38 32 | 38 35 37 64
007_ 32 30 37 63 | 39 31 64 35 | 37 39 20 28 | 63 6C 65 61
008_ 6E 29 20 28 | 72 65 6C 65 | 61 73 65 29 | 20 28 73 74
009_ 61 72 74 29 | .. .. .. .. | .. .. .. .. | .. .. .. ..
....
040_ .. .. .. .. | .. .. .. .. | .. .. .. ..
command code: 0, text: May  7 2025 17:49:31 
Copyright (c) 2012 Broadcom
version 53e4b446260a1033259c90e682857d207c91d579 (clean) (release) (start)
```

Thinking about `RPI_FIRMWARE_GET_GENCMD_RESULT` - should it perhaps set the result value size to the length of the response instead of just copying the request buffer size?

pelwell, 2025-05-08:
> Thinking about RPI_FIRMWARE_GET_GENCMD_RESULT - should it perhaps set the result value size to the length of the response instead of just copying the request buffer size?

That was my intention, but from what I could see the internal function that implementing the gencmd does not provide that implementation. Is that correct, @popcornmix ?

popcornmix, 2025-05-08:
While the internal function doesn't directly report the length, it should be a null terminated string, so we could probably use strlen to generate the actual response size.

pelwell, 2025-05-08:
A second trial version with a calculated response length can be downloaded here: https://drive.google.com/file/d/1ueQSexHaHB2GVAzJF6rwuENEK6pqF5JR/view?usp=sharing

zyga, 2025-05-08:
Testing!

zyga, 2025-05-08:
Looks good. I have not seen any regressions in all the other mbox tags I exercise.

pelwell, 2025-05-08:
Thanks - that's been merged, and will be in future releases. In the meantime you can continue to use the trial version.

zyga, 2025-05-08:
Thanks!

I think I found another one: `RPI_FIRMWARE_GET_CLOCKS` does not really tell the number of clocks that are returned and one has to stop parsing when many zeros are encountered:

```
2025/05/08 15:53:53 property request
addr _0 _1 _2 _3 | _4 _5 _6 _7 | _8 _9 _A _B | _C _D _E _F
000_ 18 02 .. .. | .. .. .. .. | 07 .. 01 .. | .. 02 .. ..
....
021_ .. .. .. .. | .. .. .. ..
2025/05/08 15:53:53 property response
addr _0 _1 _2 _3 | _4 _5 _6 _7 | _8 _9 _A _B | _C _D _E _F
000_ 18 02 .. .. | .. .. .. 80 | 07 .. 01 .. | .. 02 .. ..
001_ .. 02 .. 80 | .. .. .. .. | 01 .. .. .. | .. .. .. ..
002_ 02 .. .. .. | .. .. .. .. | 03 .. .. .. | .. .. .. ..
003_ 04 .. .. .. | .. .. .. .. | 05 .. .. .. | .. .. .. ..
004_ 06 .. .. .. | .. .. .. .. | 07 .. .. .. | .. .. .. ..
005_ 08 .. .. .. | .. .. .. .. | 09 .. .. .. | .. .. .. ..
006_ 0A .. .. .. | .. .. .. .. | 0B .. .. .. | .. .. .. ..
007_ 0C .. .. .. | .. .. .. .. | 0D .. .. .. | .. .. .. ..
008_ 0E .. .. .. | .. .. .. .. | 0F .. .. .. | .. .. .. ..
....
021_ .. .. .. .. | .. .. .. ..
```

closed 2025-05-08 by zyga