raspberrypi/firmware #1969: RPI_FIRMWARE_GET_CLOCKS

zyga, 2025-05-08:
**Describe the bug**

The `RPI_FIRMWARE_GET_CLOCKS` mailbox tag does not correctly communicate the response size required for variable-sized property.

**To reproduce**

```
$ vcmailbox 0x00010007 8 0 0 0
0x00000020 0x80000000 0x00010007 0x00000008 0x80000008 0x00000000 0x00000001 0x00000000
```

Here we're providing value for one clock-pair (parent, self) by providing two words (0x00000008) of storage. The response word `0x80000008` is insufficient to know if more clocks are to be expected.

Similarly a larger value buffer is padded with empty pairs.

```
$ vcmailbox 0x00010007 255 $(for _ in $(seq $((256/4))); do echo 0; done) 0
0x00000118 0x80000000 0x00010007 0x000000ff 0x800000ff 0x00000000 0x00000001 0x00000000 0x00000002 0x00000000 0x00000003 0x00000000 0x00000004 0x00000000 0x00000005 0x00000000 0x00000006 0x00000000 0x00000007 0x00000000 0x00000008 0x00000000 0x00000009 0x00000000 0x0000000a 0x00000000 0x0000000b 0x00000000 0x0000000c 0x00000000 0x0000000d 0x00000000 0x0000000e 0x00000000 0x0000000f 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
```

Here clocls seem to end at clock-id `0x0f`

**Expected behaviour**

The tag status/response word should provide the size of the value buffer sufficient to obtain a tree of all the clocks.

**Actual behaviour**

The tag status/response word is a copy of the value buffer size bit-wise ORed with the response bit. One must provide a large buffer and guess when to stop interpreting results.

**System**

https://paste.debian.net/hidden/69ae0d85/

**Logs**

N/A

**Additional context**

This is a follow-up from https://github.com/raspberrypi/firmware/issues/1968

pelwell, 2025-05-08:
It's a quirk of the kernel's firmware mailbox API that it doesn't report the actual length of the results, so we've never cared about this. A trivial fix is up for internal review. In the meantime, we're curious about your use case - what are you working on?

pelwell, 2025-05-08:
As if by magic, another trial firmware appears: https://drive.google.com/file/d/1o3uKLTZjgY6cdnwLIeyXDGH3jLFJ-uIn/view?usp=sharing

zyga, 2025-05-08:
Woot

Looking

pelwell, 2025-05-08:
What the limited documention (https://github.com/raspberrypi/firmware/wiki/Mailbox-property-interface#get-clocks) is saying is that if the value in the parent field is 0 then this is a root clock, i.e. there is no parent.

zyga, 2025-05-08:
Right, I understand that. I asked because if all the (currently reported) clocks are root clocks then it's just a flat list. That's fine, perhaps in the future a different clock tree is reported. I was just curious.

pelwell, 2025-05-08:
I'm not sure if this is overengineering or if there was once a use for it.

zyga, 2025-05-08:
Thank you for fixing all the issues so rapidly. This allowed me to make the library interface much cleaner.

closed 2025-05-12 by pelwell