raspberrypi/firmware #1970: RPI_FIRMWARE_NOTIFY_DISPLAY_DONE takes more than 1000 ms on USB boot

lategoodbye, 2025-05-18:
Hi,
recently we (Linux Mailing list) received a [bugfix](https://lore.kernel.org/linux-arm-kernel/aCIiEp3CXD2o9dTw@Z926fQmE5jqhFMgp6/) to the Raspberry Pi firmware driver. Since I never saw this timeout before, this drawn my attention.

**Describe the bug**
During USB boot on the Raspberry Pi 4 the firmware property RPI_FIRMWARE_NOTIFY_DISPLAY_DONE takes more that 1000 ms (USB 2.0 port or 3.0 port doesn't matter). Strangely this is not reproducible while booting from SD card.
````
[    2.861407] ------------[ cut here ]------------
[    2.865512] Firmware transaction 0x00030066 timeout
[    2.865549] WARNING: CPU: 3 PID: 42 at drivers/firmware/raspberrypi.c:128 rpi_firmware_property_list+0x21c/0x29c
[    2.880751] CPU: 3 UID: 0 PID: 42 Comm: kworker/u16:1 Not tainted 6.15.0-rc6 #1 PREEMPT
[    2.888944] Hardware name: Raspberry Pi 4 Model B Rev 1.5 (DT)
[    2.894848] Workqueue: events_unbound deferred_probe_work_func
[    2.900752] pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[    2.907801] pc : rpi_firmware_property_list+0x21c/0x29c
[    2.913089] lr : rpi_firmware_property_list+0x21c/0x29c
[    2.918376] sp : ffffffc0803139c0
[    2.921725] x29: ffffffc0803139e0 x28: ffffff8040bbef50 x27: ffffff80410c0f40
[    2.928953] x26: ffffffd7055d9e28 x25: ffffffc0801e0008 x24: 0000000000001000
[    2.936179] x23: ffffff80410c1080 x22: 000000000000000a x21: ffffff80410c0f00
[    2.943405] x20: 000000000000000c x19: ffffffc0801e0000 x18: ffffffc08030d0a0
[    2.950632] x17: 0000000000000000 x16: 0000000000000000 x15: 0000000000000000
[    2.957858] x14: 0000000000000000 x13: 0000000000000000 x12: 0000000000000000
[    2.965085] x11: 0000000000000000 x10: 0000000000000000 x9 : 0000000000000000
[    2.972311] x8 : 0000000000000000 x7 : 0000000000000000 x6 : 0000000000000000
[    2.979537] x5 : 0000000000000000 x4 : 0000000000000000 x3 : 0000000000000000
[    2.986764] x2 : 0000000000000000 x1 : 0000000000000000 x0 : 0000000000000000
[    2.993992] Call trace:
[    2.996458]  rpi_firmware_property_list+0x21c/0x29c (P)
[    3.001747]  rpi_firmware_property+0x70/0xd8
[    3.006064]  vc4_drm_bind+0x12c/0x378
[    3.009765]  try_to_bring_up_aggregate_device+0x22c/0x308
[    3.015230]  __component_add+0xec/0x224
[    3.019106]  component_add+0x14/0x30
[    3.022720]  vc4_hdmi_dev_probe+0x1c/0x40
[    3.026773]  platform_probe+0x68/0xf0
[    3.030474]  really_probe+0xc0/0x3ac
[    3.034088]  __driver_probe_device+0x7c/0x174
[    3.038495]  driver_probe_device+0x40/0x100
[    3.042725]  __device_attach_driver+0x10c/0x1e0
[    3.047308]  bus_for_each_drv+0x88/0x100
[    3.051273]  __device_attach+0xa0/0x1c8
[    3.055151]  device_initial_probe+0x14/0x30
[    3.059381]  bus_probe_device+0xc8/0xcc
[    3.063259]  deferred_probe_work_func+0xb8/0x12c
[    3.067930]  process_one_work+0x160/0x2d4
[    3.071983]  worker_thread+0x2d8/0x400
[    3.075773]  kthread+0x12c/0x208
[    3.079034]  ret_from_fork+0x10/0x20
[    3.082647] ---[ end trace 0000000000000000 ]---
````
I was able to bisect and the regression was introduced by Add memory barrier to the mbox handler 9995946dce284505d3dd46c7563d74a691f38a70. Downgrading to kernel: Bump to 6.6.78 70631e717cc9e758d18a08071a859f69030a0f06 doesn't show this issue during USB boot.

**To reproduce**
- use a Raspberry Pi 4 (arm64)
- install a recent Raspberry Pi firmware (at least firmware: Add memory barrier to the mbox handler 9995946dce284505d3dd46c7563d74a691f38a70 )
- boot from USB with HDMI monitor connected to HDMI 0

**Expected behaviour**
RPI_FIRMWARE_NOTIFY_DISPLAY_DONE should around 40 ms like on SD card boot

**Actual behaviour**
RPI_FIRMWARE_NOTIFY_DISPLAY_DONE takes around 1000 ms 

* Which model of Raspberry Pi? Raspberry Pi 4
* Which firmware version (`vcgencmd version`)? 9995946dce284505d3dd46c7563d74a691f38a70 and higher
* Which kernel version (`uname -a`)? at least 6.12 or 6.15

6by9, 2025-05-19:
Thanks Stefan. I'd seen the thread, but wasn't in the office last week.

My normal Pi4 is booting off USB and I'm not seeing the issue at present. I'll have a bit more of a dig into it and look at what those commits are doing.

lategoodbye, 2025-05-19:
@6by9 Since the response time is near to the 1s timeout, it's possible that the timeout isn't triggered everytime.

Please try the following hack to measure the response times: https://gist.github.com/lategoodbye/444af7d8ccb1f888645b28ebc1f27f53

6by9, 2025-05-19:
Fresh USB stick updated to latest Raspberry Pi OS, and then my 6.12.28 kernel with your gist. 
It still responds in ~40ms with a 1080p60 monitor. With a 4k monitor (which will default to 4k30) I get ~60ms. (IIRC the HVS will always finish the current frame before shutting down, hence there is variation with frame rate).

I'd be surprised if this was a board revision issue, but for reference, this is on a board with rev d03114.

lategoodbye, 2025-05-19:
I must confess that i didn't rule out every possible dependency. I only tested with Mainline kernel (arm64) + Mainline DT as the original reporter. He send me his configuration, i will try if this is relevant.

USB keyboard + mouse are also connected for the USB ports.

Here the relevant changes to the config.txt

````
arm_64bit=1
disable_overscan=1
enable_uart=1
device_tree=bcm2711-rpi-4-b.dtb
````

6by9, 2025-05-19:
OK, I'll switch to a mainline kernel, although memory says the default config can't USB boot (some PCIe drivers are modules rather than built-in).

I have keyboard, mouse, and ethernet connected up to the USB-C connector and hence dwc2 driver, as that's the easiest way I've found to swap around configurations. (Uses a https://www.amazon.co.uk/dp/B0957DFS8V?th=1 to do USB-PD and USB services).

lategoodbye, 2025-05-19:
> OK, I'll switch to a mainline kernel, although memory says the default config can't USB boot (some PCIe drivers are modules rather than built-in).

Good point. Here is the Kernel [configuration](https://gist.github.com/lategoodbye/45d22ed5c9a7175eaa9da9e9f84a99e4) for 6.15-rc6. Everything is builtin.

etbuira, 2025-05-19:
> Everything is built-in.

Excepted dwc2 gadget.

6by9, 2025-05-19:
Thanks. Mainline (a 6.15.0-rc1 branch I had around for some media patches) with everything built in triggers it.
1128ms delay, started at boot+0.5s, ending at boot+1.59s

And I agree 100% with your bisection of the firmware commits. https://github.com/raspberrypi/rpi-firmware/commit/b5e0052 doesn't work, but https://github.com/raspberrypi/rpi-firmware/commit/b5e0052~1 does.
I can try building a few test firmwares to see which of the handful of commits in that window cause the issue, and then prod the appropriate people.

6by9, 2025-05-19:
FW commit causing the regression found.
No idea why at present. I've emailed @pelwell with some details, but it'll wait for the morning.

pelwell, 2025-05-20:
It took half a day to (mostly) understand. The bug is very simple - not treating the absence of a `status` property the same as `status = "okay";` - but the chain of events that leads to the observed symptom is convoluted.

There's a trial fix available to download here: https://drive.google.com/file/d/1k1TvV477uR3YvBUEg8gYSFaoN8cIoaGm/view?usp=drive_link

It's just start4.elf and fixup4.dat.

etbuira, 2025-05-21:
Hi, thank you, i can confirm the fix works.

Gabmwiha, 2026-10-08:
Hello