raspberrypi/firmware #1973: Raspberry Pi Zero 2 W WiFi networking/SSH unusable or crashing/rebooting

sysgarpi, 2025-05-30:
I have multiple Raspberry Pi Zero W 2, installed with Raspberry Pi OS (32bit, bookworm) using the rpi-imager. I am experiencing very inconsistent stability, ranging from no problems at all, to extreme delays in the network/SSH communication up to crashes/reboots.
I've done extensive tests to troubleshoot the issue and I believe to have narrowed it down to the WiFi access point the Pi's are connected to.

More details on the process of pinpointing the culprit is documented here: https://forums.raspberrypi.com/viewtopic.php?p=2318044 (Thank you @gitbls for all the support!).

To summarize my approaches; I've tried a variety of combinations between different RPi02W boards, micro SD cards, power supplies, USB cables, wired/wireless network and wireless access points.  The common denominator for cases where I experienced the issue was the WiFi access point of my FRITZ!Box 6490.

As I found other issues discussing WiFi, I hope I chose the correct repo to submit this, show me the proper repo otherwise.

I am unable to purposely reproduce the delays or crash/reboots. I have one Pi set up for a while now, attempting to log the crash/reboot, but it hasn't happened since then, while it happened frequently before.

To quantify unusable network or extreme delays I write this quick script:

```
!/usr/bin/bash

TIMEFORMAT=%R

i=0
while true; do
        ((i=i+1))
        echo "Timing SSH #${i}.. "
        { time timeout 60 sshpass -p 'password' ssh sysgar@192.168.178.134 'uptime'; } 2>> times_savior.log
        sleep 10
done

```

The resulting times, if the delays occur are, for example:

```
2.253, 10.155, 9.361, 1.651, 16.094, 2.956, 8.395, 60.006, 60.006, 3.302, 6.843, 60.004, 15.030, 3.201, 3.904, 2.531, 9.490, 2.585, 9.388, 31.759, 2.594, 5.792, 2.244, 31.263, 30.349, 60.006, 2.553, 2.605, ...
```

If I wouldn't put a timeout to the connection, it would sometimes finish anywhere between 10-30 minutes, but also had some cases where it exceeded 80 minutes.
But with the same setup, I've witnessed stretches of time, where the connection was perfectly fine, with consistent completion times of <2 seconds.

As elsewhere suggested I added the following file `/etc/modprobe.de/02w-wifi-fix.conf`, but this help in my case:

```
options brcmfmac debug=0x104404
options brcmfmac feature_disable=0x2000
```

**Logs**

The following shows an excerpt of the journalctl logs of one of the Pi's experiencing the delays, but no crash/reboot so far: `journalctl | grep brcm`. I have `brcmfmac.debug=0x100000` added to `/boot/firmware/cmdline.txt`.

```
May 13 02:17:54 raspberrypi kernel: Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 cgroup_disable=memory snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0  smsc95xx.macaddr=01:23:45:67:89 vc_mem.
mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  console=ttyS0,115200 console=tty1 root=PARTUUID=0b22e5e3-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=DE systemd.run=/boot/firmware/firstrun.sh systemd.run_success_ac
tion=reboot systemd.unit=kernel-command-line.target brcmfmac.debug=0x100000
May 13 02:17:56 raspberrypi kernel: Bluetooth: hci0: BCM43430A1 'brcm/BCM43430A1.raspberrypi,model-zero-2-w.hcd' Patch
May 13 02:17:56 raspberrypi kernel: brcmfmac: F1 signature read @0x18000000=0x1541a9a6                                                                                                                                                         May 13 02:17:56 raspberrypi kernel: brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
May 13 02:17:56 raspberrypi kernel: usbcore: registered new interface driver brcmfmac
May 13 02:17:57 raspberrypi kernel: brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
May 13 02:17:57 raspberrypi kernel: brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Jun 14 2023 07:27:45 version 7.45.96.s1 (gf031a129) FWID 01-70bd2af7 es7
May 13 02:17:57 raspberrypi kernel: brcmfmac: CONSOLE: hndarm_armr addr: 0x18003000, cr4_idx: 0
May 13 02:17:57 raspberrypi kernel: brcmfmac: CONSOLE: 000000.001
May 13 02:17:57 raspberrypi kernel: brcmfmac: CONSOLE: RTE (SDIO-CDC) 7.45.96.s1 on BCM43430 r1 @ 37.4/81.6/81.6MHz
May 13 02:17:57 raspberrypi kernel: brcmfmac: CONSOLE: 000000.001 sdpcmdcdc0: Broadcom SDPCMD CDC driver
May 13 02:17:57 raspberrypi kernel: brcmfmac: CONSOLE: 000000.007 reclaim section 0: Returned 45474 bytes to the heap
May 13 02:17:57 raspberrypi kernel: brcmfmac: CONSOLE: 000000.009 wlc_bmac_info_init: host_enab 1
May 13 02:17:57 raspberrypi kernel: brcmfmac: CONSOLE: 000000.061 wl0: Broadcom BCM43430 802.11 Wireless Controller 7.45.96.s1 (gf031a129)
May 13 02:17:57 raspberrypi kernel: brcmfmac: CONSOLE: 000000.062 wl_nd_ra_filter_init: Enter..
May 13 02:17:57 raspberrypi kernel: brcmfmac: CONSOLE: 000000.062 TCAM: 256 used: 215 exceed:0
May 13 02:17:57 raspberrypi kernel: brcmfmac: CONSOLE: 000000.063 reclaim section 1: Returned 79816 bytes to the heap
May 13 02:17:57 raspberrypi kernel: brcmfmac: CONSOLE: 000000.156 wl0: wlc_enable_probe_req: state down, deferring setting of host flags
May 13 02:17:57 raspberrypi kernel: brcmfmac: CONSOLE: 000000.293 wl0: wlc_enable_probe_req: state down, deferring setting of host flags
May 13 02:18:24 subject kernel: Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 cgroup_disable=memory snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0  smsc95xx.macaddr=01:23:45:67:89 vc_mem.mem_
[... 64 lines cut ...]
May 29 21:19:13 subject kernel: brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
May 29 21:19:13 subject kernel: brcmfmac: CONSOLE: 000002.798 wl0: wl_open
May 29 21:19:14 subject kernel: brcmfmac: CONSOLE: 000003.121 wl0: wl_open
May 29 21:19:14 subject kernel: brcmfmac: CONSOLE: 000003.144 wl0: wlc_enable_probe_req: state down, deferring setting of host flags
May 29 21:19:16 subject kernel: brcmfmac: CONSOLE: 000005.050 E/GTK: wlc_bcol_process_gtk_rekey: 4 handshake frame, skip
May 29 21:19:16 subject kernel: brcmfmac: CONSOLE: 000005.088 E/GTK: wlc_bcol_process_gtk_rekey: 4 handshake frame, skip
May 30 03:32:16 subject kernel: brcmfmac: CONSOLE: 022350.842 E/GTK: wlc_bcol_process_gtk_rekey: 4 handshake frame, skip
May 30 03:32:16 subject kernel: brcmfmac: CONSOLE: 022350.909 E/GTK: wlc_bcol_process_gtk_rekey: 4 handshake frame, skip
May 30 12:31:40 subject NetworkManager[433]: <info>  [1748601100.5014] rfkill1: found Wi-Fi radio killswitch (at /sys/devices/platform/soc/3f300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/ieee80211/phy0/rfkill1) (driver brcmfmac)
May 30 12:31:40 subject kernel: brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
May 30 12:31:41 subject kernel: brcmfmac: CONSOLE: 000003.179 wl0: wl_open
May 30 12:31:41 subject kernel: brcmfmac: CONSOLE: 000003.487 wl0: wl_open
May 30 12:31:41 subject kernel: brcmfmac: CONSOLE: 000003.613 wl0: wlc_enable_probe_req: state down, deferring setting of host flags
May 30 12:31:43 subject kernel: brcmfmac: CONSOLE: 000005.385 E/GTK: wlc_bcol_process_gtk_rekey: 4 handshake frame, skip
May 30 12:31:43 subject kernel: brcmfmac: CONSOLE: 000005.406 E/GTK: wlc_bcol_process_gtk_rekey: 4 handshake frame, skip
May 30 12:33:40 subject NetworkManager[426]: <info>  [1748601220.2208] rfkill1: found Wi-Fi radio killswitch (at /sys/devices/platform/soc/3f300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/ieee80211/phy0/rfkill1) (driver brcmfmac)
May 30 12:33:40 subject kernel: brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
May 30 12:33:40 subject kernel: brcmfmac: CONSOLE: 000002.764 wl0: wl_open
May 30 12:33:41 subject kernel: brcmfmac: CONSOLE: 000003.119 wl0: wl_open
May 30 12:33:41 subject kernel: brcmfmac: CONSOLE: 000003.142 wl0: wlc_enable_probe_req: state down, deferring setting of host flags
May 30 12:33:42 subject kernel: brcmfmac: CONSOLE: 000005.021 E/GTK: wlc_bcol_process_gtk_rekey: 4 handshake frame, skip
May 30 12:33:43 subject kernel: brcmfmac: CONSOLE: 000005.122 E/GTK: wlc_bcol_process_gtk_rekey: 4 handshake frame, skip
May 30 12:49:51 subject kernel: brcmfmac: CONSOLE: 000906.835 E/GTK: wlc_bcol_process_gtk_rekey: 4 handshake frame, skip
May 30 12:49:51 subject kernel: brcmfmac: CONSOLE: 000906.861 E/GTK: wlc_bcol_process_gtk_rekey: 4 handshake frame, skip
```

sysgarpi, 2025-05-31:
> > As elsewhere suggested I added the following file /etc/modprobe.de/02w-wifi-fix.conf, but this help in my case:
> 
> this _didn't_ help?

Correct. Unless there is something else I have to do before the changes take effect, other than creating the file with this content and rebooting the system. I've read that with Raspberry Pi OS it's not necessary to _rebuild initramfs_ (?) and the file permissions shouldn't matter either.

So, with these options present as described, the delays occur just the same as I witnessed before making the changes.

J1mbo, 2025-06-02:
I also have this problem - very unstable WiFi with the Zero 2W after some period of time, in my case when connected to Unifi U6-lite AP.

Unifi configuration for the network is 2.4GHz, WPA2, PMF disabled, 802.11r disabled. All disabled: client device isolation, proxy arp, bss transition, UAPSD, speed limit, multicast enhancement, multicast/broadcast control. On Auto: 802.11 DTIM period and minimum data rate control.

The '2W is within 2 meters of the U6-lite AP. There are no brown-out messages in the logs.

The device has Bookworm with latest 'full-upgrade' applied as of time of writing.

dmesg:

 $ dmesg | grep -i brcmfmac
[    7.540579] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[    7.546009] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    7.547555] usbcore: registered new interface driver brcmfmac
[    7.785833] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[    7.786588] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Jun 14 2023 07:27:45 version 7.45.96.s1 (gf031a129) FWID 01-70bd2af7 es7
[   11.015860] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   12.389805] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled

After applying 'full-upgrade', lowering the RPi transmit power to 12dB and disabling power save, and restarting the AP, connectivity has been dramatically improved (no ping drops in 12 hours). Time will tell if it's a permanent fix.

sysgarpi, 2025-06-15:
I've come across the [Alpine Linux Raspberry Pi documentation](https://wiki.alpinelinux.org/wiki/Raspberry_Pi), which notes the following file/configuration for improved WiFi stability:

```
$cat /etc/modprobe.d/brcmfmac.conf
options brcmfmac roamoff=1 feature_disable=0x282000
```

I've undone all previous configuration and tried, this, but it still results in terribly inconsistent times for SSH connections, tested with the initially mentioned script:

`1.661, 0.782, 0.689, 1.824, 1.637, 1.671, 60.006, 2.412, 1.998, 59.662, 60.003, 60.006, 60.009, 9.265, 60.006, 1.974, 60.006, 60.006, 60.006, 2.053, 3.224, 28.336, 2.255, 2.863, 6.032`

sysgarpi, 2025-06-15:
Just to verify, I've set both feature_disable values and updated initramfs, as I've read that that _could_ make a difference:

```
$cat /etc/modprobe.d/brcmfmac.conf
options brcmfmac roamoff=1 feature_disable=0x282000
options brcmfmac feature_disable=0x2000
```

After updating the configuration file:

```
sudo update-initramfs -u
sudo reboot
```

SSH remains unusable via WiFi.

sysgarpi, 2025-06-16:
> [@sysgarpi](https://github.com/sysgarpi) - according to ChatGPT, "Because modprobe processes options in order and “the last value set wins,” your second line effectively overwrites the first mask, so the only feature you end up disabling is SWSUP (the 4-way WPA handshake offload)". So try instead the single line:
> 
> options brcmfmac roamoff=1 feature_disable=0x282000
> 
> This disables roaming, SWSUP (4-way WPA handshake (EAPOL) in firmware), SAE_EXT (SAE (WPA3) authentication in firmware - note this will result in not being able to join WPA3 networks), and DUMP_OBSS (overlapping-BSS scanning in firmware).

Hi @J1mbo, thanks for the feedback, that's good to know! Unfortunately I've had both options active as a single line in the .conf file as well with the same result.

AllanJard, 2025-09-14:
Also been struggling with a Unifi network and a Pi Zero 2 W with ssh. I found [this post](https://www.reddit.com/r/Ubiquiti/comments/1d9uke1/comment/lkuk8fa/) and did as it said adding `IPQoS 0x00 0x00` to the end of `/etc/ssh/sshd_config`, reboot and it is working great now!

Massive pain to need to do that. Something funny about the Unifi network or does it apply some QoS that is being problematic?

sysgarpi, 2025-11-08:
> [@sysgarpi](https://github.com/sysgarpi)
> 
> i've read about a solution for brcmfmac and pi zero 2's not playing nice with each other (i cannot locate the post now). it was to disable NetworkManager and use wpa_supplicant and systemd-networkd, I've had this running on a machine for a good bit now and it's been smooth sailing.

Thanks for the info, I'll give that a shot as well!

sysgarpi, 2025-12-05:
> [@sysgarpi](https://github.com/sysgarpi) seems disabling bluetooth helps in my, similar to yours, case. In `/boot/firmware/config.txt` add line `dtoverlay=disable-bt`, save and reboot.

Thank you! I'll test that as well, once I have the setup in question in active again 👍

MichaIng, 2026-09-07:
Can we please narrow down the stability issues to a single feature, instead of disabling 4 features including SAE, that way kinda sabotaging the efforts done [here](https://github.com/raspberrypi/linux/issues/7610)? Of course that modprobe.d file can just be removed or overridden, but the average user who finds the linked issue does not know about driver parameters, set by newly added config files on package upgrades. I only saw this by chance, as GitHub shows commits in the conversation history, which mention an issue/PR. Otherwise I could have searched endlessly wondering why previously perfectly working WPA3 suddenly does not work anymore.