raspberrypi/firmware #1974: autoboot problem after updating to new firmware

Muzad, 2025-06-01:
I've created a yocto image with this partitions (with alt-root and all-boot partitions):

```
part /     --source rootfs --exclude-path boot/ --ondisk mmcblk0 --fstype=ext4 --label rootfs1 --align 4096 --fixed-size 400M
part /     --source rootfs --exclude-path boot/ --ondisk mmcblk0 --fstype=ext4 --label rootfs2 --align 4096 --fixed-size 400M
part /boot --source bootimg-partition --ondisk mmcblk0 --fstype=vfat --label boot1 --active --align 4096 --fixed-size 100M
part /boot --source bootimg-partition --ondisk mmcblk0 --fstype=vfat --label boot2 --active --align 4096 --fixed-size 100M

bootloader --ptable gpt
```

Recently i updated the firmware (start4.elf / fixup4.dat) to version 1.20250430. when 'boot_partition=0', rpi boots from first partition and every thing works as expected:
```bash
root@raspberrypi4:/boot# cat autoboot.txt 
boot_partition=0
```

boot log:
```
Read start4.elf bytes  2263968 hnd 0x00000000
Read fixup4.dat bytes     5456 hnd 0x00000000
Firmware: 5560078dcc8591a00f57b9068d13e5544aeef3aa Apr 30 2025 13:33:39
0x00b03140 0x00000000 0x00001fff
MEM GPU: 76 ARM: 948 TOTAL: 1024
Starting start4.elf @ 0xfec00200 partition 0
+

MESS:00:00:02.681911:0: arasan: arasan_emmc_open
MESS:00:00:02.683573:0: arasan: arasan_emmc_set_clock C0: 0x00800000 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 400000 max: 400000 delay: 5
MESS:00:00:02.803812:0: arasan: arasan_emmc_set_clock C0: 0x00800000 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 400000 max: 400000 delay: 5
MESS:00:00:02.816683:0: arasan: arasan_emmc_set_clock C0: 0x00800f00 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 390000 max: 400000 delay: 5
MESS:00:00:02.867495:0: boot-part: 0 fs-type: 5
MESS:00:00:02.989326:0: boot-part: 0 fs-type: 2
MESS:00:00:02.990866:0: arasan: arasan_emmc_set_clock C0: 0x00800f02 C1: 0x000e0407 emmc: 200000000 actual: 25000000 div: 0x00000004 target: 25000000 min: 0 max: 25000000 delay: 1
MESS:00:00:03.121837:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:03.124570:0: brfs: File read: 84 bytes
MESS:00:00:03.150133:0: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:03.154642:0: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:03.171753:0: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:03.176255:0: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:03.181909:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:03.186859:0: gpioman: gpioman_get_pin_num: pin DISPLAY_SDA not defined
MESS:00:00:03.193836:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:03.663654:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:03.669296:0: *** Restart logging
MESS:00:00:03.671939:0: brfs: File read: 84 bytes
MESS:00:00:03.681556:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:03.686587:0: hdmi: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:03.697210:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:03.702235:0: hdmi: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:03.707834:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:03.721624:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:03.726647:0: hdmi: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:03.737268:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:03.742298:0: hdmi: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:03.747895:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:03.756660:0: HDMI0: hdmi_pixel_encoding: 300000000
MESS:00:00:03.762129:0: HDMI1: hdmi_pixel_encoding: 300000000
MESS:00:00:03.787680:0: brfs: File read: /mfs/sd/bcm2711-rpi-cm4.dtb
MESS:00:00:03.790930:0: Loaded 'bcm2711-rpi-cm4.dtb' to 0x100 size 0xbed9
MESS:00:00:03.809873:0: brfs: File read: 48857 bytes
MESS:00:00:03.883736:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:03.900667:0: brfs: File read: 84 bytes
MESS:00:00:03.902384:0: Failed to open command line file 'cmdline.txt'
MESS:00:00:04.303755:0: brfs: File read: /mfs/sd/kernel7l.img
MESS:00:00:04.306397:0: Loaded 'kernel7l.img' to 0x8000 size 0x86538
MESS:00:00:04.312487:0: Device tree loaded to 0x2eff3a00 (size 0xc549)
MESS:00:00:04.321209:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:04.327792:0: uart: Baud rate change done...
MESS:00:00:04.329814:0:
[... 102 lines cut ...]
MESS:00:00:04.372728:0: brfs: File read: 88 bytes
MESS:00:00:04.374410:0: Failed to open command line file 'cmdline.txt'
MESS:00:00:05.448644:0: brfs: File read: /mfs/sd/kernel7l.img
MESS:00:00:05.451290:0: Loading 'kernel7l.img' to 0x8000 size 0x86538
MESS:00:00:05.457460:0: Device tree loaded to 0x2eff3c00 (size 0xc3b3)
MESS:00:00:05.465471:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:05.472769:0: uart: Baud rate change done...
MESS:00:00:05.474791:0:

U-Boot 2021.07 (Jul 05 2021 - 15:11:28 +0000)
...
```

**System**
Machine model: Raspberry Pi Compute Module 4 Rev 1.0

$ uname -a
Linux raspberrypi4 5.10.17-v7l #1 SMP Mon Mar 1 09:44:55 UTC 2021 armv7l GNU/Linux

$ cat /etc/os-release 
ID=poky
NAME="Poky (Yocto Project Reference Distro)"
VERSION="3.2.4 (gatesgarth)"
VERSION_ID=3.2.4
PRETTY_NAME="Poky (Yocto Project Reference Distro) 3.2.4 (gatesgarth)"

timg236, 2025-06-02:
boot_partition 0 means the default partition - which should map to 'boot1'
boot_partition 1 is the first FAT partition in GPT
boot_partition 2 is the second FAT partition in GPT - which would be selected by boot_partition=2 or 'sudo reboot 2'

Muzad, 2025-06-05:
Ahh, Thanks a lot. yes, using boot_partition=1 and 2 resolved the problem on the new firmware. i didn't know that this indexes has changed in the new firmware version. Thanks again and i think this issue can be closed too now.