raspberrypi/firmware #1975: Raspberry Pi 3 latest Firmware disables all USB Ports

solisinvictum, 2025-06-06:
After updating to latest firmware with rpi-update, all USB Ports stops working.

lsusb shows nothing. like nothing would exist.

fixed it for me with rpi-update c7c5c236b82c19212bd0eb54442e9dd2c77bc241

pelwell, 2025-06-06:
That release only changed the kernel, not the firmware. Can you take a look at this page?: https://github.com/raspberrypi/rpi-firmware/commits/master/
The hexadecimal numbers on the right can be used to install that release using rpi-update. If you have the patience, try working forwards from your last known good release to see exactly where it breaks.

timg236, 2025-06-07:
Perhaps this change?

What does your config.txt file say?

https://github.com/raspberrypi/rpi-firmware/commit/f810e0a27ac6e6cf95e1d0af53c7bd72c2821932
kernel: configs: arm64: Remove CONFIG_USB_DWCOTG
See: https://github.com/raspberrypi/linux/pull/6883

solisinvictum, 2025-06-07:
The config.txt:

`# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[cm5]
dtoverlay=dwc2,dr_mode=host

[all]
`

pelwell, 2025-06-07:
Add the following to the `[all]` section of config.txt:
```
dtoverlay=dwc2,dr_mode=host
```

solisinvictum, 2025-06-07:
> Add the following to the `[all]` section of config.txt:
> 
> ```
> dtoverlay=dwc2,dr_mode=host
> ```

sure?, it is already under [cm5](?). nvm, i add it, update to latest firmware and try it out.

And sure you right, after adding this, it works with the latest firmware/kernel.

pelwell, 2025-06-13:
How did you update? There are many different ways...

solisinvictum, 2025-06-13:
rpi-update. sudo rpi-update to be precise.

pelwell, 2025-06-13:
It's fine for me, with or without the dtoverlay=dwc2 line:
```
pi@raspberrypi:~$ uname -a && vcgencmd version
Linux raspberrypi 6.12.33-v8+ #1887 SMP PREEMPT Thu Jun 12 14:21:08 BST 2025 aarch64 GNU/Linux
May 14 2025 12:25:02
Copyright (c) 2012 Broadcom
version 17084b403fb60475b8ee2641c26049a7d54bf153 (clean) (release) (start)
```
Actually, that's not quite true. Although the (USB-attached) Ethernet is working, my mouse isn't generating events with dw2 - try without it.