raspberrypi/firmware #1976: framebuffer not working

zasdf22, 2025-06-17:
the framebuffer node in start;s fdt is not tied to the video output HDMI. 
I can't do anything with it from u-boot. Guessing I need some vc4 magic to make it actually work ? The rainbow screen works, but I can't write my own data for a splash screen.
What can I do ?

zasdf22, 2025-06-17:
Yes I have a fdt /chosen/framebuffer@XXXXXX but when I write anything to its memory, it is not updated on the HDMI output as I (novice) would expect. All it does is show the rainbow square until later on linux starts the vc/v3d modules in an init script or udev I forget which.
I would like to know if there is anyway to draw to it as if it were a simple-video device ?

6by9, 2025-06-17:
I know nothing about how u-boot would be handling this. At one point I believe it used the bcm2708_fb mailbox calls, and those would potentially trash what had been set up for the simple-framebuffer.

Pi4 Raspberry Pi OS Lite.
Created `/etc/modprobe.d/blacklist-vc4.conf` containing `blacklist vc4` to stop the module loading and hence simple-framebuffer to persist.
`ls /proc/device-tree/chosen/` gives me `framebuffer@3e194000`.
`sudo memtool mw 0x3e194000 0xffffffff` gives me a white dot in the top left. Increase the offset and the white dot is further across the image.

Do remember that these are physical addresses.

pelwell, 2025-06-17:
Would you mind posting the rest of config.txt, or at least the dtoverlay/dtparam lines? There are some interesting messages encoded in your fdt:
```
dterror:   override i2c_bus: data is truncated or mangled
dterror: can't find symbol 'bt'
Failed to resolve overlay 'miniuart-bt'
```

pelwell, 2025-06-18:
Thanks. Let's go through those errors:
```
dterror:   override i2c_bus: data is truncated or mangled
```
I can now see that that message is triggered by the m41t80 overlay, which isn't one of ours. The message is saying that there is a kind of syntax error in the overlay, but it might not be enough to stop it working for you. I'm curious about where it came from, but it's not important.
```
dterror: can't find symbol 'bt'
Failed to resolve overlay 'miniuart-bt'
```
This is to be expected: the CM4S does not have on-board Bluetooth, hence the error, but because of that UART0 is already available for use on GPIOs 14 & 15 - you just need `enable_uart=1`. I presume from the use of the `uart1` overlay that you have something connected to the UART on GPIOs 32 & 33. If it's a Bluetooth modem, is it working for you? The CM4S DTB file is missing some of the infrastructure required to make it Bluetooth-capable in the modern way.

zasdf22, 2025-06-18:
GP32-33 is just a plain uart for modbus. I should clean up this file a little.
```$ cat m41t80-overlay.dts
/dts-v1/;
/plugin/;

/ {
    compatible = "brcm,bcm2711";

    fragment@0 {
        target = <&i2c_arm>;
        __overlay__ {
            #address-cells = <1>;
            #size-cells = <0>;

            rtc@68 {
                compatible = "st,m41t80";
                reg = <0x68>;
            };
        };
    };

    __overrides__ {
        i2c_bus = <&i2c_arm>,"";
    };
};

```
It does work in linux ! Thanks

zasdf22, 2025-06-18:
okay. do the vc/v3d drivers do any fiddling with the gpu to be able to write to the fdt-provided fb or does it just work, if you happen to know?

pelwell, 2025-06-18:
I think the problem is that although U-boot will create or update a simple-framebuffer node in the FDT, it won't make use of any framebuffer passed in that way. Instead it uses the mailbox API to get the firmware to create a new framebuffer with its chosen dimensions: https://github.com/u-boot/u-boot/blob/master/arch/arm/mach-bcm283x/msg.c#L159

zasdf22, 2025-06-19:
```
$ cat /etc/modprobe.d/blacklist-vc4.conf
blacklist vc4
blacklist v3d
```

dmesg
```
Populating /dev using udev: [    4.347940] udevd[141]: starting version 3.2.14
[    4.380809] udevd[143]: starting eudev-3.2.14
[    4.571718] vc_sm_cma: module is from the staging directory, the quality is unknown, you hav
e been warned.
[    4.603599] mc: Linux media interface: v0.10
[    4.603859] rpi-gpiomem fe200000.gpiomem: window base 0xfe200000 size 0x00001000
[    4.604422] bcm2835_vc_sm_cma_probe: Videocore shared memory driver
[    4.604437] [vc_sm_connected_init]: start
[    4.631429] [vc_sm_connected_init]: installed successfully
[    4.658807] rpi-gpiomem fe200000.gpiomem: initialised 1 regions as /dev/gpiomem
[    4.662196] CAN device driver interface
[    4.706621] rtc-m41t80 1-0068: registered as rtc0
[    4.721977] rtc-m41t80 1-0068: setting system clock to 2025-06-19T08:49:54 UTC (1750322994)
[    4.731994] videodev: Linux video capture interface: v2.00
[    4.732221] brcmstb-i2c fef04500.i2c:  @97500hz registered in polling mode
[    4.775217] mcp251x spi0.1 can0: MCP2515 successfully initialized.
[    4.794392] dwc2 fe980000.usb: supply vusb_d not found, using dummy regulator
[    4.803964] dwc2 fe980000.usb: supply vusb_a not found, using dummy regulator
[    4.824394] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown
, you have been warned.
[    4.829382] snd_bcm2835: module is from the staging directory, the quality is unknown, you h
ave been warned.
[    4.863205] bcm2835_isp: module is from the staging directory, the quality is unknown, you h
ave been warned.
[    4.868595] rpivid_hevc: module is from the staging directory, the quality is unknown, you h
ave been warned.
[    4.884419] bcm2835_v4l2: module is from the staging directory, the quality is unknown, you 
have been warned.
[    4.894950] bcm2835-isp bcm2835-isp: Device node output[0] registered as /dev/video13
[    4.905072] bcm2835_codec: module is from the staging directory, the quality is unknown, you
 have been warned.
[    4.910737] bcm2835-isp bcm2835-isp: Device node capture[0] registered as /dev/video14
[    4.929379] rpivid feb10000.codec: Device registered as /dev/video19
[    4.936130] bcm2835-codec bcm2835-codec: Device registered as /dev/video10
[    4.940452] bcm2835-isp bcm2835-isp: Device node capture[1] registered as /dev/video15
[    4.946350] bcm2835-codec bcm2835-codec: Loaded V4L2 decode
[    4.953971] bcm2835-isp bcm2835-isp: Device node stats[2] registered as /dev/video16
[    4.976634] bcm2835-isp bcm2835-isp: Register output node 0 with media controller
[    4.985871] bcm2835-isp bcm2835-isp: Register capture node 1 with media controller
[    4.996123] bcm2835-isp bcm2835-isp: Register capture node 2 with media controller
[    4.997132] bcm2835-codec bcm2835-codec: Device registered as /dev/video11
[    5.003785] bcm2835-isp bcm2835-isp: Register capture node 3 with media controller
[    5.013847] bcm2835-codec bcm2835-codec: Loaded V4L2 encode
[    5.026498] bcm2835-isp bcm2835-isp: Device node output[0] registered as /dev/video20
[    5.033923] dwc2 fe980000.usb: EPs: 8, dedicated fifos, 4080 entries in SPRAM
[    5.035441] bcm2835-isp bcm2835-isp: Device node capture[0] registered as /dev/video21
[    5.058430] bcm2835-isp bcm2835-isp: Device node capture[1] registered as /dev/video22
[    5.068473] bcm2835-isp bcm2835-isp: Device node stats[2] registered as /dev/video23
[    5.068743] dwc2 fe980000.usb: DWC OTG Controller
[    5.076320] bcm2835-isp bcm2835-isp: Register output node 0 with media controller
[    5.081182] bcm2835-codec bcm2835-codec: Device registered as /dev/video12
[    5.083814] dwc2 fe980000.usb: new USB bus registered, assigned bus number 1
[    5.083855] dwc2 fe980000.usb: irq 53, io mem 0xfe980000
[    5.084051] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 6.06
[    5.084066] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    5.084078] usb usb1: Product: DWC OTG Controller
[    5.084086] usb usb1: Manufacturer: Linux 6.6.28-v7l dwc2_hsotg
[    5.084095] usb usb1: SerialNumber: fe980000.usb
[    5.084671] hub 1-0:1.0: USB hub found
[    5.084714] hub 1-0:1.0: 1 port detected
[    5.088573] bcm2835-isp bcm2835-isp: Register capture node 1 with media controller
[    5.095935] bcm2835-codec bcm2835-codec: Loaded V4L2 isp
[... 23 lines cut ...]
compatible            overlay_prefix        user-warnings
framebuffer@3e9ea000  phandle
# ls /proc/device-tree/chosen/framebuffer@3e9ea000
compatible  height      reg         stride
format      name        status      width
# ./memtool mw 0xe39ea000 0xffffffff
# ./memtool mw 0xe39ea001 0xffffffff
# ./memtool mw 0xe39ea002 0xffffffff
# ./memtool mw 0xe39ea003 0xffffffff
# ./memtool mw 0xe39ea004 0xffffffff
# ./memtool mw 0xe39ea005 0xffffffff
# chvt 1
# chvt 7
# ./vcgencmd dispmanx_list
display:0 format:XRGB8888 transform:0 layer:-127 800x600 src:0,0,800,600 dst:0,0,640,480 cost:7
42 lbm:10240 
display:2 format:RGB565 transform:0 layer:-127 2x2 src:0,0,2,2 dst:100,0,600,600 cost:453 lbm:1
024 
# echo $?
0
# 
```

the `memtool` calls had no visible effect. 
any ideas? Thanks for your time

popcornmix, 2025-06-19:
Note: The rainbow is not displayed in the framebuffer.
It is a simple 2x2 pixel test pattern scaled up.

If you are seeing the rainbow then you'll have no ability to alter the pixels by poking memory indicated by `/chosen/framebuffer`.
The normal procedure is for the rainbow to be replaced with that framebuffer before the kernel is launched, but that is not happening with your configuration for some reason.

zasdf22, 2025-06-19:
> Note: The rainbow is not displayed in the framebuffer. It is a simple 2x2 pixel test pattern scaled up.
>
> If you are seeing the rainbow then you'll have no ability to alter the pixels by poking [...]

I used `db` variety of start and fixup (4) files. Maybe this boot log has any useful info ?
```
  0.52 RPi: BOOTSYS release VERSION:69471177 DATE: 2025/05/08 TIME: 16:21:35
  0.52 BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1746717695 serial 4cdc7421 boardrev a03150 stc 528159
  0.53 PM_RSTS 00000020
  0.54 POWER_OFF_ON_HALT: 1 WAIT_FOR_POWER_BUTTON 0 power-on-reset 0
  0.54 part 00000000 reset_info 00000000
  0.54 uSD voltage 3.3V
  0.57 Initialising SDRAM rank 1 total-size: 8 Gbit 3200 (0x07 0x00)
  0.57 DDR 3200 0 0 8 152 BL:3
  1.53 OTP boardrev a03150 bootrom c8b0 c8b0
  1.53 Customer key hash 3c26a85c347c623373c9a66d3cd572f79b3f8daa22208f84562260a5c57c411f
  1.54 VC-JTAG unlocked
  1.57 bootconf.sig
  1.57 hash: b503f8ad7f8aea93a272a5ba5248cc5222d0c55af28a4345d0a496bdba9d16bf
  1.57 rsa2048: 6bd18f209d06f4dc391b25493debf0c4dc8a63cb2a3c65ba4bffa5f7b016a73cac477c00e39d6ba2dc06d21b15cf7bb1c912d11cc34e7ef04ee0388e4b9f497a8c38d262bfe3792a3c8af33cff4b44b81653e6b57ec366f3016d206120691b5e8b5563fbc116313c5a1b0dcbda3ed07f0c0862386c138cbabcdd1217d927596e4213d0967639e10775495a78c5b35c2cfef9673251a61945c4350fd3877a19808446e5d0a62b4ae585a27508d6217370c4406d40b70e11f06169dabdb10cf71bf5043a2d0ee15828898adbbf5ba13d1d78e3f01302da157311fd2c3b934b609c54898b2433904e0347ec95e9a6104307094227961964e2c26da9ae6274c51c10
  1.62 RSA verify
  1.39 rsa-verify pass (0x0)

  1.88 RPi: BOOTLOADER release VERSION:69471177 DATE: 2025/05/08 TIME: 16:21:35
  1.88 BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1746717695 serial 4cdc7421 boardrev a03150 stc 1888124
  1.98 Unsupported boot order 4
  1.17 Unsupported boot order 6
  1.20 Unsupported boot order 2
  1.28 Boot mode: SD (01) order f5
  2.07 SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
  2.08 SD HOST: 200000000 CTL0: 0x00800f00 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
  2.19 EMMC
  2.29 SD retry 1 oc 0
  2.32 SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
  2.33 OCR c0ff8080 [0]
CID: 00150100384754463452061493ba5d6a
  2.35 SD HOST: 200000000 CTL0: 0x00800f04 BUS: 25000000 Hz actual: 25000000 HZ div: 8 (4) status: 0x1fff0000 delay: 4
  2.36 SD HOST: 200000000 CTL0: 0x00800f04 BUS: 50000000 Hz actual: 50000000 HZ div: 4 (2) status: 0x1fff0000 delay: 2
  2.37 MBR: 0x00000001,  262144 type: 0x0c
  2.37 MBR: 0x00040001,    2048 type: 0x0c
  2.37 MBR: 0x00040801,    2048 type: 0x0c
  2.38 MBR: 0x00041001,13778949 type: 0x0f
  2.86 Trying partition: 0
  2.00 type: 16 lba: 1 'mkfs.fat' '  V       ^ ' clusters 65399 (4)
  2.40 rsc 4 fat-sectors 256 root dir cluster 1 sectors 32 entries 512
  2.40 FAT16 clusters 65399
  2.41 [sdcard] autoboot.txt not found
  2.41 Select partition rsts 0 C(boot_partition) 0 EEPROM config 0 result 1
  2.22 Trying partition: 1
  2.24 type: 16 lba: 1 'mkfs.fat' '  V       ^ ' clusters 65399 (4)
  2.43 rsc 4 fat-sectors 256 root dir cluster 1 sectors 32 entries 512
  2.43 FAT16 clusters 65399
  2.39 secure-boot
  2.41 Loading boot.img ...
  2.44 boot.sig
  2.44 hash: 8ff84416629136f3d22e62101416306fa671a492324a69b3efe56135f4464139
  2.45 ts: 1750327846
  2.45 rsa2048: a90225f2c857e6013983a4a1afac329c88fdaf6a224fab247db3d9759dad727481d78de33930a91e7cbf91756952ae568925bee925fcbd0944b518dcd22bfa10a72efacf934f4afe47c6e27a3d07239126a88667f819b4d8ee27ed85a203549c8c92eabe1dad0486f9d9a734e6d1fb6c459881c9a2e64aab986f43d465ba6e5ac539de7c5bf76181ee029cdddbbc4122ddb2eeaf0aabf797a227c05a9b74b1e153c667519b7e70fb40abd46081a43fc96ea94b27b94e6f7d65a0e6857e45db791dafe10f4728652fbbaed01cc7d9164d995e764b1b1528ce685fd2b178eabb30b3d6d5ffafea5c556a330ab525527f3486acde241e2967f48d8f8829832add53
  3.13 Verifying
  3.99 RSA verify
  4.08 rsa-verify pass (0x0)
  4.03 MBR: 0x00000000,       0 type: 0x00
  4.03 MBR: 0x00000000,       0 type: 0x00
  4.04 MBR: 0x00000000,       0 type: 0x00
  4.04 MBR: 0x00000000,       0 type: 0x00
  4.49 Trying partition: 0
  4.52 type: 16 lba: 0 'mkfs.fat' '  V       ^ ' clusters 4589 (4)
  4.05 rsc 4 fat-sectors 20 root dir cluster 1 sectors 32 entries 512
  4.06 FAT16 clusters 4589
  4.68 Read config.txt bytes      833 hnd 0xedd
[... 70 lines cut ...]
MESS:00:00:07.843323:0: Loaded overlay 'ads7846'
MESS:00:00:07.844862:0: dtparam: cs=0
MESS:00:00:07.849465:0: dtparam: penirq=26
MESS:00:00:07.854571:0: dtparam: xohms=600
MESS:00:00:07.857176:0: dtparam: settle=5
MESS:00:00:07.867303:0: Unknown dtparam 'settle' - ignored
MESS:00:00:07.869708:0: dtparam: speed=100000
MESS:00:00:07.990969:0: brfs: File read: 2402 bytes
MESS:00:00:07.997865:0: brfs: File read: /mfs/sd/overlays/uart1.dtbo
MESS:00:00:08.018016:0: Loaded overlay 'uart1'
MESS:00:00:08.019384:0: dtparam: txd1_pin=32
MESS:00:00:08.023930:0: dtparam: rxd1_pin=33
MESS:00:00:08.110350:0: brfs: File read: 988 bytes
MESS:00:00:08.112274:0: brfs: File read: /mfs/sd/cmdline.txt
MESS:00:00:08.117474:0: Read command line from file 'cmdline.txt':
MESS:00:00:08.123346:0: 'rootwait console=tty1 fbcon=map:0 console=ttyAMA0,115200 cma=256M'
MESS:00:00:08.191931:0: CMA size reduced to 256MB (total_mem 1024MB, gpu_mem 64MB)
MESS:00:00:08.288561:0: brfs: File read: 66 bytes
MESS:00:00:08.295001:0: brfs: File read: /mfs/sd/u-boot.bin
MESS:00:00:08.297468:0: Loaded 'u-boot.bin' to 0x8000 size 0x91eb4
MESS:00:00:08.303385:0: Device tree loaded to 0x2600000 (size 0xd961)
MESS:00:00:08.311648:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:08.318607:0: uart: Baud rate change done...
MESS:00:00:08.320626:0: uart: Baud rate change done...
```

6by9, 2025-06-19:
```
# ./vcgencmd dispmanx_list
display:0 format:XRGB8888 transform:0 layer:-127 800x600 src:0,0,800,600 dst:0,0,640,480 cost:742 lbm:10240 
display:2 format:RGB565 transform:0 layer:-127 2x2 src:0,0,2,2 dst:100,0,600,600 cost:453 lbm:1024 
```
You have 2 displays configured. 
From your config.txt
```
# force 800x600 60Hz
hdmi_force_hotplug=1
hdmi_group=2
hdmi_mode=9
framebuffer_width=800
framebuffer_height=600
enable_dpi_lcd=1
display_lcd_rotate=0
```
display 0 is the DPI output and has an 800x600 plane on it.
display 2 is HDMI and still has the rainbow screen on it (it should have magenta in the top left as it is the secondary screen). The simple-framebuffer is only ever routed to one display.
Are you looking at the wrong screen?

6by9, 2025-06-19:
Thinking further, I'm not sure how you're expecting the DPI display to work when vc4 does take over - you appear to have no configuration that tells it to enable DPI. I'd expect a `dtoverlay=vc4-kms-dpi-generic,<timings>` or similar.

6by9, 2025-06-19:
Didn't make it clear?

> There is only one screen. DPI -> TFP401-> LVDS driver-> display

That explains most things.

Based on the config.txt, DPI would be the primary display and get the framebuffer. HDMI would be the secondary display and get the secondary rainbow screen (magenta in top left, not red).

Remove `enable_dpi_lcd=1` and HDMI should become the primary display and get the framebuffer.
If you haven't got HDMI hotplug asserted, then you'll need `hdmi_force_hotplug=1` to get the firmware to initialise the HDMI output. 

You'll also need to force enabling HDMI via a `video=` cmdline.txt line otherwise vc4 won't initialise the output.

zasdf22, 2025-06-20:
> Didn't make it clear?
>
>
> That explains most things.
>
> Based on the config.txt, DPI would be the primary display and get the framebuffer. HDMI would be the secondary display and get the secondary [...]

yeah sorry im a twat wasting your time. really, an idiot. but with all that still nothing. functions perfectly in linux.
can you get a hdmi fb before linux?

zasdf22, 2025-06-20:
I got it to show a picture on the hdmi fb now. It's a bit wonky but I will edit this comment and post my settings for everyone that may run into this problem later. can close if im allowed to edit posts afterwards

zasdf22, 2025-06-20:
@darkelvenangel you wanted to know how to do it