raspberrypi/firmware #1980: Raspberry Pi 4b ends up in a reboot loop since commit #3885768 running Zephyr

Snenz, 2025-07-11:
I'm trying to boot [Zephyr](https://docs.zephyrproject.org/latest/boards/raspberrypi/rpi_4b/doc/index.html) on a Raspberry Pi 4b. With the changes made to `boot/start4.elf` in commit https://github.com/raspberrypi/firmware/commit/3885768a3aad70f42a9d9a85b550046b6b9d83fe the Pi ends up in a reboot loop, continuously rebooting after around 17 seconds of uptime.

This appears to be related to the new watchdog option mentioned in the commit message? \
"`firmware: arm_loader: Add config option to enable the watchdog timer from the firmware`"

After adding the option `dtparam=watchdog=off` to my `config.txt`, everything seems to be working fine again. Is there a new option to disable the watchdog in a different way in the `config.txt` (regarding the commit message)?

My main concern with this change is that the behaviour unexpectedly changed from the opt-in to an opt-out behaviour. Was this intended?

timg236, 2025-07-11:
Does this happen with RPi OS? The watchdog shouldn't be enabled automatically. Please also attach config.txt files.

timg236, 2025-07-11:
That's strange. I'm not seeing reboot loops with the following firmware and PiOS kernel. The firmware shouldn't be enabling the watchdog register unless it has kernel_watchdog in config.txt 

```
pi@tim-pi4:~ $ vcgencmd version
Jul  7 2025 11:44:22
Copyright (c) 2012 Broadcom
version ab660caaca8f4f8d1f3dd7a350f8461c92e0b551 (clean) (release) (start)
pi@tim-pi4:~ $ vcgencmd bootloader_version
2025/07/03 15:56:45
version ab660caaca8f4f8d1f3dd7a350f8461c92e0b551 (release)
timestamp 1751554605
update-time 1751556477
capabilities 0x0000007f

```

Please could you run 'sudo vclog -m' and share the output if you have that on Zephyr ?

timg236, 2025-07-11:
Ok, I'll have to give this some more thought and possibly attach a fix candidate next week. The watchdog shouldn't be getting set, if it is then the bcm2835_wdt driver should take ownership so first step will be check watchdog hardware just prior to starting the kernel

iamgcb, 2025-07-12:
I am running Debian 12 on my RPi 4 Model B Rev 1.1 and encountering the same issue boot-loop after complete booting.
Changed back to 6.12.35.

timg236, 2025-07-14:
It would also be useful to know if the watchdog driver is enabled and how quickly it reboots e.g. random or within 30seconds of power on.

```
dmesg | grep -i watchdog
cat /proc/cmdline
```

timg236, 2025-07-14:
I've attached a possible firmware fix here, but it would still be useful to know the exact kernel versions in case there is some unexpected interaction there.

[pi4-watchdog-issue-1980.zip](https://github.com/user-attachments/files/21219337/pi4-watchdog-issue-1980.zip)

iamgcb, 2025-07-14:
> It would also be useful to know if the watchdog driver is enabled and how quickly it reboots e.g. random or within 30seconds of power on.
> 
> ```
> dmesg | grep -i watchdog
> cat /proc/cmdline
> ```
Reboot Loop: Constantly rebooting around 60 seconds after complete booting (at login screen/SSH access made available for remote)
Rollback: 6.12.35+ (Stable)

Below are the information requested:
dmesg | grep -i watchdog
[    0.734487] bcm2835-wdt bcm2835-wdt: Broadcom BCM2835 watchdog timer

cat /proc/cmdline
coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 cgroup_disable=memory numa_policy=interleave nvme.max_host_mem_size_mb=0 snd_bcm2835.enable_hdmi=0  numa=fake=2 system_heap.max_order=0 smsc95xx.macaddr=*********** vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=fe4104a4-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait cfg80211.ieee80211_regdom=US

uname -a
Linux myRPi 6.12.36-v8+ #1892 SMP PREEMPT Mon Jul  7 17:12:23 BST 2025 aarch64 GNU/Linux
Linux myRPi 6.12.37-v8+ #1894 SMP PREEMPT Fri Jul 11 16:47:45 BST 2025 aarch64 GNU/Linux

vcgencmd bootloader_version
2025/07/03 15:56:45
version ab660caaca8f4f8d1f3dd7a350f8461c92e0b551 (release)
timestamp 1751554605
update-time 1751713993
capabilities 0x0000007f

Adding dtparam=watchdog=off fixed the issue in **6.12.36/6.12.37**.

Snenz, 2025-07-15:
I will also test the possible fix with Zephyr, but can't do that until tomorrow.

Snenz, 2025-07-16:
> I've attached a possible firmware fix here, but it would still be useful to know the exact kernel versions in case there is some unexpected interaction there.
> 
> [pi4-watchdog-issue-1980.zip](https://github.com/user-attachments/files/21219337/pi4-watchdog-issue-1980.zip)

Unfortunately this fix candidate does not resolve the problem. The Pi still reboots after exactly the same time as before (around 17 seconds after successful bootup).

iamgcb, 2025-07-16:
> Thanks for the update.
> 
> How did you do the rollback e.g. rpi-update?
> 
> Does using attached firmware in the previous comment resolve the issue on 6.12.37 ?

Yes, I am using the rpi-update to rollback.
Also had some issues cause pulling out and plugging in broke my SD card slot, so I have migrated my setup from SD card to boot from USB now.

Tested with the attached file on 6.12.37, still rebooting constantly with watchdog config disabled (#dtparam=watchdog=off)

Snenz, 2025-07-16:
> [...] that would help determine whether this is kernel/firmware change or both.

Well, as I'm using Zephyr with its own kernel which has nothing to do with Linux, doesn't this imply that the problem must be in the firmware change as I'm only using the firmware files? Using the same Zephyr binary but changing the firmware version is what caused the issues for me. Or did you mean something different?

(Currently the Raspberry Pi 4's watchdog functionality isn't even supported by Zephyr.)

timg236, 2025-07-16:
The firmware will enable the watchdog the device-tree contains the watchdog symbol and it is enabled.

The device-tree listed here for Zephyr seems to have the watchdog node enabled so I would expect the watchdog to be enabled and therefore Zephyr would need to cancel the watchdog OR not enable it in device-tree.
https://docs.zephyrproject.org/latest/boards/raspberrypi/rpi_4b/doc/index.html#tf-card

I'm checking whether the watchdog could have not been enabled by device-tree in older firmware.

timg236, 2025-07-16:
The change in behaviour is related to #1970 and it's a coincidence that this appeared in the same release as other watchdog changes.

If a device-tree node has a property but no status node, then the kernel should treat it as "enabled". The watchdog node in Zephyr is present but does not have a status node so following #1970 the watchdog will now be enabled. 

My guess is that Debian 12 has a similar device-tree but doesn't have the downstream bcm2835_wdt driver or possibly something is taking more than 17s before the driver runs e.g. u-boot?

I'll ask others at PiTowers to think about whether we should do something special for the watchdog node.  Although, I would recommend that the Zephyr project remove the unused watchdog device-tree node.

Snenz, 2025-07-16:
Oh, so this behaviour was changed in June, but just now surfaced in this commit's change to start4.elf? Thanks for digging :)

rpodgorny, 2025-07-17:
i've been hit by this lately on my arch linux arm system. i confirm adding `dtparam=watchdog=off` to config.txt resolved the boot loop and now my systems boots up properly again.

Snenz, 2025-07-21:
Works for me too, thanks.

closed 2025-07-21 by Snenz

timg236, 2025-07-22:
Thanks for verifying!