raspberrypi/firmware #1998: RPi 5 bootloader unusable due to (lack of consideration for) overscan

ali1234, 2025-09-30:
[image]

**Describe the bug**
The RPi5 bootload menu and QR code thing are unreadable because they are outside the television caption safe area. See screenshot.

1. Have a RPi 5.
2. Plug it into a TV and turn it on.

**Expected behaviour**
Anything important on the screen should be displayed inside the caption safe area, so that it is visible on any TV.

**Actual behaviour**
All the important information is outside the caption safe area, so it is not visible.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* RPi 5 4GB
* OS: None
* Firmware: Whatever version I got by running `apt update` about five minutes ago.
* Kernel: None

ali1234, 2025-09-30:
Yes, this is on a HDMI connection. Yes, HDMI can have overscan on "HD ready" TVs.

pelwell, 2025-09-30:
Is that the best that TV can do, or does playing with the aspect ratio button render the UI visible?

timg236, 2025-09-30:
Thanks for the feedback. I think this out of scope for the current bootloader menu but could be considered as part of a wider redesign.